stack-of-tasks/pinocchio #1798: Shared pointers from boost and std

jmirabel, 2022-11-25:
Since https://github.com/humanoid-path-planner/hpp-fcl/issues/218, hpp-fcl uses `std::shared_ptr`. `GeometryObject::geometry` is however still a `boost::shared_ptr`.

This prevents someone from doing
```c++
// go is a GeometryObject
go.geometry = bvh_model->convex
```

This is done, e.g., [here](https://github.com/humanoid-path-planner/hpp-pinocchio/blob/2aa93acdc7b1a21260f416323046eb4e81fc856f/src/device.cc#L507)

I'm surprised I'm the only one to have encountered this issue. @nim65s may understand the reason.

nim65s, 2022-11-25:
You're not the only one. Everybody is blocked for this: HPP users must use hpp-fcl < 2 for pinocchio < 3.
We discussed a workaround for this in https://github.com/humanoid-path-planner/hpp-pinocchio/pull/158, but this wasn't a good idea. Pinocchio 3 will solve this by using `std::shared_ptr` for `GeometryObject::geometry`.

jmirabel, 2022-11-25:
At the place where I have issue, the fix is easy. I don't know whether there are many places like this. Let me check.

jmirabel, 2022-11-25:
I made a series of PR in HPP that implements a work around for the issue.

jcarpent, 2023-01-16:
Solve via #1822.

closed 2023-01-16 by jcarpent