stack-of-tasks/pinocchio #1800: Experimental Test, Kinova Jaco 2 spherical

Betancourt20, 2022-11-28:
Hi,
Thanks a lot for the framework !! 
Sorry for my following dumb question:
I did some examples in python with different controller for the kinova jaco 2 spherical (6 dof) and know I compiled everything in C++ with the API of kinova. The thing is that, I do not know if I should still using the URDF model?  
I left the arm in a home position and I got the states (q and $\dot{q}$) . Then I compute the torque as it is mentioned in the example but the torques that I'm obtaining are not at all that ones I should get. 

Thanks in advance for everything !:
Julio

jcarpent, 2022-11-28:
Hi @Betancourt20,
Could you provide a code associated to your experience with Pinocchio?
Without it, we cannot be very helpful.

Betancourt20, 2022-11-28:
Thank you for your answer.

I'll put the essential code, but If it is needed, I will share as much as you ask. 

```cpp
#include "pinocchio/fwd.hpp"
#include "pinocchio/parsers/sample-models.hpp"
#include "pinocchio/algorithm/joint-configuration.hpp"
#include "pinocchio/algorithm/kinematics.hpp"
#include "pinocchio/algorithm/rnea.hpp"
#include "pinocchio/algorithm/aba.hpp"
#include "pinocchio/algorithm/crba.hpp"
#include "pinocchio/parsers/urdf.hpp"
#include <iostream>
#include "Node.hpp"
#include "kinova_driver/kinova_comm_eigen.h"
#include <yaml-cpp/yaml.h>
#include <dlfcn.h>
#include <unistd.h>
#include <math.h>
#include <std_msgs/msg/float64_multi_array.hpp>
#include <Eigen/Dense>

# define M_PI           3.14159265358979323846  
using std::placeholders::_1;
ManipulatorSkillsetLichieNode::ManipulatorSkillsetLichieNode()
                : SKILLSET_NODE("manipulator_skillset", "lichie"),
                api_mutex_(),
                ref_reach_(false),
                active_control_(false),
                type_controller(ANGULAR_)

{
    this->declare_parameter<std::string>("kinova_robotType_", "");
    this->declare_parameter<std::string>("CommParameter", "");

    std::string commParametersPath_;
    this->get_parameter("CommParameter", commParametersPath_);
    RCLCPP_INFO(this->get_logger(), "Parsing CommParameter file %s", commParametersPath_.c_str());

    pugi::xml_document doc;
    pugi::xml_parse_result result_xml = doc.load_file(commParametersPath_.c_str());
    if(!result_xml){
        std::cout << "File input " << commParametersPath_.c_str() <<  std::endl;
        throw std::runtime_error("XML input file not found");
    }

    ee_pub_ = this->create_publisher<PoseStamped>("~/manipulator_skillset/end_effector", 10);
    joint_pub_ = this->create_publisher<sensor_msgs::msg::JointState>("~/manipulator_skillset/joint", 10);
    ee_wrench_pub_ = this->create_publisher<WrenchStamped>("~/manipulator_skillset/end_effector_wrench", 10);
    torque_pub_ = this->create_publisher<sensor_msgs::msg::JointState>("~/manipulator_skillset/torques", 10);

    this->comm_= std::make_unique<kinova::KinovaComm>(commParametersPath_, api_mutex_, kinova_robotType_);

    QuickStatus quick_status;
    GeneralInformations info;
    this->comm_->getQuickStatus(quick_status);

    if (this->comm_->isHomed()){
      std::cout << "Arm is already in \"home\" position" << std::endl;
    }
    else{
      this->comm_->setAngularControl(); // I'm not sure this is the best way
      this->comm_->homeArm(); // home position by definition
    }

    timer_ = this->create_wall_timer(
        10ms, [this]()
        {try {
          this->state_callback_();
          usleep(5000);
          this->control_callback_();            
        }
        catch (kinova::KinovaCommException &e)
        {
            std::cout << "Ignore KinovaCommException"<<std::endl;
        }
 	});

   timer_Pub = this->create_wall_timer(
        100ms, [this]()
        {
            this->pose_callback_();
        });

}
void ManipulatorSkillsetLichieNode::state_callback_()
{
  this->comm_->getJointAngles(joint_pos_);
  this->comm_->getJointVelocities(joint_vel_);
  this->comm_->getCartesianPosition(ee_pos_);
  this->comm_->getCartesianForce(cart_force);
  this->comm_->getJointTorques(tqs_commands);

  Eigen::Matrix<float, 6, 1> mes_art;
  Eigen::Matrix<float, 6, 1> mes_vel;
  joint_pos_.getArt(mes_art);
  joint_vel_.getArt(mes_vel);

  const std::string urdf_filename = "/home/unknown/Documents/Posdoc/ModelsCads/urdf" +   std::string("/kinova_description/robots/kinova.urdf");
  pinocchio::Model model;
  pinocchio::urdf::buildModel(urdf_filename,model);
  pinocchio::Data data(model);

  Eigen::VectorXd q = pinocchio::neutral(model);
  Eigen::VectorXd q_des = pinocchio::neutral(model);
  Eigen::VectorXd qdot = Eigen::VectorXd::Zero(model.nv);
  Eigen::VectorXd qddot = Eigen::VectorXd::Zero(model.nv);
  Eigen::VectorXd a = Eigen::VectorXd::Zero(model.nv);

  q << mes_art(0), mes_art(1), mes_art(2), mes_art(3), mes_art(4), mes_art(5);
  qdot << mes_vel(0), mes_vel(1), mes_vel(2), mes_vel(3), mes_vel(4), mes_vel(5);

  real_taus <<  tqs_commands.Actuator1, tqs_commands.Actuator2, tqs_commands.Actuator3,
  tqs_commands.Actuator4, tqs_commands.Actuator5, tqs_commands.Actuator6;
  
  std::cout << "real_taus: " << real_taus.transpose() << std::endl;

  const Eigen::VectorXd & tau = pinocchio::rnea(model,data,q,qdot,a);
  std::cout << "tau = " << tau.transpose() << std::endl;

```
I'm trying to compare in open loop what pinocchio gives me against what the API says.
Thanks a lot !

jcarpent, 2022-11-28:
If the robot configuration vectors, velocity, acceleration do not match between the simulated and the real robots, this is expected that the torques are not the same.

Betancourt20, 2022-11-28:
I completely agree with you. But here, I'm trying to compare only experimental tests: what the Kinova API gives vs what Pinocchio gives me.
My question is, even in experimental tests, am I going to use all the time model from the URDF file ? 
Thanks

jcarpent, 2022-11-28:
I don't understand your question. I do not think it is related to Pinocchio itself.
You should rather see with your post-doc advisor at ONERA.

jcarpent, 2022-11-28:
Otherwise, please provide the values that you obtained.

Betancourt20, 2022-11-28:
Thank you for your answer. For sure, I was not clear with my questions.
Let's try to reformulate it.  The manipulator is in a home position (deg) :
```
q = 76.8487 197.207 316.553 94.8308 102.313 72.0666
```
The kinova API is reading that as the position and the torque sensors are reading
```
real_taus:  0.109529  0.709634  -4.71262  0.707568 -0.138458  -0.09295
```
but when computing 
```cpp
  const Eigen::VectorXd & tau = pinocchio::rnea(model,data,q,qdot,a);
  std::cout << "tau = " << tau.transpose() << std::endl;
```
gives me 
```
tau =   -0.0107606      9.42613     -1.27369    -0.034494      1.76573 -2.80236e-07
```
I do not know if it is related with the model or something else?
The configuration of the arm is as
[image]

jcarpent, 2022-11-28:
Your q is in degrees !!!

jcarpent, 2022-11-28:
You should translate them into radians of course.

closed 2022-11-28 by Betancourt20