stack-of-tasks/pinocchio #1805: Unable to compile pinocchio3-preview branch with hpp-fcl

MariaBana, 2022-12-09:
Hello.
I tried to compile pinocchio3-preview with hpp-fcl on ubuntu 20.04, but I got this error.
```
/home/mao/Desktop/Third-Party/pinocchio/build/include/pinocchio/multibody/geometry.hxx: In constructor ‘pinocchio::GeometryData::GeometryData(const pinocchio::GeometryModel&)’:
/home/mao/Desktop/Third-Party/pinocchio/build/include/pinocchio/multibody/geometry.hxx:47:75: error: no matching function for call to ‘hpp::fcl::CollisionObject::CollisionObject(const CollisionGeometryPtr&)’
   47 |       collisionObjects.push_back(fcl::CollisionObject(geom_object.geometry));
      |                                                                           ^
In file included from /home/mao/Desktop/Third-Party/pinocchio/build/include/pinocchio/multibody/fcl.hpp:37,
                 from /home/mao/Desktop/Third-Party/pinocchio/build/include/pinocchio/multibody/geometry.hpp:8,
                 from /home/mao/Desktop/Third-Party/pinocchio/build/include/pinocchio/parsers/urdf.hpp:10,
                 from /home/mao/Desktop/Third-Party/pinocchio/src/parsers/urdf/model.cpp:6:
```
I should have executed cmake normally, right? This is the output of the cmake step.
```
mao@mao:~/Desktop/Third-Party/pinocchio/build$ cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DBUILD_WITH_COLLISION_SUPPORT=ON -DBUILD_PYTHON_INTERFACE=ON
-- Package version (ROS package.xml): 2.9.1
-- Default C++ standard: 201402
-- C++ standard sufficient: Minimal required 11, currently defined: 14
-- Found Boost: /usr/include (found version "1.71.0") found components: filesystem serialization system 
-- The Python bindings of Pinocchio will be compiled along the main library. If you want to disable this feature, please set the option BUILD_PYTHON_INTERFACE to OFF.
-- eigenpy FOUND. eigenpy at /opt/ros/noetic/lib/x86_64-linux-gnu/libeigenpy.so
-- PythonLibraryDirs: /usr/lib/x86_64-linux-gnu
-- PythonLibVersionString: 3.8.10
-- Python site lib: lib/python3/dist-packages
-- Python include dirs: /usr/include/python3.8
-- Checking for NumPy
--   NUMPY_INCLUDE_DIRS=/usr/lib/python3/dist-packages/numpy/core/include
--   NUMPY_VERSION=1.17.4
-- NumPy include dir: /usr/lib/python3/dist-packages/numpy/core/include
-- Found Boost: /usr/include (found version "1.71.0") found components: python38 
-- Boost_PYTHON_LIBRARY: /usr/lib/x86_64-linux-gnu/libboost_python38.so
-- Python compiler: CPython
-- hpp-fcl FOUND. hpp-fcl at /opt/ros/noetic/lib/x86_64-linux-gnu/libhpp-fcl.so
-- Found Boost: /usr/include (found version "1.71.0") found components: chrono serialization 
-- C++ standard sufficient: Minimal required 11, currently defined: 14
-- Found hpp-fcl Python bindings.
-- Found Boost: /usr/include (found version "1.71.0") found components: unit_test_framework 
-- Doxygen rendering: using MathJax backend
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mao/Desktop/Third-Party/pinocchio/build
```
I installed hpp-fcl by installing ros-noetic-hpp-fcl.
For reference, I was able to compile pinocchio3 without HPP-FCL dependency and pinocchio2 with HPP-FCL dependency.

jcarpent, 2022-12-09:
Could you provide a reproducing docker image to analyse your issue?

MariaBana, 2022-12-10:
@jcarpent 
I tried it and found that it seems to be very troublesome to do so and the generated image will be very large. 
What do you think might be the cause of this compilation problem and can you give me some ideas to solve it? Thank you very much for your guidance.

jcarpent, 2022-12-11:
Please provide the docker image

jcarpent, 2022-12-11:
@MariaBana you can also use https://hub.docker.com/ to share easily your images.

fabinsch, 2022-12-12:
Hello @jcarpent and @MariaBana, I created a docker container with ros-noetic, ros-noetic-hpp-fcl and the source code of Pinocchio on the pinocchio3-preview branch and I can see the same compile errors. You can reproduce it by:
```
docker pull fabinsch/pinocchio3-preview-hppfcl-ros-noetic
docker run -it fabinsch/pinocchio3-preview-hppfcl-ros-noetic /bin/bash
cmake .. -DCMAKE_BUILD_TYPE=Release -DBUILD_PYTHON_INTERFACE=OFF -DBUILD_TESTING=OFF -DBUILD_WITH_COLLISION_SUPPORT=ON
```

fabinsch, 2022-12-12:
I think the compile problems that we are experiencing here are related to the discussions in #1798 . The error in the docker image is:
```
/home/workspace/pinocchio/build/include/pinocchio/multibody/geometry.hxx:47:75: error: no matching function for call to ‘hpp::fcl::CollisionObject::CollisionObject(const CollisionGeometryPtr&)’
   47 |       collisionObjects.push_back(fcl::CollisionObject(geom_object.geometry));
      |                                                                           ^
In file included from /home/workspace/pinocchio/build/include/pinocchio/multibody/fcl.hpp:37,
                 from /home/workspace/pinocchio/build/include/pinocchio/multibody/geometry.hpp:8,
                 from /home/workspace/pinocchio/build/include/pinocchio/parsers/urdf.hpp:10,
                 from /home/workspace/pinocchio/src/parsers/urdf/model.cpp:6:
/opt/ros/noetic/include/hpp/fcl/collision_object.h:229:3: note: candidate: ‘hpp::fcl::CollisionObject::CollisionObject(const std::shared_ptr<hpp::fcl::CollisionGeometry>&, const Matrix3f&, const Vec3f&, bool)’
  229 |   CollisionObject(const shared_ptr<CollisionGeometry>& cgeom_,
      |   ^~~~~~~~~~~~~~~
/opt/ros/noetic/include/hpp/fcl/collision_object.h:229:3: note:   candidate expects 4 arguments, 1 provided
/opt/ros/noetic/include/hpp/fcl/collision_object.h:223:3: note: candidate: ‘hpp::fcl::CollisionObject::CollisionObject(const std::shared_ptr<hpp::fcl::CollisionGeometry>&, const hpp::fcl::Transform3f&, bool)’
  223 |   CollisionObject(const shared_ptr<CollisionGeometry>& cgeom_,
      |   ^~~~~~~~~~~~~~~
/opt/ros/noetic/include/hpp/fcl/collision_object.h:223:3: note:   candidate expects 3 arguments, 1 provided
/opt/ros/noetic/include/hpp/fcl/collision_object.h:217:3: note: candidate: ‘hpp::fcl::CollisionObject::CollisionObject(const std::shared_ptr<hpp::fcl::CollisionGeometry>&, bool)’
  217 |   CollisionObject(const shared_ptr<CollisionGeometry>& cgeom_,
      |   ^~~~~~~~~~~~~~~
/opt/ros/noetic/include/hpp/fcl/collision_object.h:217:56: note:   no known conversion for argument 1 from ‘const CollisionGeometryPtr’ {aka ‘const boost::shared_ptr<hpp::fcl::CollisionGeometry>’} to ‘const std::shared_ptr<hpp::fcl::CollisionGeometry>&’
  217 |   CollisionObject(const shared_ptr<CollisionGeometry>& cgeom_,
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
```
The last line saying: 
`no conversion from ‘const boost::shared_ptr<hpp::fcl::CollisionGeometry>’} to ‘const std::shared_ptr<hpp::fcl::CollisionGeometry>&’`

Maybe @nim65s you have an idea how we could solve this problem here ?

jmirabel, 2022-12-13:
In pinocchio3, we have to make a choice:

1. use hpp-fcl shared pointers
2. do not use CollisionObject from hpp-fcl and reimplement them in pinocchio. In pinocchio 2, this was done for the ComputeCollision and ComputeDistance objects.

I am strongly in favor of 2 as it facilitates using hpp-fcl API with objects stored in a Pinocchio geometry model.

fabinsch, 2022-12-16:
hi @MariaBana, https://github.com/stack-of-tasks/pinocchio/pull/1813 should fix your problem. You can try again now if it works for you.