stack-of-tasks/pinocchio #1806: pin2.6.4 w. Python 3.7 fails to import

JonathanKuelz, 2022-12-09:
Dear pinocchio team,

since today (last time it worked as expected was 2 days ago, so not sure when exactly), using pin2.6.4 with Python 3.7 fails for me. Checked it again in a docker container, seems to be related to a problem with `libeigenpy`. Here's a dockerfile to reproduce:

```
FROM python:3.7

RUN pip install --no-cache-dir pin==2.6.4

CMD [ "python", "-c", "from pinocchio import *" ]
```

raises:
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.7/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ImportError: libeigenpy.so: cannot open shared object file: No such file or directory
```

JonathanKuelz, 2022-12-09:
It's not a random configuration insofar as pin 2.6.4 is the latest one that is accessible on pypi for Python 3.7 - as long as `Python >=3.7 ` is listed as the py-version requirement on PyPI, I think it is appropriate that a `pip install <package>` would work on the specified version.

(Dockerfile above fails without the version specification as well)

Running in a Python 3.7 environment:
```
pip install 'pin>2.6.4'
Looking in indexes: https://pypi.org/simple, https://packagecloud.io/github/git-lfs/pypi/simple
ERROR: Could not find a version that satisfies the requirement pin>2.6.4 (from versions: 2.6.0, 2.6.1, 2.6.3, 2.6.4)
ERROR: No matching distribution found for pin>2.6.4
```

nim65s, 2022-12-09:
Ok, thanks for the notice, I don't see why we don't have a newer version of pin for python 3.7. I'll check this.

jcarpent, 2022-12-09:
@JonathanKuelz Why to not use conda?

nim65s, 2022-12-09:
because conda wouldn't allow timor to be distributed on PyPI I guess :)

jcarpent, 2022-12-09:
What is it?

nim65s, 2022-12-09:
It is @JonathanKuelz 's project: https://pypi.org/project/timor-python/

JonathanKuelz, 2022-12-09:
@nim65s exactly, :) Builds on pinocchio, and so do other tools we build or use internally - as some of them do not have other conda dependencies, it would be unfortunate if we had to either give up PyPI distributions or move to another library than pinocchio if we want to support Python 3.7

nim65s, 2022-12-09:
Yes, so it turns out support for Python 3.7 was just forgotten at some point in cmake-wheel. I'm adding it back and re-uploading the packages. It is going to take some hours I guess.

JonathanKuelz, 2022-12-09:
Great! On that occasion, thanks a lot for the open source tools (pin, hppfcl, ...) and the great support you are offering for all of them!

jcarpent, 2022-12-09:
I will close this issue as it seems to be solved, it is just needed to wait a bit.
@JonathanKuelz Feel free to reopen it if needed.

closed 2022-12-09 by jcarpent