stack-of-tasks/pinocchio #1807: Best way to build pin3 with casadi support?

ThorbenPultke, 2022-12-09:
Hi there! First of all thank you so much for your great framework and effort you put into this work!

I used the numerical part of pin while working with crocoddyl for an MPC-based approach for optimal control. Now I would like to double check integration possibilities with Acados. For this purpose I need a symbolic expression for forward kinematics and forward dynamics with Casadi, which should be doable with pin3 - at least this is what I was able to reconstruct from older issues.

I just tried to build pin3 to the best of my knowledge from source according to your instructions on your website. I indeed managed to get pinocchio running, but ran into an exception while importing pin.casadi (see screenshot). I built pin3 with Casadi turned to ON. 

Is this fault known? Did I do something wrong while building this library? Could you point me towards best practices for building this framework from source (e.g. which builder / software to use) ?

Thank you so much in advance!

Best regards
Thorben

[image]

jcarpent, 2022-12-09:
@ThorbenPultke COuld you provide a docker image to replicate your issue? ON which commit are you?

fabinsch, 2022-12-12:
Hello @ThorbenPultke, I just compiled the pinocchio3x preview branch (`1e7c3a80`) with Casadi and it is working fine for me:
```
(pin3preview_casadi) ➜  build_casadi_pinocchio3 git:(1e7c3a80) ✗ python
Python 3.9.15 | packaged by conda-forge | (main, Nov 22 2022, 15:55:03) 
[GCC 10.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
>>> from pinocchio import casadi as cpin
>>> 
```

the steps I did are the following:
- create a new conda environment `conda create --name pin3preview_casadi python=3.9` and activate it
- `conda install cmake casadi compilers=1.4.2`
- `conda install Pinocchio --only-deps`
- run cmake and make sure you are using the compiler from your conda env (GNU 10.4.0)
```
cmake .. -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_CASADI_SUPPORT=ON -DBUILD_TESTING=OFF
```
- `make install -j<number_of_jobs_in_parallel>`

Let me know if this helps you and if you encounter any problems during the build.

AunSiro, 2022-12-15:
Hello @fabinsch, thank you for the instructions. I am also trying to use the Pinoccio-Casadi combination. I am trying to follow your steps on macOS 11.4, but got some errors at the last step. The terminal output I gost for the whole process was:
part 1: https://pastebin.com/9puQuCkx 
part 2: https://pastebin.com/EvTZdw0d

Before that, I did
`git clone -b pinocchio3-preview https://github.com/stack-of-tasks/pinocchio.git`
and 
`cd pinocchio`

I don't know if I did something wrong or the problem lies with my OS. I would greatly appreciate if you could take a look. 
Thank you!

fabinsch, 2022-12-15:
Hey @AunSiro,

the problem is fixed with #1812, this is clang related. It should be fine now.

AunSiro, 2022-12-16:
Thank you, it worked!

jcarpent, 2022-12-16:
Thanks @fabinsch for the nice job.
Best,
JUstin

closed 2022-12-16 by jcarpent

johanubbink, 2023-06-21:
Hello! Thank you for creating and maintaining this library :)

I have tried to follow the instructions above but I am running into the following errors:
```
 error: invalid use of incomplete type 'struct eigenpy::expected_pytype_for_arg<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, -1, 0>, Eigen::MatrixBase<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, -1, 0> > >'
   35 | struct expected_pytype_for_arg<
      |        ^~~~~~~~~~~~~~~~~~~~~~~~
   36 |     Eigen::Matrix<Scalar, Rows, Cols, Options, MaxRows, MaxCols> >
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
and
```
/home/johan/anaconda3/envs/pin3preview_casadi/include/boost/preprocessor/iteration/detail/limits/local_256.hpp:13:9: error: 'get_pytype' is not a member of 'boost::python::converter::expected_pytype_for_arg<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, -1, 0> >'
   13 |         BOOST_PP_LOCAL_MACRO(0)
      |         ^~~~~~~~~~~~~~~~~~~~
In file included from /home/johan/Software/pinocchio/build/include/pinocchio/spatial/inertia.hpp:12,
                 from /home/johan/Software/pinocchio/build/include/pinocchio/multibody/model.hpp:13,
                 from /home/johan/Software/pinocchio/build/include/pinocchio/algorithm/contact-cholesky.hpp:8,
                 from /home/johan/Software/pinocchio/build/include/pinocchio/bindings/python/algorithm/contact-cholesky.hpp:9,
                 from /home/johan/Software/pinocchio/bindings/python/algorithm/expose-constrained-dynamics.cpp:8:
/home/johan/Software/pinocchio/build/include/pinocchio/spatial/symmetric3.hpp: In instantiation of 'pinocchio::Symmetric3Tpl<Scalar, Options>::Symmetric3Tpl(const pinocchio::Symmetric3Tpl<Scalar, Options>&) [with _Scalar = casadi::Matrix<casadi::SXElem>; int _Options = 0]':

```
the exact commands I used are:

```
conda create --name pin3preview_casadi python=3.9
conda activate pin3preview_casadi
conda install cmake
conda install -c conda-forge casadi
conda install -c conda-forge compilers=1.4.2
conda install -c conda-forge Pinocchio --only-deps
git clone git@github.com:stack-of-tasks/pinocchio.git -b pinocchio3-preview
cd pinocchio
git submodule update --init --recursive
mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_CASADI_SUPPORT=ON -DBUILD_TESTING=OFF
make install -j3
```

Do you have any suggestions what I could try? Not sure if it is relevant, but I use ubuntu 20.04, with boost version 1.78.0 and casadi 3.6.3.

Any help would be appreciated.

Best regards,
Johan

Mr-Y-B-L, 2023-10-25:
> Hello! Thank you for creating and maintaining this library :)
> 
> I have tried to follow the instructions above but I am running into the following errors:
> 
> ```
> /home/johan/anaconda3/envs/pin3preview_casadi/include/eigenpy/eigen-from-python.hpp:35:8: error: invalid use of incomplete type 'struct eigenpy::expected_pytype_for_arg<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, -1, 0>, Eigen::MatrixBase<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, -1, 0> > >'
>    35 | struct expected_pytype_for_arg<
>       |        ^~~~~~~~~~~~~~~~~~~~~~~~
>    36 |     Eigen::Matrix<Scalar, Rows, Cols, Options, MaxRows, MaxCols> >
>       |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> ```
> 
> and
> 
> ```
> /home/johan/anaconda3/envs/pin3preview_casadi/include/boost/preprocessor/iteration/detail/limits/local_256.hpp:13:9: error: 'get_pytype' is not a member of 'boost::python::converter::expected_pytype_for_arg<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, -1, 0> >'
>    13 |         BOOST_PP_LOCAL_MACRO(0)
>       |         ^~~~~~~~~~~~~~~~~~~~
> In file included from /home/johan/Software/pinocchio/build/include/pinocchio/spatial/inertia.hpp:12,
>                  from /home/johan/Software/pinocchio/build/include/pinocchio/multibody/model.hpp:13,
>                  from /home/johan/Software/pinocchio/build/include/pinocchio/algorithm/contact-cholesky.hpp:8,
>                  from /home/johan/Software/pinocchio/build/include/pinocchio/bindings/python/algorithm/contact-cholesky.hpp:9,
>                  from /home/johan/Software/pinocchio/bindings/python/algorithm/expose-constrained-dynamics.cpp:8:
> /home/johan/Software/pinocchio/build/include/pinocchio/spatial/symmetric3.hpp: In instantiation of 'pinocchio::Symmetric3Tpl<Scalar, Options>::Symmetric3Tpl(const pinocchio::Symmetric3Tpl<Scalar, Options>&) [with _Scalar = casadi::Matrix<casadi::SXElem>; int _Options = 0]':
> ```
> 
> the exact commands I used are:
> 
> ```
> conda create --name pin3preview_casadi python=3.9
> conda activate pin3preview_casadi
> conda install cmake
> conda install -c conda-forge casadi
> conda install -c conda-forge compilers=1.4.2
> conda install -c conda-forge Pinocchio --only-deps
> git clone git@github.com:stack-of-tasks/pinocchio.git -b pinocchio3-preview
> cd pinocchio
> git submodule update --init --recursive
> mkdir build
> cd build
> cmake .. -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_CASADI_SUPPORT=ON -DBUILD_TESTING=OFF
> make install -j3
> ```
> 
> Do you have any suggestions what I could try? Not sure if it is relevant, but I use ubuntu 20.04, with boost version 1.78.0 and casadi 3.6.3.
> 
> Any help would be appreciated.
> 
> Best regards, Johan

Hi friend, I am very sorry for asking a question in a closed issue. Have you resolved this issue please, I am experiencing the same problem

Mr-Y-B-L, 2023-10-25:
> Hello! Thank you for creating and maintaining this library :)
>
> I have tried to follow the instructions above but I am running into the following errors:
>
> ``` [...]

Hi friend, I am very sorry for asking a question in a closed issue. Have you resolved this issue please？ I am experiencing the same problem. Any help from you is necessary.