stack-of-tasks/pinocchio #1815: undefined symbol error for pip installed pinocchio

BolunDai0216, 2022-12-19:
I was trying to install pinocchio using `pip install pin`, and when I tried to import it, I get the following error

```console                                                                                                
Python 3.9.16 (main, Dec 19 2022, 09:26:36) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/bolun/.pyenv/versions/3.9.16/lib/python3.9/site-packages/cmeel.prefix/lib/python3.9/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ImportError: /home/bolun/.pyenv/versions/3.9.16/lib/python3.9/site-packages/cmeel.prefix/lib/python3.9/site-packages/pinocchio/pinocchio_pywrap.cpython-39-x86_64-linux-gnu.so: undefined symbol: _ZNK3hpp3fcl15ComputeDistanceclERKNS0_11Transform3fES4_RKNS0_15DistanceRequestERNS0_14DistanceResultE
```

Is this a bug in the code? I get this kind of error when something is not defined in the header file.

Would it be possible to help me take a look at what is the issue?

Thanks in advance!

fabinsch, 2022-12-19:
I just installed pinocchio via pip in a clean conda environment and it is working as expected.

nim65s, 2022-12-19:
ok, it's not helping, sorry. what about `python -c "import sys;print(sys.path)` ?

Python3.9 is not an issue, I tried in a `python:3.9` docker and it works out of the box

nim65s, 2022-12-19:
Ok, then your issue is here: you have python 3.8 stuff from robotpkg and from ROS. you need to clean your environment, either by going into a virtual env or by cleaning your PYTHONPATH

BolunDai0216, 2022-12-19:
I cleaned my `PYTHONPATH`, now the output is

```
python -c "import sys;print(sys.path)"
['', '/home/bolun/.pyenv/versions/3.9.16/lib/python3.9/site-packages/cmeel.prefix/lib/python3.9/site-packages', '/home/bolun/.pyenv/versions/3.9.16/lib/python39.zip', '/home/bolun/.pyenv/versions/3.9.16/lib/python3.9', '/home/bolun/.pyenv/versions/3.9.16/lib/python3.9/lib-dynload', '/home/bolun/.pyenv/versions/3.9.16/lib/python3.9/site-packages', '/home/bolun/.pyenv/versions/3.9.16/lib/python3.9/site-packages/cmeel.prefix/lib/python3.9/site-packages']
```

And after uninstalling and installing pinocchio using pip, I still get the error 

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/bolun/.pyenv/versions/3.9.16/lib/python3.9/site-packages/cmeel.prefix/lib/python3.9/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ImportError: /home/bolun/.pyenv/versions/3.9.16/lib/python3.9/site-packages/cmeel.prefix/lib/python3.9/site-packages/pinocchio/pinocchio_pywrap.cpython-39-x86_64-linux-gnu.so: undefined symbol: _ZNK3hpp3fcl15ComputeDistanceclERKNS0_11Transform3fES4_RKNS0_15DistanceRequestERNS0_14DistanceResultE
```

nim65s, 2022-12-19:
you also need to clean your `LD_LIBRARY_PATH` then

BolunDai0216, 2022-12-19:
That works! Thank you so much for the help!

nim65s, 2022-12-19:
Ok. Try not to mix different installs, or when you really want that please use environment isolations. You'll get into less troubles.

xinsongyan, 2023-01-06:
I had similar problems, I am using virtualenv with python 3.8 in ubuntu 20.04. 
With a new virtualenv, I installed pinocchio with` pip install pin`, 
when importing pinocchio I got a similar import error: 
```
Traceback (most recent call last):
  File "/home/xxx/PycharmProjects/pythonProject3/main.py", line 5, in <module>
    import pinocchio
  File "/home/xxx/PycharmProjects/pythonProject3/venv/lib/python3.8/site-packages/cmeel.prefix/lib/python3.8/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
  File "/home/xxx/PycharmProjects/pythonProject3/venv/lib/python3.8/site-packages/cmeel.prefix/lib/python3.8/site-packages/hppfcl/__init__.py", line 34, in <module>
    from .hppfcl import *
ImportError: /home/xxx/PycharmProjects/pythonProject3/venv/lib/python3.8/site-packages/cmeel.prefix/lib/python3.8/site-packages/hppfcl/hppfcl.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN5boost7archive18text_oarchive_implINS0_13text_oarchiveEE4saveERKSs

```
with `pip freeze` I got: 
```
cmeel==0.23.2
cmeel-assimp==5.2.5
cmeel-boost==1.80.0
cmeel-console-bridge==1.0.2
cmeel-octomap==1.9.8.1
cmeel-tinyxml==2.6.2
cmeel-urdfdom==3.1.0
eigenpy==2.8.1
hpp-fcl==2.2.0
numpy==1.23.4
pin==2.6.12
tomli==2.0.1
```

I went back to check my previous working environment: 
```
cmeel==0.23.2
cmeel-assimp==5.2.4
cmeel-boost==1.80.0
cmeel-console-bridge==1.0.2
cmeel-octomap==1.9.8
cmeel-tinyxml==2.6.2
cmeel-urdfdom==3.1.0
eigenpy==2.8.0
hpp-fcl==2.1.3
numpy==1.23.4
pin==2.6.12
tomli==2.0.1
```

Looks like a few libraries got different version numbers: cmeel-assimp, cmeel-octomap, eigenpy, hpp-fcl. 

Not sure if this is helpful to solve undefined symbol errors.

nim65s, 2023-01-09:
Hi @xinsongyan,

This is another issue, it would have been better to open another ticket.

Anyway, your symptoms follow a compiler change after https://github.com/stack-of-tasks/pinocchio/issues/1819, and it can be fixed by just updating everything:
```
root@b23eacacfd9c:/# python -m pip freeze
cmeel==0.23.2
cmeel-assimp==5.2.5
cmeel-boost==1.80.0.1
cmeel-console-bridge==1.0.2.1
cmeel-octomap==1.9.8.1
cmeel-tinyxml==2.6.2.1
cmeel-urdfdom==3.1.0.1
eigenpy==2.8.1
hpp-fcl==2.2.0
numpy==1.24.1
pin==2.6.12.1
tomli==2.0.1