stack-of-tasks/pinocchio #1816: Installing from source without Python

JakobThumm, 2022-12-19:
Hi,
I'm trying to install pinocchio from source without python using
```
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DBUILD_PYTHON_INTERFACE=OFF
```
This is successful with the output:
```
INSTALL_PREFIX=/usr/local -DBUILD_PYTHON_INTERFACE=OFF
-- Package version (ROS package.xml): 2.6.12
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.6.3") 
-- Performing Test R-pedantic
-- Performing Test R-pedantic - Success
-- Performing Test R-Wno-long-long
-- Performing Test R-Wno-long-long - Success
-- Performing Test R-Wall
-- Performing Test R-Wall - Success
-- Performing Test R-Wextra
-- Performing Test R-Wextra - Success
-- Performing Test R-Wcast-align
-- Performing Test R-Wcast-align - Success
-- Performing Test R-Wcast-qual
-- Performing Test R-Wcast-qual - Success
-- Performing Test R-Wformat
-- Performing Test R-Wformat - Success
-- Performing Test R-Wwrite-strings
-- Performing Test R-Wwrite-strings - Success
-- Performing Test R-Wconversion
-- Performing Test R-Wconversion - Success
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen dot 
-- Since urdfdom >= 1.0.0, the default C++ standard is C++11. The project is then compiled with C++11 standard.
-- Found Boost: code/boost_1_78_0 (found version "1.78.0") found components: filesystem serialization system 
-- Pinocchio won't be compiled with its Python bindings. If you want to enable this feature, please set the option BUILD_PYTHON_INTERFACE to ON.
-- Found Boost: code/boost_1_78_0 (found version "1.78.0") found components: unit_test_framework 
-- Doxygen rendering: using MathJax backend
-- Configuring done
-- Generating done
-- Build files have been written to: code/pinocchio/build
```
But the build fails with
```
code/pinocchio/build/include/pinocchio/utils/cast.hpp:8:10: fatal error: Eigen/Core: No such file or directory
    8 | #include <Eigen/Core>
      |          ^~~~~~~~~~~~

```
Apparently, Eigen is not correctly linked. Is that a known issue when building without Python?

nim65s, 2022-12-19:
Hi,
Are you sure you have the latest version of the sources, without changes ? Can you show use what do you have on `git show -q` and `git status` ?

JakobThumm, 2022-12-19:
Hi, I pulled latest master.
```
git pull
Already up to date.

git status
On branch master
Your branch is up to date with 'origin/master'.
nothing to commit, working tree clean
```

nim65s, 2022-12-19:
your configuration logs show no trace of searching for eigen3, while the latest master has a requirement for it in
https://github.com/stack-of-tasks/pinocchio/blob/c0e7066c86018516abd08332966bc592fb68b0fb/CMakeLists.txt#L111

Don't you have this line ?

`git show -q` shows this `c0e7066c86018516abd08332966bc592fb68b0fb` commit ? You might be on a wrong remote.
Otherwise, do you have correctly set up the git submodule, either from a `--recursive` in your `git clone` or with `git submodule update --init --recursive` ?

JakobThumm, 2022-12-19:
Hi,
I'm using Eigen 3.4 for some other packages.
So, I'm using this version: https://gitlab.com/libeigen/eigen/-/archive/3.4.0/eigen-3.4.0.zip
Do you know how I can tell CMake to use the custom Eigen path?
I set the Eigen3 Include Dir already:
```
echo $EIGEN3_INCLUDE_DIR
/usr/include/eigen3/eigen-3.4.0
```

nim65s, 2022-12-19:
we can check if cmake correctly found your eigen 3.4 by looking into its cache: `grep -i eigen3 CMakeCache.txt`.

JakobThumm, 2022-12-19:
```
grep -i eigen3 CMakeCache.txt
//The directory containing a CMake configuration file for Eigen3.
Eigen3_DIR:PATH=/usr/lib/cmake/eigen3
_PACKAGE_CONFIG_DEPENDENCIES_FIND_DEPENDENCY:INTERNAL=SET(Boost_USE_STATIC_LIBS OFF);SET(Boost_USE_MULTITHREADED ON);SET(Boost_NO_BOOST_CMAKE ON);find_dependency(Eigen3 REQUIRED);find_dependency(urdfdom_headers REQUIRED);find_dependency(urdfdom REQUIRED);find_dependency(Boost REQUIRED COMPONENTS filesystem serialization system)
_PACKAGE_CONFIG_DEPENDENCIES_FIND_PACKAGE:INTERNAL=SET(Boost_USE_STATIC_LIBS OFF);SET(Boost_USE_MULTITHREADED ON);SET(Boost_NO_BOOST_CMAKE ON);find_package(Eigen3 REQUIRED);find_package(urdfdom_headers REQUIRED);find_package(urdfdom REQUIRED);find_package(Boost REQUIRED COMPONENTS filesystem serialization system)
_PACKAGE_CONFIG_DEPENDENCIES_PROJECTS:INTERNAL=Eigen3;urdfdom_headers;urdfdom;Boost
```

nim65s, 2022-12-19:
Thanks for those debug. I'll try to reproduce with a custom eigen.

JakobThumm, 2022-12-19:
Thank you for your help!

JakobThumm, 2022-12-21:
Hi,
I forgot to set the `CMAKE_PREFIX_PATH`.
Now it works as expected :+1: 
Thank you so much!

closed 2022-12-21 by JakobThumm