stack-of-tasks/pinocchio #1817: Open Source Code From "Differentiable Collision Detection: a Randomized Smoothing Approach"?

alberthli, 2022-12-19:
[This recent paper](Differentiable Collision Detection: a Randomized Smoothing Approach) states that the code for differentiable collision detection will be released within the Pinocchio framework. Has this been done already? After a quick search through this repo and the HPP-FCL repo, I couldn't find the implementation. Thank you!

jcarpent, 2022-12-20:
We are currently working on finishing pinocchio 3, which should also include these contact derivatives.

jcarpent, 2022-12-20:
@alberthli Which applications do you have in mind of these derivatives?

alberthli, 2022-12-20:
I see - is there a timeline either for when the implementation of the collision detection derivatives will be rolled into the preview branch or for when pinocchio 3's release is planned?

I was hoping to use the derivatives for robot configuration adjustment based on collisions during planning. In particular, there have been a couple of recent papers on differentiable collision detection, and only pinocchio has Python bindings, which is a necessity to be integrated into my stack.

jcarpent, 2022-12-20:
> In particular, there have been a couple of recent papers on differentiable collision detection, and only pinocchio has Python bindings, which is a necessity to be integrated into my stack.

From a scientific perspective, the different approaches do not provide the same information.
I don't know precisely which papers you refer to. Still, in our case, we provide derivatives of witness points, which can be used for many usages like distance variation between geometries or contact point variations for simulation.

jcarpent, 2022-12-20:
> I see - is there a timeline either for when the implementation of the collision detection derivatives will be rolled into the preview branch or for when pinocchio 3's release is planned?

Hard to say: the quickest as we can do, but maintaining large software like Pinocchio and hpp-fcl while also doing research is very demanding ;) Hope you will understand.

alberthli, 2022-12-20:
> From a scientific perspective, the different approaches do not provide the same information.
I don't know precisely which papers you refer to. Still, in our case, we provide derivatives of witness points, which can be used for many usages like distance variation between geometries or contact point variations for simulation.

Yes, the approaches I've seen all have their differences. Aside from the question of bindings, the witness point approach presented by your collaborators and you is the closest to what I had in mind when thinking about this problem, and would be very useful for my application.

> Hard to say: the quickest as we can do, but maintaining large software like Pinocchio and hpp-fcl while also doing research is very demanding ;) Hope you will understand.

Of course! Thanks for everyone's hard work. It's a great service to the community and I will be watching with interest.