stack-of-tasks/pinocchio #1818: how to add pinocchio as library in cmake

LeoMaglanoc, 2022-12-21:
Hello,
I have to include pinocchio in a c++ project via cmake, but it fails currently.

I've installed pinocchio c++ without the python bindings via

```
sudo apt install -qqy robotpkg-pinocchio
```

I've tried out the simple examples from the documentation https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/index.html#OverviewComplex

If I compile it like this with the source code from the simplest example, it works:

```
g++ -std=c++11 overview-simple.cpp -o overview-simple $(pkg-config --cflags --libs pinocchio)  
```

For including Pinocchio in a separate c++ project, I've tried the following things:

1. finding it as a package like this:

```
find_package(pinocchio 2.6.10)
```

but it fails with this message:
```
By not providing "FindPinocchio.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Pinocchio", but CMake did not find one.

  Could not find a package configuration file provided by "Pinocchio"
  (requested version 2.6.10) with any of the following names:

    PinocchioConfig.cmake
    pinocchio-config.cmake

  Add the installation prefix of "Pinocchio" to CMAKE_PREFIX_PATH or set
  "Pinocchio_DIR" to a directory containing one of the above files.  If
  "Pinocchio" provides a separate development package or SDK, be sure it has
  been installed.
```

Can I somehow specify the path of the installed pinocchio library in the cmake file?

2. adding the compile options/flags just like in the example from the documentation

```
target_compile_options(${PROJECT_NAME} PUBLIC $(pkg-config --cflags --libs pinocchio))
```

but that also throws an error message. If it helps ```pkg-config --cflags --libs pinocchio```, gives the following:

```
-DPINOCCHIO_WITH_URDFDOM -DPINOCCHIO_WITH_HPP_FCL -DHPP_FCL_HAS_OCTOMAP -DHPP_FCL_HAVE_OCTOMAP -DFCL_HAVE_OCTOMAP -DOCTOMAP_MAJOR_VERSION=1 -DOCTOMAP_MINOR_VERSION=9 -DOCTOMAP_PATCH_VERSION=7 -I/opt/openrobots/lib/pkgconfig/../../include -I/opt/openrobots/include -I/usr/local/include/eigen3 -I/usr/lib/x86_64-linux-gnu/pkgconfig/../../../include -L/opt/openrobots/lib/pkgconfig/../../lib -L/opt/openrobots/lib -L/usr/lib/x86_64-linux-gnu/pkgconfig/../../../lib/x86_64-linux-gnu -Wl,-rpath,/opt/openrobots/lib/pkgconfig/../../lib -lpinocchio -Wl,-rpath,/usr/lib/x86_64-linux-gnu -lboost_filesystem -lboost_serialization -lboost_system -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -lconsole_bridge -Wl,-rpath,/opt/openrobots/lib -lhpp-fcl -loctomap -loctomath
```

Thank you very much!

olivier-stasse, 2022-12-21:
Dear @LeoMaglanoc thank you for your interest in pinocchio.
The answer is in the message. You can try:
`
export CMAKE_PREFIX_PATH=/opt/openrobots/:$CMAKE_PREFIX_PATH
`
Hope it helps.

LeoMaglanoc, 2022-12-21:
Thanks,

I've set the prefix path via export already. If I use the console for compiling with the command ```cmake ..  -D CMAKE_PREFIX_PATH=/opt/openrobots```, it works. Is there a way to set the cmake prefix path inside the cmake file somehow? I should add that normally I'm using my IDE Clion to compile and run the project so maybe it's because of the IDE?

nim65s, 2022-12-21:
Hi @LeoMaglanoc,
Did you give this `CMAKE_PREFIX_PATH` to Clion ? Here is a reference for that: https://stackoverflow.com/questions/54444846/equivalent-of-cmake-prefix-path-in-clion : you can add `-DCMAKE_PREFIX_PATH=/opt/openrobots` in this `CMake options` field.

LeoMaglanoc, 2022-12-21:
Yes thanks! The cmake building works now

```
-- pinocchio FOUND. pinocchio at /opt/openrobots/lib/libpinocchio.so
```

but I still can't include Pinocchio in the source file

```
fatal error: pinocchio/parsers/urdf.hpp: No such file or directory
    8 | #include "pinocchio/parsers/urdf.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

I've included Pinocchio in cmake via target_link_libraries. Is there anything else I have to do?

```
find_package(pinocchio)
target_link_libraries(${PROJECT_NAME} pinocchio) 
add_executable(debug src/debug.cc)
target_link_libraries(debug ${PROJECT_NAME} pinocchio)
```

jcarpent, 2022-12-21:
please have a look here https://github.com/stack-of-tasks/pinocchio/blob/master/unittest/packaging/cmake/CMakeLists.txt

LeoMaglanoc, 2022-12-21:
Tha

closed 2022-12-21 by LeoMaglanoc

Smit-tay, 2024-11-06:
In fact this issue is addressed by following the Pinocchio installation instructions on the stack-of-tasks.github.io/Pinocchio/download.html page

Explicitly the "Configure environment variables" section