stack-of-tasks/pinocchio #1819: pip wheel for ubuntu 18.04

Jiayuan-Gu, 2022-12-24:
I find that the latest pinocchio (>=2.10.0) can be installed through pip only for ubuntu >=20.04, because the base docker image used to build is `manylinux_2_28` but ubuntu 18.04 has glibc=2.27. Although ubuntu 18.04's EOL is close (Apr. 2023), certain applications may only support ubuntu 18.04 and pip (e.g., google colab). Could the developers also build manylinux2014 wheels?

BTW, I have an open-source project using `pinocchio` to support pythonic motion planning (https://github.com/Jiayuan-Gu/pymp), and it will be nice Pinocchio would like to include it in the tutorial since motion planning is missing now in the tutorial.

jcarpent, 2022-12-24:
@Jiayuan-Gu If you want, you can, in the meantime, rely on conda to install pinocchio on google colab.
Of course, you are very welcome to add new examples of references to your lib related to motion planning.
On Pinocchio's side, some developers have been using Pinocchio for a while within the HPP framework (https://github.com/humanoid-path-planner), which is dedicated to motion planning.

For the rest of the issue, I let @nim65s answer the question.

traversaro, 2022-12-24:
> @Jiayuan-Gu If you want, you can, in the meantime, rely on conda to install pinocchio on google colab.

For that you may find useful https://github.com/conda-incubator/condacolab that simplifies the installation of conda packages on conda.

nim65s, 2022-12-24:
@Jiayuan-Gu : moving back to manylinux2014 shouldn't be an issue. I'll probably handle this within a week. If you need shorter delays, I could focus on this part and get it done in a day or 2.

Jiayuan-Gu, 2022-12-24:
@jcarpent @traversaro  @nim65s Thank you for the timely reply.

- @jcarpent Thank you for pointing out `humanoid-path-planner`. When I get some time, I may try to add some examples [here](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_b-examples_m-rrt.html).
Besides, the original purpose of my pythonic motion-planning library is to reduce the overhead of compilation (at the cost of runtime efficiency) when we just want to experiment with something new (e.g., a common case in our projects is to find a collision-free starting configuration around the user-provided one). I would like to thank you since `pinocchio` and `hpp-fcl` make life easier.
- @traversaro I tried `condacolab` following your suggestion, and it worked well. Thank you.
- @nim65s One potential issue of moving back to manylinux2014 is the version of Boost. We encountered some difficulties in making a pip wheel of OMPL before due to the Boost version. Take your time, and I am not in a hurry.

nim65s, 2023-01-02:
Hi @Jiayuan-Gu : this should be done. Can you confirm and close this issue ?

Jiayuan-Gu, 2023-01-02:
Hi @nim65s I have tested the installation in Google Colab and it works. Thank you.

closed 2023-01-02 by Jiayuan-Gu