stack-of-tasks/pinocchio #1823: Rotor inertia ≠ mass

matheecs, 2023-01-04:
In branch `pinocchio3-preview`, `rotor_inertia` is equal to `mass`, such as

https://github.com/stack-of-tasks/pinocchio/blob/0d53f26acdf9a0b5740b88145f624b11d97f40f9/src/parsers/srdf.hxx#L173

Is it a bug? In my view, `rotor_inertia` is always a matrix, `mass` is a scalar.

jcarpent, 2023-01-04:
`rotor_inertia` corresponds to the inertia of the electrical motor, which URDF does not model.
This is a scalar quantity, as we currently only consider rotor inertia for single dof joints.
For instance, here is the way it has been set up for the TALOS robot:
https://github.com/pal-robotics/talos_robot/blob/cff2ba0e26f98c9f8acf1d6f46d2d8ff962bb2fc/talos_description/srdf/talos.srdf#L137-L168

I hope this is helping you in your understanding.

jcarpent, 2023-01-04:
The mass of a joint is the mass associated with the body supported by the joint.
The robot inertia is only considered in RNEA, CRBA and ABA algorithms to account for inertia effects due to motors.

matheecs, 2023-01-04:
I still confused about the definition of `rotor_inertia` and `mass`. Imagine the rotor as a thick-walled cylindrical tube with open ends, of inner radius $r_1$, outer radius $r_2$, length $h$ and mass $m$, like 

[image]

We have

$Rotor \\ Inertia = \frac{1}{2}\times m\times (r_2^2+r_1^2)$

So, rotor inertia ≠ mass.

jcarpent, 2023-01-04:
Yes, you're right about the naming. From a physical perspective, the mass and rotor inertia are two different quantities.

Unfortunately, we are not at the origin of this convention. But in the case of the SRDF format, the `mass` means the inertia value along the axis of rotation/translation.
The same weird namings also appear in URDF and SDF format, unfortunately.

matheecs, 2023-01-05:
Thanks for the help, It all makes sense now.

closed 2023-01-05 by matheecs