stack-of-tasks/pinocchio #1824: ‘computeDistances’ is not a member of ‘pinocchio’

lpredoomy, 2023-01-05:
I encountered an error when compiling OCS2
[image]

I changed CMakeLists.txt of pinocchio
OPTION(BUILD_WITH_COLLISION_SUPPORT "Build the library with the collision support (required HPP-FCL)" ON)

Still haven't solved the problem

jcarpent, 2023-01-05:
I think the PR should be opended at the OCS 2 level and you should follow their instruction for installing Pinocchio.
@wxmerkt Does Pinocchio installed from ROS is handling the support of hpp-fcl?

jcarpent, 2023-01-05:
@lpredoomy a bit of politeness when introducing your issue is more than welcome !!!

closed 2023-01-05 by jcarpent

wxmerkt, 2023-01-05:
Pinocchio from ROS has HPP-FCL support. However, based on the other issue I believe this is source-compiled Pinocchio and in this case the HPP-FCL was not activated when configuring the source build - or activated after the initial configure. In this case, one needs to either `catkin clean` or `catkin build pinocchio --force-cmake`. Using the Pinocchio binaries from ROS would fix the above issue as well

jcarpent, 2023-01-05:
Thanks a lot @wxmerkt for your guidance.

lpredoomy, 2023-01-05:
@wxmerkt Thank you for your answer！ 
1.I recompiled and installed pinocchio and made the following changes, but the same error still occurred.
[image]
2.I have compiled and installed pinocchio from the source code. Will installing pinocchio from binaries from ROS cause conflicts?
3.I don't quite understand where you want to execute the catkin clean or catkin build pinocchio --force-cmake command
Thank you very much for your help！

wxmerkt, 2023-01-05:
It's a bit unclear what went wrong with your installation. If in doubt, delete the build directory, reconfigure, rebuild, reinstall. If you have multiple installations, it might be that the wrong one is picked up. Make sure to grep the build directory (in this case of OCS) and check that the Pinocchio you want is picked up

For (3) - I assumed you had HPP-FCL and Pinocchio in your catkin workspace rather than separately built/installed via CMake. In that case you can ignore this

lpredoomy, 2023-01-06:
@wxmerkt  I temporarily solved this error by putting pinocchio and hpp-fcl in the ocs2 workspace as described [here](https://leggedrobotics.github.io/ocs2/installation.html#build-this-documentation).
I also have the following puzzles.
1.What is the difference between putting pinocchio as an ros package in the same workspace of ocs2 and compiling pinocchio from source code？I have changed the "BUILD_WITH_COLLISION_SUPPORT" option to ON when compiling the pinocchio source code, but this does not solve the error.
2.According to my understanding, ocs2 can call the pinocchio compiled from the source code through the cmake command.When I install pinocchio in `/usr/local/include`, there is also a pinocchio package in the ocs2 workspace. When compiling OCS2, what is the calling priority of the two?
Thank you very much for your help！

wxmerkt, 2023-01-06:
Great to hear you have resolved the problem.

1. Pinocchio will be built as a cmake package in either case - whether it is built manually or via catkin-tools. catkin-tools sets the build/install directories per the catkin workspace and allows for resolving the dependency tree. Otherwise, it also sets up the environment variables (CMAKE_PREFIX_PATH, LD_LIBRARY_PATH, etc.) automatically. So if you do all these steps manually and correctly, there should be no difference to downstream packages.
2. The preference depends on the order in your environment variables - by default the one in the catkin workspace should take precedence

jennareher, 2023-02-10:
Hello pinocchio team, I believe I am having an issue related to this ticket (in that I seem to be unable to use any of the functionality behind `PINOCCHIO_WITH_HPP_FCL`) which I haven't been able to resolve with the suggestions above. Let me know if I'm asking in the wrong place or need to be opening a separate issue, I didn't want to duplicate tickets. 

Specifically, I am wanting to use a source installation of pinocchio (our project is not using ROS) and am exploring using the `hpp-fcl` integration for collision checking. I have been able to successfully build & install pinocchio using the `BUILD_WITH_COLLISION_SUPPORT` flag, but am encountering a similar build error to the original post. 

Here is my cmake command and the output for how I am building pinocchio:
```
$ cmake -GNinja -DCMAKE_BUILD_TYPE=Release -DBUILD_PYTHON_INTERFACE=OFF -DBUILD_TESTING=OFF -DBUILD_WITH_COLLISION_SUPPORT=ON -Bpinocchio-build -Spinocchio
-- Package version (ROS package.xml): 2.6.9
-- The CXX compiler identification is Clang 14.0.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/clang++-14 - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Performing Test R-pedantic
-- Performing Test R-pedantic - Success
-- Performing Test R-Wno-long-long
-- Performing Test R-Wno-long-long - Success
-- Performing Test R-Wall
-- Performing Test R-Wall - Success
-- Performing Test R-Wextra
-- Performing Test R-Wextra - Success
-- Performing Test R-Wcast-align
-- Performing Test R-Wcast-align - Success
-- Performing Test R-Wcast-qual
-- Performing Test R-Wcast-qual - Success
-- Performing Test R-Wformat
-- Performing Test R-Wformat - Success
-- Performing Test R-Wwrite-strings
-- Performing Test R-Wwrite-strings - Success
-- Performing Test R-Wconversion
-- Performing Test R-Wconversion - Success
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
CMake Warning at cmake/doxygen.cmake:494 (message):
  Failed to find Doxygen, documentation will not be generated.
Call Stack (most recent call first):
  cmake/post-project.cmake:59 (_setup_project_documentation)
  CMakeLists.txt:35 (PROJECT)

-- Since urdfdom >= 1.0.0, the default C++ standard is C++11. The project is then compiled with C++11 standard.
-- Found Boost: /usr/include (found version "1.74.0") found components: filesystem serialization system 
-- Pinocchio won't be compiled with its Python bindings. If you want to enable this feature, please set the option BUILD_PYTHON_INTERFACE to ON.
-- hpp-fcl FOUND. hpp-fcl at /usr/local/lib/libhpp-fcl.so
-- Found Boost: /usr/include (found version "1.74.0") found components: chrono serialization 
-- Default C++ standard: 201402
-- Minimal C++ standard upgraded to 11
-- C++ standard sufficient: Minimal required 11, currently defined: 11
-- Found Boost: /usr/include (found version "1.74.0") found components: unit_test_framework 
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/pinocchio-build
```
which I then install (`cmake --build pinocchio-build && sudo cmake --install pinocchio-build`) successfully. Prior to this I had also installed `hpp-fcl` from source.

I then have some code which calls `pinocchio::computeCollisions(collision_geometry_, collision_data_);` that fails to build:
```
error: no member named 'computeCollisions' in namespace 'pinocchio'
  pinocchio::computeCollisions(collision_geometry_, collision_data_);
  ~~~~~~~~~~~^
1 error generated.
```
Just above this line I was able to call:
```
pinocchio::updateGeometryPlacements(robot_model_, robot_data_, collision_geometry_, collision_data_);
```
so I do have access to other geometry functions inside of `pinocchio/algorithm/geometry.hpp`. I had essentially tried replicating the C++ example [here](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_b-examples_e-collisions.html).