stack-of-tasks/pinocchio #1825: I encountered an error related to pinocchio when compiling OCS2

lpredoomy, 2023-01-05:
hello everyone
I compiled and installed pinocchio from the source code.Then, OCS2 is also installed according to the official tutorial.
An error as shown in the figure is encountered in the process of catbin build.
[image]
I follow the error prompt to view the pinocchio source code.
I think the following documents are closely related to this problem.  "/usr/local/include/pinocchio/multibody/fcl.hpp"
[image]

According to the analysis of this code, this problem is related to PINOCCHIO_WITH_HPP_FCL

So I changed the CMakeLists.txt file when installing pinocchio. Set the BUILD_WITH_COLLISION_SUPPORT option to ON
[image]
[image]
[image]
I still haven't solved the problem after the above operations. 
How can I solve this problem? Thank you

cmastalli, 2023-01-05:
Hi @lpredoomy,

We understand your wish to sort out your issue as soon as possible. However, I believe you're interacting in a improper way that might be perceived as impolite.

Creating another issue with the same topic because the other one gets close is not appropriate. First, this duplicates information, which makes everyone life harder (maintainers, developers or users). Second, the former issue seems to be not related with Pinocchio. If you believe the opposite, then I would suggest you to explain your reasons and accepts if Pinocchio developers wish to tackle.

As an open-source developer, I understand the value of this issues. This help us improve our software. But, we have limited resources and priorities. We have often other responsibilities such as supervising students, writing publications and grants, etc.

We appreciate your interest in joining our community and expect to understand the rules of the game.

lpredoomy, 2023-01-05:
I'm sorry, this is my first time to ask questions, and I'm not familiar with the process.I saw that another question I asked was closed, thought it was necessary to ask again, and added details.

lpredoomy, 2023-01-05:
@cmastalli  I'm sorry, this is my first time to ask questions, and I'm not familiar with the process.I saw that another question I asked was closed, thought it was necessary to ask again, and added details.

wxmerkt, 2023-01-05:
I provided pointers in the other issue (#1824) on how to fix it;

**Option A:** Use the binaries via the ROS buildfarm - `sudo apt install ros-noetic-pinocchio` - and clean and remove Pinocchio from your workspace (or since it appears to be installed to `/usr/local` - `make uninstall` in your custom installation

**Option B:** `catkin clean`, then make sure you have `hpp-fcl` cloned in your workspace and the CMakeLists option set to ON, then recompile

lpredoomy, 2023-01-05:
@wxmerkt I tried option A, but it didn't solve the problem.For option B, I compile and install both pinocchio and hpp-fcl from source code with the `cmake` and `make install` commands. I don't know how to use the `**catkin clean**`

Is it not feasible to install pinocchio and hpp-fcl by compiling source code?

jcarpent, 2023-01-05:
@lpredoomy Could you output the results of your cmake command?

jcarpent, 2023-01-05:
@lpredoomy If your plan is to use Pinocchio for OSC-2, please follow the guideline [here](https://leggedrobotics.github.io/ocs2/installation.html).

closed 2023-01-06 by wxmerkt