stack-of-tasks/pinocchio #1835: Definition of Joint Jacobian

LeoMaglanoc, 2023-01-16:
Hello,

there is a mismatch between my understanding of what a joint jacobian is and what actually gets computed as a joint jacobian in Pinocchio. My goal is to compute the linear and angular velocity at each joint (and also for the end-effector). Is that something I can achieve with the joint jacobian? And if not, are there other Pinocchio functions I could use to do that?

E.g., given the following (pseudo-)code:

```
Eigen::Vector<double, 6> q; 
q << 1.0, 2.0, 3.0, 4.0, 5.0, 6.0;
Eigen::Vector<double, 6> q_dot; 
q_dot << 1.0, 2.0, 3.0, 4.0, 5.0, 6.0;
pinocchio::forwardKinematics(model, data, q);
pinocchio::computeJointJacobians(model, data, q);
pinocchio::getJointJacobian(model, data, joint_id, pinocchio::WORLD, jacobian);
return jacobian * q_dot;
```

Would this return the vector consisting of the linear and angular velocity at joint joint_id?

stephane-caron, 2023-01-16:
AFAIU yes. I guess you've already checked the documentation for [getJointJacobian](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#abadfcd21d13fa17865a189e4b47acca9). Here is a proposal (to be validated by project maintainers!) to recap a bit more details about this Jacobian: https://github.com/stack-of-tasks/pinocchio/pull/1836/files

jcarpent, 2023-01-16:
@LeoMaglanoc You should change the frame of expression (`pinocchio::LOCAL` for getting the joint Jacobian expressed in the LOCAL frame of the joint or `pinocchio::LOCAL_WORLD_ALIGNED` to compute the Jacobian of the joint but expressed in the coordinate system of the WORLD).

LeoMaglanoc, 2023-01-17:
@jcarpent Thanks, `pinocchio::LOCAL_WORLD_ALIGNED` is probably what I'm looking for. Just for my understanding though, what exactly do you get if you use `pinocchio::WORLD`?

jcarpent, 2023-01-17:
It will be the velocity of the Jacobian of the point coincident with the origin of the WORLD and expressed in the WORLD coordinate system. I do think this cheat sheet will be helpful: https://github.com/stack-of-tasks/pinocchio/blob/master/doc/pinocchio_cheat_sheet.pdf

jcarpent, 2023-01-17:
I will close this issue and transfer this issue as Q&A in the discussions.

closed 2023-01-17 by jcarpent