stack-of-tasks/pinocchio #1839: module 'pinocchio.pinocchio_pywrap_default' has no attribute 'buildModelFromSdf'

jameswzhu, 2023-01-20:
Greetings,

I am trying to load a SDF model using the Pinocchio 3 Python bindings:
`model, collision_model, visual_model = pin.buildModelsFromSdf(sdf_model_path)`

However, I receive the following exception:
`Exception has occurred: AttributeError
module 'pinocchio.pinocchio_pywrap_default' has no attribute 'buildModelFromSdf'
  File "/usr/local/lib/python3/dist-packages/pinocchio/shortcuts.py", line 76, in buildModelsFromSdf
    model, constraint_models = pin.buildModelFromSdf(filename)
  File "/home/james/catkin_ws/src/quad-sdk/ilqr/microtaur.py", line 29, in <module>
    model, collision_model, visual_model = pin.buildModelsFromSdf(sdf_model_path)
AttributeError: module 'pinocchio.pinocchio_pywrap_default' has no attribute 'buildModelFromSdf'`

Did I miss installing a dependency or something similar? Help would be immensely appreciated. Thank you.

fabinsch, 2023-01-21:
Hey @jameswzhu ,
It seems you're not installing Pinocchio with SDF support . Make sure to be in the  pinocchio3-preview branch and enable the cmake option BUILD_WITH_SDF_SUPPORT. I hope this helps

jameswzhu, 2023-01-22:
Thank you for the help! Yes I needed to turn on the SDF support. But now, I am seeing this error when I attempt to load a model:

`Exception has occurred: MemoryError
exception: no description
  File "/usr/local/lib/python3/dist-packages/pinocchio/shortcuts.py", line 78, in buildModelsFromSdf
    model, constraint_models = pin.buildModelFromSdf(filename, root_joint)
  File "/home/james/catkin_ws/src/quad-sdk/ilqr/microtaur.py", line 29, in <module>
    model, collision_model, visual_model = pin.buildModelsFromSdf(sdf_model_path, root_joint=pin.JointModelFreeFlyer())
MemoryError: `

Any thoughts on this? Though this may be something with my personal setup that I will try to debug. Thanks!

jcarpent, 2023-01-22:
Hi @jameswzhu,

Could you share your robot model?
Best,

Justin

jameswzhu, 2023-01-22:
Sure, the robot model I want to load is for the Ghost Robotics Spirit. But also for debugging, I've have tried just a simple box model. But the error occurs for both models. I've attached the files below, hope this helps!
[models.zip](https://github.com/stack-of-tasks/pinocchio/files/10475001/models.zip)

jcarpent, 2023-01-23:
@jameswzhu from your previous message, it seems you are mixing two strategies to install Pinocchio. One uses catkin, and the other one uses dist-package. Could you start from a fresh environment (e.g., docker or conda)?

jameswzhu, 2023-01-23:
Thanks for the recommendation! I will try that and follow up with you on if that solved the issue.

jameswzhu, 2023-02-10:
Sorry for the delay in response, but I think I'm still struggling with the installation. When I install pinocchio from source, should it not be found in the python3/dist-packages folder?

jameswzhu, 2023-02-12:
I wanted to follow up with some more information. Does it matter which `libsdformat` release I  am using? I currently have `libsdformat6`. on Ubuntu 20.04, with C++14.

g1y5x3, 2024-12-02:
also encountered this issue with using the apt-get instruction from https://stack-of-tasks.github.io/pinocchio/download.html

nim65s, 2024-12-02:
SDF support is not available on robotpkg for now. I'll try to add it at some point, but not before next year.

g1y5x3, 2024-12-02:
I see. So I will try to install the latest release from source.

jorisv, 2024-12-03:
@g1y5x3 conda-forge is built with SDF support since Pinocchio v3.3.0.

jorisv, 2025-11-27:
Closed since conda-forge release ship SDF.

closed 2025-11-27 by jorisv