stack-of-tasks/pinocchio #1844: PyPI package seems broken due to cmeel-boost release

JonathanKuelz, 2023-01-25:
Good morning,

since yesterday evening, we are having problems with fresh PyPi-based installations of pinocchio again. The issue seems to be related to the latest update of cmeel-boost on PyPI: [](https://pypi.org/project/cmeel-boost/1.80.0.1/#history)

Installing cmeel-boost 1.80.0.1 fixes this, but the dependency in the pin package seems to be not fixed.

Dockerfile to reproduce:
```
FROM python:3.10

RUN pip install --no-cache-dir pin>2.6.12
RUN pip install --no-cache-dir cmeel-boost==1.80.0.1

CMD [ "python" ]
```

then within the docker:
```
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/site-packages/cmeel.prefix/lib/python3.10/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ImportError: libboost_python310.so.1.80.0: cannot open shared object file: No such file or directory

```

jcarpent, 2023-01-30:
@nim65s Do you have any insight on this?

nim65s, 2023-01-31:
Sorry, I missed this issue. This was transient following a boost upgrade to 1.81, and has been fixed by [v2.6.14.c1](https://github.com/cmake-wheel/pinocchio/actions/runs/4001869548)

xinsongyan, 2023-02-07:
Hi, I think I got the same issue with `pip install pin`: 

> Collecting pin
>   Using cached pin-2.6.14-1-cp38-cp38-manylinux_2_17_x86_64.whl (6.1 MB)
> Requirement already satisfied: cmeel in ./venv/lib/python3.8/site-packages (from pin) (0.31.0)
> Requirement already satisfied: cmeel-tinyxml in ./venv/lib/python3.8/site-packages (from pin) (2.6.2.1)
> Requirement already satisfied: cmeel-console-bridge in ./venv/lib/python3.8/site-packages (from pin) (1.0.2.1)
> Requirement already satisfied: cmeel-urdfdom in ./venv/lib/python3.8/site-packages (from pin) (3.1.0.1)
> Requirement already satisfied: hpp-fcl in ./venv/lib/python3.8/site-packages (from pin) (2.2.0)
> Requirement already satisfied: tomli<3.0.0,>=2.0.1 in ./venv/lib/python3.8/site-packages (from cmeel->pin) (2.0.1)
> Requirement already satisfied: cmeel-assimp in ./venv/lib/python3.8/site-packages (from hpp-fcl->pin) (5.2.5)
> Requirement already satisfied: cmeel-octomap in ./venv/lib/python3.8/site-packages (from hpp-fcl->pin) (1.9.8.1)
> Requirement already satisfied: eigenpy in ./venv/lib/python3.8/site-packages (from hpp-fcl->pin) (2.9.2)
> Requirement already satisfied: cmeel-boost in ./venv/lib/python3.8/site-packages (from eigenpy->hpp-fcl->pin) (1.81.0)
> Requirement already satisfied: numpy in ./venv/lib/python3.8/site-packages (from cmeel-boost->eigenpy->hpp-fcl->pin) (1.24.2)
> Installing collected packages: pin
> Successfully installed pin-2.6.14

when importing Pinocchio I got the import error:

> ImportError: libboost_python38.so.1.80.0: cannot open shared object file: No such file or directory

I am using Ubuntu20.04 + python3.8

jcarpent, 2023-02-07:
Could you do a `pip update`?

xinsongyan, 2023-02-07:
Do you mean `pip upgrade`? 
Currently, the pip version is 22.3.1,  with ` pip install pip --upgrade
`, it is upgraded to 23.0
but reinstalling pinocchio, the error persists.

jcarpent, 2023-02-07:
Which os do you use?

xinsongyan, 2023-02-07:
I am using Ubuntu 20.04.5 LTS. 

> Collecting pin
>   Using cached pin-2.6.14-1-cp38-cp38-manylinux_2_17_x86_64.whl (6.1 MB)
> Requirement already satisfied: cmeel-tinyxml in ./venv/lib/python3.8/site-packages (from pin) (2.6.2.1)
> Requirement already satisfied: cmeel-urdfdom in ./venv/lib/python3.8/site-packages (from pin) (3.1.0.1)
> Requirement already satisfied: hpp-fcl in ./venv/lib/python3.8/site-packages (from pin) (2.2.0)
> Requirement already satisfied: cmeel in ./venv/lib/python3.8/site-packages (from pin) (0.31.0)
> Requirement already satisfied: cmeel-console-bridge in ./venv/lib/python3.8/site-packages (from pin) (1.0.2.1)
> Requirement already satisfied: tomli<3.0.0,>=2.0.1 in ./venv/lib/python3.8/site-packages (from cmeel->pin) (2.0.1)
> Requirement already satisfied: eigenpy in ./venv/lib/python3.8/site-packages (from hpp-fcl->pin) (2.9.2)
> Requirement already satisfied: cmeel-octomap in ./venv/lib/python3.8/site-packages (from hpp-fcl->pin) (1.9.8.1)
> Requirement already satisfied: cmeel-assimp in ./venv/lib/python3.8/site-packages (from hpp-fcl->pin) (5.2.5)
> Requirement already satisfied: **cmeel-boost** in ./venv/lib/python3.8/site-packages (from eigenpy->hpp-fcl->pin) (**1.81.0**)
> Requirement already satisfied: numpy in ./venv/lib/python3.8/site-packages (from cmeel-boost->eigenpy->hpp-fcl->pin) (1.24.2)
> Installing collected packages: pin
> Successfully installed pin-2.6.14

> ImportError: libboost_python38.so.**1.80.0**: cannot open shared object file: No such file or directory

As highlighted, the installed cmeel-boost is 1.81.0, when importing, it is looking for 1.80.0, which might be the problem.

stephane-caron, 2023-02-07:
In the meantime, fellow PyPI travelers can backtrack to the following workaround:

```console
$ pip install cmeel-boost==1.80.0 hpp-fcl==2.1.4 pin==2.6.12
```

JakobThumm, 2023-02-08:
Is there any pip version currently installable in py3.8?

jcarpent, 2023-02-08:
`pip install cmeel-boost==1.80.0 hpp-fcl==2.1.4 pin==2.6.12` seems to work.
Otherwise, use conda which is properly working.

JakobThumm, 2023-02-08:
Unfortunately, this throws an `hpp-fcl` error:
```
ImportError: /opt/conda/envs/hrgym/lib/python3.8/site-packages/cmeel.prefix/lib/python3.8/site-packages/pinocchio/../../../libhpp-fcl.so: undefined symbol: _ZNK6Assimp8IOSystem16CurrentDirectoryB5cxx11Ev
```
I recall that this issue was fixed with hpp-fcl 2.2.0 and pin 2.6.14.

nim65s, 2023-02-08:
Sorry folks, I got distracted this WE by the birth of my son while I was cleaning boost on pypi which was full due to too many trials on a boring issue between py 3.8 and apple silicon.
And it looks like eigenpy was rebuild with the wrong boost version. hpp-fcl & pin are fine with boost 1.81.0, so  this rebuild should fix your issues: https://github.com/cmake-wheel/eigenpy/actions/runs/4123796315

nim65s, 2023-02-08:
```dockerfile
FROM python:3.8

RUN python -m pip install -U pip
RUN python -m pip install pin
RUN python -c "import pinocchio"
```
works again. If it doesn't work for you, that's probably because you don't have the latest version of some packages.

closed 2023-02-08 by nim65s

JakobThumm, 2023-02-08:
Congratulations on the birth of your child! 
The fix worked like a charm :)