stack-of-tasks/pinocchio #1848: update model after loading URDF

hxrkpib, 2023-01-29:
There is a case that I want to add a payload to a quadruped robot. The payload is dynamically loaded
to body link(random mass/com/inertia).Before that, I import a model by loading URDF.
Now I have two choice:
1) update body link mass/com/inertia (treat body and payload as one link)
2) add a new link(fixed-joint) to body's child

but i don't find related API to achieve them.

hxrkpib, 2023-01-29:
if I choice 2. add a new link(fixed-joint) to body's child, the new joint/link should can be deleted.

jcarpent, 2023-01-29:
You can use `appendBodyToJoint` in the Model API

hxrkpib, 2023-01-29:
> You can use `appendBodyToJoint` in the Model API

thanks for your reply! If I use `appendBodyToJoint`, the payload body should be deleted for update the payload body inertia. 
in other words, the body should be add-delete-add... cycle like this to update payload body inertia(the payload mass/position maybe changed).
[image]
for example, when the robot is running, add a backpack to the body, and then remove it.

jcarpent, 2023-01-29:
We should add the feature to remove inertia from a given joint inertia.
In the meantime, you can mimic this effect by saving the inertia of a given joint `const pinocchio::Inertia saved_inertia = model.inertias[joint_id]` and using the method `appendBodyToJoint`. When you want to remove its effect, you need to restore the saved value as `model.inertias[joint_id] = saved_inertia`. Is it fine?

jcarpent, 2023-01-30:
@hxrkpib I will close the issue and transfer it as Q&A in the discussions.