stack-of-tasks/pinocchio #1849: Segfault on Ubuntu 18

duburcqa, 2023-01-30:
I'm getting a segfault when I'm using Pinocchio 2.6.14 in conjunction with eigenpy 2.9.0. It happens whether I compile them myself or install them using `robotpkg`. It occurs in particular when I call `Zero` or `Random` static methods on `pinocchio::Force` or `pinocchio::Motion` objects, but not `pinocchio::SE3`. I suspect it is related to memory aligment issues.

By the way, I noticed a few segfault on the CI of Pinocchio itself for Ubuntu 18, which seems to be related but I may be wrong.

It is a little bit problematic because eigenpy 2.9.0 is the only release available on Ubuntu 18, and I would like to wait until EOL in april to drop its support.

If you need more information to reproduce the bug, just tell me and I could probably take some time to write a MRE.

jcarpent, 2023-01-30:
Yes, we are aware of the issue, which also appears in the CI.
With @nim65s, we are working on it.

On Ubuntu-18, I guess you could also use an older release, no?

duburcqa, 2023-01-30:
> Yes, we are aware of the issue, which also appears in the CI.

Ok perfect !

> On Ubuntu-18, I guess you could also use an older release, no?

Currently, only eigenpy 2.9.0 is available for ubuntu 18 on robotpkg repository [here](http://robotpkg.openrobots.org/packages/debian/pub/dists/bionic/robotpkg/binary-amd64/).

nim65s, 2023-01-30:
@duburcqa we can retrieve older versions if necessary. But I think this segfault only occurs in python 3.6, wouldn't 2.7 work for you ? As this is the default version for this ubuntu, and the only one supported by ROS there, there are more tests on our side for it.

duburcqa, 2023-01-30:
> we can retrieve older versions if necessary. 

Just 2.8.1 would be great. I just need one version that is working properly :)

> But I think this segfault only occurs in python 3.6, wouldn't 2.7 work for you ?

Not really, I have many dependencies requiring Python 3.6+ including my own codebase, plus I'm not using ROS. I would rather consider dropping support of Ubuntu 18 and Python 3.6 rather than switching to Python 2.7.

jcarpent, 2023-01-30:
The issue seems to be related to Boost.Python. I'm checking it twice, but will provide a fix soon.

jcarpent, 2023-01-30:
Solved via #1851

closed 2023-01-30 by jcarpent

duburcqa, 2023-01-30:
Nice, I will check by myself tomorrow. If so, I will wait for the next release integrating the patch before starting testing on Ubuntu 18 again. Thanks !