stack-of-tasks/pinocchio #1872: Derivatives using CasADi is NOT equal to analytical derivatives for inverse dynamics

matheecs, 2023-02-20:
I use two method to calculate the $\frac{d\mathbf{tau}}{d\mathbf{q}}$ with following codes (pinocchio3)

```python
import pinocchio as pin
from pinocchio import casadi as cpin

from casadi import *
import numpy as np

np.set_printoptions(precision=2)
np.set_printoptions(suppress=True)

model = pin.buildSampleModelHumanoidRandom(False)
data = model.createData()

cmodel = cpin.Model(model)
cdata = cmodel.createData()

qmax = np.full((model.nq, 1), np.pi)
q = pin.randomConfiguration(model, -qmax, qmax)
v = np.random.rand(model.nv)
a = np.random.rand(model.nv)
tau = np.random.rand(model.nv)

cq = SX.sym("q", cmodel.nq, 1)
cv = SX.sym("v", cmodel.nv, 1)
ca = SX.sym("a", cmodel.nv, 1)
ctau = SX.sym("tau", cmodel.nv, 1)

pin.computeRNEADerivatives(model, data, q, v, a)
pin3_J = data.dtau_dq

ctau = cpin.rnea(cmodel, cdata, cq, cv, ca)
Fj = jacobian(ctau, cq)
myFunction = Function("myFunction", [cq, cv, ca], [Fj])
casadi_J = np.array(myFunction(q, v, a))

print(f"pin3_J_first_row=\n{pin3_J[0,:]}\ncasadi_J_first_row=\n{casadi_J[0,:]}")
```

But two method give two different result, like

```text
pin3_J_first_row=
[   0.      0.      0.      0.   -254.27   60.12   14.47    5.1     1.17
   -4.46   -2.86    1.95   -0.76   -4.81    0.39   -3.45    1.52    1.84
   30.52  -19.24   -4.1     0.07    0.24   -0.7     1.88    0.44  -16.21
   -7.92   -9.98   -5.07    3.32    1.64]
casadi_J_first_row=
[  0.     0.     0.   -14.68 -18.53  -8.5   14.47   5.1    1.17  -4.46
  -2.86   1.95  -0.76  -4.81   0.39  -3.45   1.52   1.84  30.52 -19.24
  -4.1    0.07   0.24  -0.7    1.88   0.44 -16.21  -7.92  -9.98  -5.07
   3.32   1.64]
```

This problem will causes some problems in the optimization process.

jcarpent, 2023-02-20:
I think the issue is coming from the fact the JointModelCompisite is not yet supported for the derivatives and will need some additional work to be officially supported. You should use instead a pure FreeFlyer joint, and to work with integrate and difference operation to work on the tangent space of the configuration space.

jcarpent, 2023-02-20:
Please have a look at this C++ example:
https://github.com/stack-of-tasks/pinocchio/blob/0d53f26acdf9a0b5740b88145f624b11d97f40f9/unittest/casadi/rnea-derivatives.cpp#L24-L174

matheecs, 2023-02-22:
Do I have to select **error state** $\in$ tangent space instead of euler angle or quaternion as the variables in optimization?

closed 2023-04-09 by matheecs