stack-of-tasks/pinocchio #1873: Numpy version against in Ubuntu 22.04

botoph4, 2023-02-21:
Hey,

Here is log I got:
``
RuntimeError: module compiled against API version 0x10 but this version of numpy is 0xe . Check the section C-API incompatibility at the Troubleshooting ImportError section at https://numpy.org/devdocs/user/troubleshooting-importerror.html#c-api-incompatibility for indications on how to solve this problem .
``

This error  easy to reproduce while `import pinocchio`.

System: Ubuntu 22.04
Python version: 3.10.6
numpy version: 1.21.5

jcarpent, 2023-02-21:
COuld you tell us how did you install pinocchio?

jcarpent, 2023-02-21:
I can reproduce the issue in a docker image. 
@nim65s Do you know the origin of the problem?

jcarpent, 2023-02-21:
The issue seems related to cmeel-boost 1.81

jcarpent, 2023-02-21:
@botoph4 In the meantime, you can use the command:
```python
pip install numpy --upgrade
```
to get the latest numpy version and this should work.

duburcqa, 2023-02-21:
Numpy C API is NOT backward compatible but is forward compatible. This is specified deep somewhere inside the documentation of Numpy. So you should compile pinocchio with an older version of Numpy.

nim65s, 2023-02-21:
We don't support older versions of numpy in cmeel.

@jcarpent latest comment is not a temporary workaround, but the expected solution in this case.

If one wants a pinocchio version compatible with system packages from one particular linux distribution, they can either use packages specifically build for that distribution, like those provided by robotpkg, or build the package themselves, but this is not in the scope of cmeel, which aims at compatibility with the pip / PyPI ecosystem and following relevant PEPs.

To me, the issue here is that the versions constraints are not correctly published. I wasn't aware of those API version. According to https://github.com/numpy/numpy/blob/29cb0ef6bb47a942d77bf7479a870a0c88af6fc7/numpy/core/include/numpy/numpyconfig.h#L81-L83, we're compatible with >= 1.23. I'll add that to cmeel-boost & eigenpy, and trigger a rebuild of the latter.

duburcqa, 2023-02-21:
If you don't want to support numpy <= 1.23 then yes it is just a version constraint issue. Yet, i'm surprise that you really need such a strict requirement. 1.23 is the latest version of numpy that it fully supported by all common libraries. Many are still not compatible with 1.24.

nim65s, 2023-02-21:
I don't understand your comment.

At least, it looks like you understood that 1.23 support is not included, which is a wrong interpretation of "we're compatible with >= 1.23".

duburcqa, 2023-02-21:
I just saying that some people may not be able to fullfill such a strict requirement because 1.23 is considered very recent for some Python packages reluctant to update their dependencies, e.g. Tensorflow and many other industry-oriented packages.

botoph4, 2023-02-22:
Thx for all the discussions here. 
@jcarpent yes, bump up to 1.24.2 as the lastest numpy version and it works. Thx!
Since @nim65s found the solution already, I am going to close this issue.

closed 2023-02-22 by botoph4