stack-of-tasks/pinocchio #1879: Catkin build failure when generating python stubs

james-p-foster, 2023-03-02:
After seeing the recent PR adding the generation of python stubs here: https://github.com/stack-of-tasks/pinocchio/pull/1869 I have tried to build the `devel` branch with catkin, but I get build errors.

I have set up a new workspace with just `pinocchio` and its dependencies, `hpp-fcl` and `eigenpy`, and replicated the issue.

Concretely, when I run:
```
catkin build -j10 -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python3.8 -DBUILD_WITH_COLLISION_SUPPORT=ON -DGENERATE_PYTHON_STUBS=ON
```

I get the following:
```
Traceback (most recent call last):
  File "/home/jfoster/dummy-workspace/build/pinocchio/bindings/python/stubgen/src/pybind11_stubgen/__init__.py", line 1010, in <module>
    main()
  File "/home/jfoster/dummy-workspace/build/pinocchio/bindings/python/stubgen/src/pybind11_stubgen/__init__.py", line 988, in main
    _module = ModuleStubsGenerator(_module_name)
  File "/home/jfoster/dummy-workspace/build/pinocchio/bindings/python/stubgen/src/pybind11_stubgen/__init__.py", line 731, in __init__
    self.module = importlib.import_module(module_or_module_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/jfoster/dummy-workspace/src/pinocchio/bindings/python/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
SystemError: initialization of pinocchio_pywrap raised unreported exception
make[2]: *** [bindings/python/CMakeFiles/generate_stubs__home_jfoster_dummy-workspace_build_pinocchio_bindings_python_pinocchio.dir/build.make:73: bindings/python/CMakeFiles/generate_stubs__home_jfoster_dummy-workspace_build_pinocchio_bindings_python_pinocchio] Error 1
make[1]: *** [CMakeFiles/Makefile2:1798: bindings/python/CMakeFiles/generate_stubs__home_jfoster_dummy-workspace_build_pinocchio_bindings_python_pinocchio.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:149: all] Error 2
```

When I drop the `-DGENERATE_PYTHON_STUBS=ON` flag, everything builds perfectly.

james-p-foster, 2023-03-03:
I still haven't resolved the error, but note this part of the traceback:
```
  File "/home/jfoster/dummy-workspace/src/pinocchio/bindings/python/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
```
For some reason, the `__init__.py` from the `pinocchio` source code is being called, where a relative import of the `pywrap` module will obviously fail because the .so module doesn't exist there. Rather, the `__init__.py` file in `<devel or install>/lib/python3/dist-packages/...> should be called, where the `pywrap` module is actually built. I wonder if I'm running into some conflict between my `src` and `install` spaces when building pinocchio.

james-p-foster, 2023-03-03:
To test my above comment, I dropped building via ROS `catkin` and instead downloaded `pinocchio` from source and tried to build via `cmake`. I got exactly the same kind of traceback.
```
Traceback (most recent call last):
  File "/home/jfoster/pinocchio-workspace/build/bindings/python/stubgen/src/pybind11_stubgen/__init__.py", line 1010, in <module>
    main()
  File "/home/jfoster/pinocchio-workspace/build/bindings/python/stubgen/src/pybind11_stubgen/__init__.py", line 988, in main
    _module = ModuleStubsGenerator(_module_name)
  File "/home/jfoster/pinocchio-workspace/build/bindings/python/stubgen/src/pybind11_stubgen/__init__.py", line 731, in __init__
    self.module = importlib.import_module(module_or_module_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/jfoster/pinocchio-workspace/bindings/python/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
SystemError: initialization of pinocchio_pywrap raised unreported exception
make[2]: *** [bindings/python/CMakeFiles/generate_stubs__home_jfoster_pinocchio-workspace_build_bindings_python_pinocchio.dir/build.make:73: bindings/python/CMakeFiles/generate_stubs__home_jfoster_pinocchio-workspace_build_bindings_python_pinocchio] Error 1
make[2]: Leaving directory '/home/jfoster/pinocchio-workspace/build'
make[1]: *** [CMakeFiles/Makefile2:1772: bindings/python/CMakeFiles/generate_stubs__home_jfoster_pinocchio-workspace_build_bindings_python_pinocchio.dir/all] Error 2
```
It's again trying to do a relative import for the `pywrap` module where it doesn't exist, I think.

wxmerkt, 2023-03-05:
Hi @james-p-foster, many thanks for the detailed issue report. A brief test on CI (https://github.com/stack-of-tasks/pinocchio/actions/runs/4318776324/jobs/7537472293) showed that there shouldn't be an issue for building with Python stub generation in a clean catkin environment [we may want to merge #1880  to test stub generation on CI]. I haven't tried it locally though so cannot comment further at this point. Perhaps someone who's using Python stubs on a more regular basis could comment since it also appears to happen for the regular CMake workflow? @jcarpent @duburcqa ?

jcarpent, 2023-03-11:
@james-p-foster Could you provide a docker image that reproduces the issue? This will help us to dive into the bug directly.
Thanks in advance.

Best,
Justin

jcarpent, 2023-03-13:
> To test my above comment, I dropped building via ROS `catkin` and instead downloaded `pinocchio` from source and tried to build via `cmake`. I got exactly the same kind of traceback.
> 
> When building with:
> 
> ```
> cmake .. 
> -DCMAKE_INSTALL_PREFIX=/home/jfoster/pinocchio-install
> -DCMAKE_PREFIX_PATH=/opt/openrobots/include  # to give access to eigenpy and hpp-fcl via robotpkg
> -DCMAKE_BUILD_TYPE=Release
> -DPYTHON_EXECUTABLE=/usr/bin/python3.8
> -DBUILD_WITH_COLLISION_SUPPORT=ON
> -DGENERATE_PYTHON_STUBS=ON
> ```
> 
> I get:
> 
> ```
> Traceback (most recent call last):
>   File "/home/jfoster/pinocchio-workspace/build/bindings/python/stubgen/src/pybind11_stubgen/__init__.py", line 1010, in <module>
>     main()
>   File "/home/jfoster/pinocchio-workspace/build/bindings/python/stubgen/src/pybind11_stubgen/__init__.py", line 988, in main
>     _module = ModuleStubsGenerator(_module_name)
>   File "/home/jfoster/pinocchio-workspace/build/bindings/python/stubgen/src/pybind11_stubgen/__init__.py", line 731, in __init__
>     self.module = importlib.import_module(module_or_module_name)
>   File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>   File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 991, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
>   File "<frozen importlib._bootstrap_external>", line 848, in exec_module
>   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
>   File "/home/jfoster/pinocchio-workspace/bindings/python/pinocchio/__init__.py", line 6, in <module>
>     from .pinocchio_pywrap import *
> SystemError: initialization of pinocchio_pywrap raised unreported exception
> make[2]: *** [bindings/python/CMakeFiles/generate_stubs__home_jfoster_pinocchio-workspace_build_bindings_python_pinocchio.dir/build.make:73: bindings/python/CMakeFiles/generate_stubs__home_jfoster_pinocchio-workspace_build_bindings_python_pinocchio] Error 1
> make[2]: Leaving directory '/home/jfoster/pinocchio-workspace/build'
> make[1]: *** [CMakeFiles/Makefile2:1772: bindings/python/CMakeFiles/generate_stubs__home_jfoster_pinocchio-workspace_build_bindings_python_pinocchio.dir/all] Error 2
> ```
> 
> It's again trying to do a relative import for the `pywrap` module where it doesn't exist, I think.

@nim65s Are you able to reproduce this issue via robotpkg?

jcarpent, 2023-03-23:
I will close this issue as there is no more activity on it.

closed 2023-03-23 by jcarpent