stack-of-tasks/pinocchio #1884: BOOST_MPL_LIMIT_*_SIZE without BOOST_MPL_CFG_NO_PREPROCESSED_HEADERS in cmakelists.txt

olivierrochel-inria, 2023-03-08:
Hello,

#852 added the definition of BOOST_MPL_LIMIT_LIST_SIZE and BOOST_MPL_LIMIT_LIST_SIZE in the CMakeLists.txt (devel branch), but these seem to be useless if BOOST_MPL_CFG_NO_PREPROCESSED_HEADERS is not defined (boost doc says: Overriding will take effect only if the library is configured not to use preprocessed headers). 

BOOST_MPL_CFG_NO_PREPROCESSED_HEADERS is indeed correctly set in boost-container-limits.hpp where the MPL LIMITS are also redundantly defined, but not in CMakeLists.txt.

I've had a case where the fix in #852 was helpful, but required this additional define. A better fix on our side may have been to make sure boost-container-limits.hpp is correctly take into account but in any case, IIRC, #852 may be in need of a complementary BOOST_MPL_CFG_NO_PREPROCESSED_HEADERS define to be truly relevant in all situations ?

jcarpent, 2023-03-08:
Thanks for this notification @olivierrochel-inria.
If I understand, you advised to add `BOOST_MPL_CFG_NO_PREPROCESSED_HEADERS` in the compile definitions within the CMakeLists.txt?

olivierrochel-inria, 2023-03-08:
Yes, that would align things with the include file at least.
I hope these settings might become unneeded one day... but in the short term it should be more coherent like this.

jcarpent, 2023-03-09:
See discussions in #1885 for concluding remark.

closed 2023-03-09 by jcarpent