stack-of-tasks/pinocchio #1895: Floating-base position in "ComputeFrameJacobian"

niteofhunter, 2023-03-21:
Hi, I have a question about "ComputeFrameJacobian"

I included the bellow model with Pinocchio and tried to get the jacobian matrix of the front-left foot with "computeFrameJacobian(model, data, q, frameId=WORLD,reference_frame=left_foot_id, J)".

[image]

At first, I input the body's COM position in the world frame in q => q = [0,0,0.33,0,0,0,1,{joints}]

At that time, the result is as below, which does not reflect the body's length(0.208) and width(0.09).

[image]

However, when I put the body's position relative to the "front-left foot"(q = [-0.208,-0.09,0.33,0,0,0,1,{joints}]), it returns the right Jacobian
 
[image]

My question is, "Should I really put the body's relative position to the foot to get the right jacobian matrix??? or Is it just a coincidence???(Is there a standard way to get a jacobian matrix of floating base type robot with computeFrameJacobian???)" No one saying about this issue, So I'm very confused.

Thank you ^^

MegMll, 2023-03-22:
Hi @niteofhunter, 

Not sure I truly understand what you are trying to do, but using WORLD as a frame does not seem to be the best way to go, since it means that you are computing the Jacobian of the point coincident with the origin of the WORLD and expressed in WORLD coordinates.

To get the Jacobian you  probably want, you should use the convention LOCAL_WORLD_ALIGNED to compute the Jacobian and express it in the WORLD coordinates.

niteofhunter, 2023-03-22:
> Hi @niteofhunter,
>
> Not sure I truly understand what you are trying to do, but using WORLD as a frame does not seem to be the best way to go, since it means that you are computing the Jacobian of the [...]

WOW!!! That's what I truly want!!!

I didn't think that "LOCAL_WORLD_ALIGNED" existed.

You save my life^^. Thank you^^

closed 2023-03-22 by MegMll