stack-of-tasks/pinocchio #1896: Installation Ubuntu 22.04 / Installation from Source

hubernikus, 2023-03-21:
When installing through instructions as given here:
https://stack-of-tasks.github.io/pinocchio/download.html

I get the dependcy issues:

`$ cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local
-- Configuring "pinocchio" (http://github.com/stack-of-tasks/pinocchio)
-- Package version (ROS package.xml): 2.6.17
CMake Error at cmake/package-config.cmake:110 (find_package):
  By not providing "Findurdfdom_headers.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "urdfdom_headers", but CMake did not find one.

  Could not find a package configuration file provided by "urdfdom_headers"
  with any of the following names:

    urdfdom_headersConfig.cmake
    urdfdom_headers-config.cmake

  Add the installation prefix of "urdfdom_headers" to CMAKE_PREFIX_PATH or
  set "urdfdom_headers_DIR" to a directory containing one of the above files.
  If "urdfdom_headers" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:117 (ADD_PROJECT_DEPENDENCY)

-- Configuring incomplete, errors occurred!
See also "/home/lukas/Code/robot_control/libs/pinocchio/build/CMakeFiles/CMakeOutput.log".
`

jcarpent, 2023-03-21:
You should use the `CMAKE_MODULE_PATH` variable as mentioned in the error for installation from source.
Please see the doc here https://cmake.org/cmake/help/latest/variable/CMAKE_MODULE_PATH.html.

Concerning apt installation, it seems you are working with ubuntu 18.04 which is almost EOL.

nim65s, 2023-03-21:
py36 is 16.04, which is EOL. Are you sure you are on 22.04 ?

jcarpent, 2023-03-22:
@hubernikus On 22.04, you should run:

```shell
sudo apt install robotpkg-py310-pinocchio
```

hubernikus, 2023-03-22:
Yes, I'm on Ubuntu 22.04.

I was surprised too, but setting up the robotpkg from http://robotpkg.openrobots.org/debian.html, the only options I had were py27 / py36, which are both EOL... 

I also guess the installation procedure in 
https://stack-of-tasks.github.io/pinocchio/download.html
is out of date.

In the end, I managed by installing from source, and adapting an install file below

```
#!/bin/bash
SCRIPT=$(readlink -f "${BASH_SOURCE[0]}")
SOURCE_PATH=$(dirname "$SCRIPT")

BUILD_ROBOT_MODEL="ON"

EIGEN_VERSION=3.4.0
OSQP_TAG=0.6.2
OSQP_EIGEN_TAG=0.6.4
PINOCCHIO_TAG=2.6.9

# cleanup any previous build folders
rm -rf "${SOURCE_PATH}"/tmp

echo ">>> INSTALLING BASE DEPENDENCIES"

if [ -z $(which pkg-config) ]; then
	echo ">>> INSTALLING pkg-config tool"
	apt-get update && apt-get install "${AUTO_INSTALL}" pkg-config || exit 1
fi

pkg-config eigen3 --atleast-version="${EIGEN_VERSION}"
if [ "$?" != 0 ]; then
	echo ">>> INSTALLING EIGEN"
	mkdir -p "${SOURCE_PATH}"/tmp/lib && cd "${SOURCE_PATH}"/tmp/lib || exit 1
	wget -c "https://gitlab.com/libeigen/eigen/-/archive/${EIGEN_VERSION}/eigen-${EIGEN_VERSION}.tar.gz" -O - | tar -xz || exit 1
	cd "eigen-${EIGEN_VERSION}" && mkdir -p build && cd build && env CXXFLAGS=-DEIGEN_MPL2_ONLY cmake .. && make install || exit 1
fi

EIGEN_PATH=$(cmake --find-package -DNAME=Eigen3 -DCOMPILER_ID=GNU -DLANGUAGE=C -DMODE=COMPILE)
if [ "${EIGEN_PATH::14}" != "-I/usr/include" ]; then
	rm -rf /usr/include/eigen3 && ln -s ${EIGEN_PATH:2} /usr/include/eigen3 || exit 1
fi

if [ "${BUILD_ROBOT_MODEL}" == "ON" ]; then
  echo ">>> INSTALLING ROBOT MODEL DEPENDENCIES"
  apt-get update && apt-get install "${AUTO_INSTALL}" libboost-all-dev liburdfdom-dev || exit 1

  pkg-config pinocchio --atleast-version=${PINOCCHIO_TAG}
  if [ "$?" != 0 ]; then
    mkdir -p "${SOURCE_PATH}"/tmp/lib && cd "${SOURCE_PATH}"/tmp/lib || exit 1

    echo ">>> INSTALLING OSQP [1/3]"
    git clone --depth 1 -b v${OSQP_TAG} --recursive https://github.com/oxfordcontrol/osqp \
        && cd osqp && mkdir build && cd build && cmake -G "Unix Makefiles" .. && cmake --build . --target install \
        && cd ../.. && rm -r osqp || exit 1

    echo ">>> INSTALLING OSQP_EIGEN [2/3]"
    git clone --depth 1 -b v${OSQP_EIGEN_TAG} https://github.com/robotology/osqp-eigen.git \
        && cd osqp-eigen && mkdir build && cd build && cmake .. && make -j && make install \
        && cd ../.. && rm -r osqp-eigen || exit 1

    echo ">>> INSTALLING PINOCCHIO [3/3]"
    git clone --depth 1 -b v${PINOCCHIO_TAG} --recursive https://github.com/stack-of-tasks/pinocchio \
        && cd pinocchio && mkdir build && cd build \
        && cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DBUILD_PYTHON_INTERFACE=OFF \
        && make -j1 && make install && cd ../.. && rm -r pinocchio || exit 1
  fi
  ldconfig
fi

# Cleanum and reset
rm -rf "${SOURCE_PATH}"/tmp
cd "${SOURCE_PATH}" || return
```

jcarpent, 2023-03-22:
OK. Thanks for the feedback.

closed 2023-03-22 by jcarpent

nim65s, 2023-03-22:
> I was surprised too, but setting up the robotpkg from http://robotpkg.openrobots.org/debian.html, the only options I had were py27 / py36, which are both EOL...

In that link, I guess you missed one detail:

> You must of course replace the codename "bionic" by you actual distribution codename!

hubernikus, 2023-03-24:
Indeed, I missed changing the codename to the correct distribution; that should have solved it too. Thank you for pointing it out.