stack-of-tasks/pinocchio #1897: When building a model the addJoint of non-Fixed joints need to be called in a strict depth-first order

traversaro, 2023-03-22:
While working on https://github.com/ami-iit/idynfor/pull/26, I tested the pinocchio Model building API with many randomly generated models. While doing that, I noticed that:
~~~
auto dataPin = pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>(modelPin);
if (!modelPin.check(dataPin))
{
    // handle error
}
~~~
failed if the `addJoint` used to build the pinocchio model were not in a strictly depth-first order, in particular the check was failing in https://github.com/stack-of-tasks/pinocchio/blob/78d62096002ffa3790638e392f0b6e4a5efc3d34/src/algorithm/check.hxx#L156 .

I can prepare a self-contained example of this behaviour, but I anyhow started by opening an issue as I am not sure if this behaviour is intended (and in that case I would be happy to add some documentation on this) or instead it is not intended.

jcarpent, 2023-03-22:
Hi Silvio!
Thanks for the issue and the related questions. 
This is an expected behavior indeed. We enforce this strategy to get a compact structure when operating on subtrees. This is one of the reasons for Pinocchio's efficiency (e.g. CRBA). Yet, this is not mandatory for (ABA, RNEA, etc.).

Hope my explanations are clear enough,
Best,
Justin

traversaro, 2023-03-22:
Ack, I imagined this. So if it is ok for you we can keep this issue open, and I will add a note in the AddJoint method to mention this, thanks!

jcarpent, 2023-03-22:
Perfect !

jcarpent, 2023-03-22:
Solved via #1899.