stack-of-tasks/pinocchio #1902: Inconsistent collision checking when updating from hpp-fcl 2.2 to 2.3

kingjin94, 2023-03-24:
Dear Pinocchio Team,

It seems that somewhere in the collision checking stack changes have occurred that significantly alter collision checking results. As I am not totally into the depths I can only provide the following example at the moment based on the pinocchio wrapper timor-python:

```
!pip install hpp-fcl==2.2.0 (or 2.3.0)  #Force a specific hpp-fcl version
!pip install timor-python

import timor
assembly = timor.ModuleAssembly.from_serial_modules(timor.ModulesDB.from_name('PROMODULAR'), ["1", "4", "16", "3", "8", "5", "45", "5", "52", "5", "39", "5", "39"])
assembly.robot.has_self_collision()  # Will be true if installed with hpp-fcl 2.3 and false with 2.2
```

The `assembly.robot` object is a wrapper for pinocchio's RobotWrapper. The essential properties for pinocchio are (as needed for pin.computeCollisions):

- assembly.robot.model -> pinocchio.pinocchio_pywrap.Model
- assembly.robot.data -> assembly.robot.data
- assembly.robot.collision -> pinocchio.pinocchio_pywrap.GeometryModel
- assembly.robot.collision_data -> pinocchio.pinocchio_pywrap.GeometryData

The self-collision pairs can be visualized as such: assembly.robot.visualize_self_collisions(); here same color = two parts of the robot in collision. From this robot it seems all new collisions are between meshes and cylinders that lie on neighboring links (if that helps).

Are you aware of any recent changes in pinocchio / hpp-fcl that might result in such different behaviors? Do you have any ideas how to debug this further?

jcarpent, 2023-03-24:
Thanks @kingjin94 for raising this issue.
With @lmontaut, we are investing the issue according to the diff: https://github.com/humanoid-path-planner/hpp-fcl/compare/v2.2.0..v2.3.0

jcarpent, 2023-03-25:
The issue is solved via https://github.com/humanoid-path-planner/hpp-fcl/releases/tag/v2.3.1.
@kingjin94 You should also consider changing from BVHModelOBBRSS to Convex.

closed 2023-03-25 by jcarpent

kingjin94, 2023-03-27:
Thanks to you and your team as well. Your response time and dedication is amazing!

jcarpent, 2023-03-27:
Thanks @kingjin94 for acknowledging our collective efforts to produce and maintain open-source toolboxes which might help the robotics community to move forward.