stack-of-tasks/pinocchio #1913: RNEA not taking external forces into account in Python?

stephane-caron, 2023-04-03:
I'm trying out `pin.rnea` in Python with a list of external forces applied to each joint:

- Zero external force on each joint
- Except at the left/right wheel joints (robot model is Upkie) where there is a $F_z$ vertical force

Here is a minimal example with $F_z = 100$ N:

```python
import numpy as np
import pinocchio as pin
from robot_descriptions.loaders.pinocchio import load_robot_description

robot = load_robot_description("upkie_description")
model = robot.model
theta = 1.0  # rad
q = np.array([theta, -2 * theta, 0, theta, -2 * theta, 0])
v = np.zeros(model.nv)
a = np.zeros(model.nv)

f_ext = [pin.Force.Zero() for j in range(model.njoints)]
f_ext[model.getJointId("left_wheel")].linear[2] += 100.0  # N
f_ext[model.getJointId("right_wheel")].linear[2] += 100.0  # N

tau = pin.rnea(model, robot.data, q, v, a, f_ext)
print(f"{tau=}")
```

The issue is that the outcome of this example is the same ``tau`` regardless of $F_z$ :thinking: You can try it with 10, 100, 1000, ...

stephane-caron, 2023-04-03:
My guess would be that RNEA has some undocumented requirements, for example maybe FK should be run beforehand?

jcarpent, 2023-04-03:
By using:
```python
import numpy as np
import pinocchio as pin
from robot_descriptions.loaders.pinocchio import load_robot_description

robot = load_robot_description("upkie_description")
model = robot.model
theta = 1.0  # rad
q = np.array([theta, -2 * theta, 0, theta, -2 * theta, 0])
v = np.zeros(model.nv)
a = np.zeros(model.nv)

f_ext = [pin.Force.Zero() for j in range(model.njoints)]
f_ext[model.getJointId("left_knee")].linear[:] += 100.0  # N
f_ext[model.getJointId("right_knee")].linear[:] += 100.0  # N

tau = pin.rnea(model, robot.data, q, v, a, f_ext)
print(f"{tau=}")
```
you will that it works well.
I guess this is due to the current configuration where normal contact forces have no impact on the joint torques

stephane-caron, 2023-04-03:
You're right, I forgot that external forces are in local joint frames, so I should have checked the orientation of those frames rather than assuming it would be vertical :sweat_smile: 

As it turns out, the z-axis in the local frames is lateral (source: [`display_joints.py`](https://github.com/tasts-robots/upkie_description/blob/main/examples/display_joints.py)):

[image]

When correctly applying $F_z$ on the local y- rather than z-axis it works as expected :+1: Thanks!

closed 2023-04-03 by stephane-caron