stack-of-tasks/pinocchio #1915: Linking error Pinocchio version 2.6.17

mfocchi, 2023-04-04:
Dear all

I updated Pinocchio  to the latest package (version 2.6.17) available for Ubuntu 20, but it started to give me a linking error:

/opt/openrobots/lib/python3.8/site-packages/pinocchio/pinocchio_pywrap.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZNK3hpp3fcl15ComputeDistanceclERKNS0_11Transform3fES4_RKNS0_15DistanceRequestERNS0_14DistanceResultE

any idea?

Michele

fabinsch, 2023-04-04:
Hey @mfocchi, how did you update Pinocchio?

I just tried in a docker container (Ubuntu 20.04) to install Pinocchio 2.6.17 in a clean conda environment and it works without any error.

```
ARG UBUNTU_VER=20.04
ARG CONDA_VER=latest
ARG OS_TYPE=x86_64
ARG PY_VER=3.8

FROM ubuntu:${UBUNTU_VER}

RUN apt-get update && apt-get install -yq curl wget jq vim git gnupg2

ARG CONDA_VER
ARG OS_TYPE
RUN curl -LO "http://repo.continuum.io/miniconda/Miniconda3-${CONDA_VER}-Linux-${OS_TYPE}.sh"
RUN bash Miniconda3-${CONDA_VER}-Linux-${OS_TYPE}.sh -p /miniconda -b
RUN rm Miniconda3-${CONDA_VER}-Linux-${OS_TYPE}.sh
ENV PATH=/miniconda/bin:${PATH}
RUN conda update -y conda
RUN conda install -c conda-forge mamba
RUN mamba install -c anaconda -y python=${PY_VER}
RUN mamba install -c conda-forge -y pinocchio 
RUN python -c "import pinocchio; print(pinocchio.__version__)" 
```

the output is:
```
(base) ➜  docker_debug docker run -it debug_pin /bin/bash
root@504aef0527ac:/# python -c "import pinocchio; print(pinocchio.__version__)" 
2.6.17
```

mfocchi, 2023-04-04:
Hi @fabinsch 

I was not using conda but the usual apt package install command, in a ubuntu 20 docker environment where I had previously installed Pinocchio 2.6.3. 
I just run: apt install robotpkg-py38-pinocchio

and this command involved the update of Pinocchio to 2.6.17
Setting up robotpkg-pinocchio (2.6.17) ...

and also of these other packages that were previously installed: 

Setting up robotpkg-py38-eigenpy (2.9.2) 
Setting up robotpkg-py38-hpp-fcl (2.3.0) ...
Setting up robotpkg-py38-pinocchio (2.6.17) ...
Setting up robotpkg-tsid (1.6.3) ...
Setting up robotpkg-py38-example-robot-data (4.0.5) ...

still, when I run:

python3 -c "import pinocchio; print(pinocchio.__version__)" 

I get the above linking error, environment variables are properly set.

fabinsch, 2023-04-04:
Hi @mfocchi , okay I understand. I did the installation via `apt install -qqy robotpkg-py38-pinocchio` in a docker container (Ubuntu 20.04) and again it works fine for me.

```
ARG UBUNTU_VER=20.04

FROM ubuntu:${UBUNTU_VER}

# System packages 
ENV TZ=Europe/Paris
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone
RUN apt-get update && apt-get install -yq curl wget jq vim git gnupg2
RUN apt-get update && apt install -qqy lsb-release gnupg2 curl

RUN echo "deb [arch=amd64] http://robotpkg.openrobots.org/packages/debian/pub $(lsb_release -cs) robotpkg" | tee /etc/apt/sources.list.d/robotpkg.list
RUN curl http://robotpkg.openrobots.org/packages/debian/robotpkg.key | apt-key add -
RUN apt update && apt install -qqy robotpkg-py38-pinocchio

ENV PATH=/opt/openrobots/bin:$PATH
ENV PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig:$PKG_CONFIG_PATH
ENV LD_LIBRARY_PATH=/opt/openrobots/lib:$LD_LIBRARY_PATH
ENV PYTHONPATH=/opt/openrobots/lib/python3.8/site-packages:$PYTHONPATH
ENV CMAKE_PREFIX_PATH=/opt/openrobots:$CMAKE_PREFIX_PATH

RUN python3 -c "import pinocchio; print(pinocchio.__version__)" 
```

fabinsch, 2023-04-04:
what do you get if you run `ldd pinocchio_pywrap.cpython-38-x86_64-linux-gnu.so.2.6.17` in `/opt/openrobots/lib/python3.8/site-packages/pinocchio` ?

mfocchi, 2023-04-04:
I get this: 

mfocchi@docker:/opt/openrobots/lib/python3.8/site-packages/pinocchio$ ldd pinocchio_pywrap.cpython-38-x86_64-linux-gnu.so.2.6.17
	linux-vdso.so.1 (0x00007ffcba13a000)
	libeigenpy.so => /opt/openrobots/lib/libeigenpy.so (0x00007f365fc4d000)
	libpinocchio.so.2.6.17 => /opt/openrobots/lib/libpinocchio.so.2.6.17 (0x00007f365fc0f000)
	libboost_python38.so.1.71.0 => /usr/lib/x86_64-linux-gnu/libboost_python38.so.1.71.0 (0x00007f365fbce000)
	liburdfdom_world.so.1.0 => /usr/lib/x86_64-linux-gnu/liburdfdom_world.so.1.0 (0x00007f365fba8000)
	libhpp-fcl.so => /opt/openrobots/lib/libhpp-fcl.so (0x00007f365f948000)
	libboost_serialization.so.1.71.0 => /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.71.0 (0x00007f365f902000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f365f720000)
	libm.so.6 => /usr/lib/x86_64-linux-gnu/libm.so.6 (0x00007f365f5d1000)
	libgcc_s.so.1 => /usr/lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f365f5b6000)
	libc.so.6 => /usr/lib/x86_64-linux-gnu/libc.so.6 (0x00007f365f3c4000)
	libboost_filesystem.so.1.71.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007f365f3a6000)
	liburdfdom_model.so.1.0 => /usr/lib/x86_64-linux-gnu/liburdfdom_model.so.1.0 (0x00007f365f37f000)
	libpthread.so.0 => /usr/lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f365f35c000)
	libtinyxml.so.2.6.2 => /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007f365f344000)
	libconsole_bridge.so.0.4 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007f365f33e000)
	libassimp.so.5 => /usr/lib/x86_64-linux-gnu/libassimp.so.5 (0x00007f365e6f1000)
	libqhull_r.so.7 => /usr/lib/x86_64-linux-gnu/libqhull_r.so.7 (0x00007f365e497000)
	liboctomap.so.1.9 => /opt/openrobots/lib/liboctomap.so.1.9 (0x00007f365e42d000)
	liboctomath.so.1.9 => /opt/openrobots/lib/liboctomath.so.1.9 (0x00007f365e425000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f36619b5000)
	libz.so.1 => /usr/lib/x86_64-linux-gnu/libz.so.1 (0x00007f365e409000)
	libminizip.so.1 => /usr/lib/x86_64-linux-gnu/libminizip.so.1 (0x00007f365e1fe000)

fabinsch, 2023-04-04:
alright, that looks the same on my side. what do you get for `python3 -c "import hppfcl;print(hppfcl.__version__)"` ?

mfocchi, 2023-04-04:
ImportError: /opt/openrobots/lib/python3.8/site-packages/hppfcl/hppfcl.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZNK3hpp3fcl11HeightFieldINS0_4AABBEE11getNodeTypeEv

mfocchi, 2023-04-05:
Dear @nim65s  

I solved after the upgrade, everything now works, I was reluctant to do it because my docker is shared with hundred students, now it works, it might be there was some dependency non automatically triggered when I updated the single packages alone?

nim65s, 2023-04-05:
In a general way, working with docker and apt is often prone to package cache issues. This is even worse for robotpkg debs, which are not meant to be updated separately, but apt doesn't know that.

I'd agree with @fabinsch here, and suggest re-building the initial `apt update && apt install robotpkg-whatever` part of your Dockerfile when you need an update, instead of adding an `apt update && apt upgrade robotpkg-something` afterwards.  This would also help with the final size of the docker images, as there would be only one version of each package per image.

mfocchi, 2023-04-06:
Ok, that's an awesome suggestion, thanks!

jcarpent, 2023-04-06:
I will close this issue as it seems to be solved.
Thanks to @nim65s and @fabinsch for taking to answer and provide assistance to @mfocchi.