stack-of-tasks/pinocchio #1916: Result of inverse dynamics is inconsistent

matheecs, 2023-04-09:
I use two ways to calculate $\tau$

* RNEA
* Equations of motion (EOM): $Ma + nonLinearEffects = \tau + J^Tf_{ext}$

but results was different. A part of implementation is follow

```python
......
......
......
f_ext_local = [pin.Force.Zero() for j in range(model.njoints)]
JtF = np.zeros(model.nq)

for i in range(len(contact_frame)):
    frameIndex = model.getFrameId(contact_frame[i])
    jointIndex = model.frames[frameIndex].parentJoint
    translation = model.frames[frameIndex].placement.translation
    rotation = data.oMi[jointIndex].rotation
    contactForce = rotation.T @ contact_force_in_world[3 * i : 3 * i + 3]
    f_ext_local[jointIndex].linear = contactForce
    f_ext_local[jointIndex].angular = np.cross(
        translation, contactForce
    )

    J = pin.getFrameJacobian(
        model, data, frameIndex, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED
    )[:3, :]
    JtF = JtF + J.T @ contact_force_in_world[3 * i : 3 * i + 3] 

nle = pin.nonLinearEffects(model, data, q, v)
M = pin.crba(model, data, q)
print("torque_from_EOM = ", M @ a + nle - JtF)
print("torque_from_rnea = ", pin.rnea(model, data, q, v, a, f_ext_local))
```

the result is different:

```text
torque_from_EOM =  [ -0.     0.     0.01   0.   -10.24 -29.11   0.   -26.46  -7.29 -57.35   9.93  -0.     9.57  -5.25 -25.1    4.41   0.    -1.52   0.36   0.    -1.06  -1.52  -0.36   0.    -1.06]
torque_from_rnea =  [  0.     0.   212.14  -0.   -24.38 -14.55  -0.   -10.87 -14.86 -37.03  -4.62  -0.     2.42  -8.72 -15.78  -2.25   0.    -1.52   0.36   0.    -1.06  -1.52  -0.36   0.    -1.06]
```

So, which is correct?

jcarpent, 2023-04-10:
`f_ext_local[jointIndex].angular = np.cross(
        translation, contactForce
    ) ` is not correct when transforming from LOCAL_WORLD_ALIGNED to LOCAL. You should only rotate the quantities.

matheecs, 2023-04-10:
Thanks.

Can you give more details to fix it? If only delete the line `f_ext_local[jointIndex].angular = np.cross( translation, contactForce)`, the result is still different.

jcarpent, 2023-04-10:
Please provide your full but tiny example

matheecs, 2023-04-11:
Thank you for the help! There is a tiny example:

Env: Pinocchio3 + Python

```python
import pinocchio as pin
import numpy as np
import os

np.set_printoptions(precision=2, suppress=True, linewidth=200)

urdf_file =  "model.urdf"
jc = pin.JointModelComposite()
jc_xyz = jc.addJoint(pin.JointModelTranslation())
jc_xyz_zyx = jc_xyz.addJoint(pin.JointModelSphericalZYX())
model, collision_model, visual_model = pin.buildModelsFromUrdf(
    urdf_file, root_joint=jc_xyz_zyx
)
data = model.createData()

q = np.array([0, 0, 0.56, 0, 0, 0, 0, 0])
v = np.zeros(model.nv)
a = np.zeros(model.nv)
normal_force = abs(pin.computeTotalMass(model) * model.gravity.linear[2] / 4)
contact_force_in_world = [0.0, 0.0, normal_force] * 4
contact_frame = ["lf_contact", "lb_contact", "rf_contact", "rb_contact"]

f_ext_local = [pin.Force.Zero() for j in range(model.njoints)]
JtF = np.zeros(model.nq)

for i in range(len(contact_frame)):
    frameIndex = model.getFrameId(contact_frame[i])
    jointIndex = model.frames[frameIndex].parentJoint
    translation = model.frames[frameIndex].placement.translation
    rotation = data.oMi[jointIndex].rotation
    contactForce = rotation.T @ contact_force_in_world[3 * i : 3 * i + 3]
    f_ext_local[jointIndex].linear = contactForce
    f_ext_local[jointIndex].angular = np.cross(translation, contactForce)

    J = pin.getFrameJacobian(
        model, data, frameIndex, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED
    )[:3, :]
    JtF = JtF + J.T @ contact_force_in_world[3 * i : 3 * i + 3]

nle = pin.nonLinearEffects(model, data, q, v)
M = pin.crba(model, data, q)
print("torque_from_EOM = ", M @ a + nle - JtF)
print("torque_from_rnea= ", pin.rnea(model, data, q, v, a, f_ext_local))
```

Output:

```text
torque_from_EOM =  [ 0.    0.   56.54  0.   -0.53  0.   -0.25 -0.25]
torque_from_rnea=  [ 0.    0.   28.27  0.   -1.38  0.   -0.68 -0.68]
```

The results from EoM and RNEA are inconsistent.

Model file (`model.urdf`):

```xml
<?xml version="1.0" encoding="utf-8"?>
<robot name="mybot">
  <link name="body">
    <inertial>
      <origin xyz="0.00037 0 0.01792" rpy="0 0 0" />
      <mass value="4.935" />
      <inertia ixx="0.036517" ixy="-3.49E-05" ixz="1.11E-05" iyy="0.0058391" iyz="6.41E-06"
        izz="0.039681" />
    </inertial>
  </link>
  <link name="left_link">
    <inertial>
      <origin xyz="0.064144 0 -0.047389" rpy="0 0 0" />
      <mass value="0.394" />
      <inertia ixx="0.000169" ixy="-1E-09" ixz="0.000195" iyy="0.0019044" iyz="0" izz="0.0018657" />
    </inertial>
  </link>
  <joint name="left_joint" type="revolute">
    <origin xyz="0 0.1 -0.5" rpy="0 0 0" />
    <parent link="body" />
    <child link="left_link" />
    <axis xyz="0 1 0" />
    <limit lower="-0.7" upper="0.7" effort="200" velocity="23" />
  </joint>
  <joint name="left_front_joint" type="fixed">
    <parent link="left_link" />
    <child link="lf_contact" />
    <origin xyz=" 0.10 0 -0.06" />
  </joint>
  <joint name="left_back_joint" type="fixed">
    <parent link="left_link" />
    <child link="lb_contact" />
    <origin xyz="-0.03 0 -0.06" />
  </joint>
  <link name="lf_contact">
    <inertial>
      <mass value="0.01" />
      <inertia ixx="1.6e-06" ixy="-0.0" ixz="-0.0" iyy="1.6e-06" iyz="-0.0" izz="1.6e-06" />
    </inertial>
  </link>
  <link name="lb_contact">
    <inertial>
      <mass value="0.01" />
      <inertia ixx="1.6e-06" ixy="-0.0" ixz="-0.0" iyy="1.6e-06" iyz="-0.0" izz="1.6e-06" />
    </inertial>
  </link>
  <link name="right_link">
    <inertial>
      <origin xyz="0.064144 0 -0.047389" rpy="0 0 0" />
      <mass value="0.394" />
      <inertia ixx="0.000169" ixy="1E-09" ixz="0.000195" iyy="0.0019044" iyz="0" izz="0.0018657" />
    </inertial>
  </link>
  <joint name="right_joint" type="revolute">
    <origin xyz="0 -0.1 -0.5" rpy="0 0 0" />
    <parent link="body" />
    <child link="right_link" />
    <axis xyz="0 1 0" />
    <limit lower="-0.7" upper="0.7" effort="200" velocity="23" />
  </joint>
  <joint name="right_front_joint" type="fixed">
    <parent link="right_link" />
    <child link="rf_contact" />
    <origin xyz=" 0.10 0 -0.06" />
  </joint>
  <joint name="right_back_joint" type="fixed">
    <parent link="right_link" />
    <child link="rb_contact" />
    <origin xyz="-0.03 0 -0.06" />
  </joint>
  <link name="rf_contact">
    <inertial>
      <mass value="0.01" />
      <inertia ixx="1.6e-06" ixy="-0.0" ixz="-0.0" iyy="1.6e-06" iyz="-0.0" izz="1.6e-06" />
    </inertial>
  </link>
  <link name="rb_contact">
    <inertial>
      <mass value="0.01" />
      <inertia ixx="1.6e-06" ixy="-0.0" ixz="-0.0" iyy="1.6e-06" iyz="-0.0" izz="1.6e-06" />
    </inertial>
  </link>
</robot>
```

jcarpent, 2023-04-11:
Here is the solution. You've forgotten two important things: calling `computeJointJacobians` before calling `getFrameJacobian` and summing the contributions of contact forces when evaluating `f_ext_local[jointIndex]`.

```
import pinocchio as pin
import numpy as np
import os

np.set_printoptions(precision=2, suppress=True, linewidth=200)

urdf_file =  "model.urdf"
jc = pin.JointModelComposite()
jc_xyz = jc.addJoint(pin.JointModelTranslation())
jc_xyz_zyx = jc_xyz.addJoint(pin.JointModelSphericalZYX())
model, collision_model, visual_model = pin.buildModelsFromUrdf(
    urdf_file, root_joint=jc_xyz_zyx
)
data = model.createData()

q = np.array([0, 0, 0.56, 0, 0, 0, 0, 0])
v = np.zeros(model.nv)
a = np.zeros(model.nv)
pin.computeJointJacobians(model, data, q)
normal_force = abs(pin.computeTotalMass(model) * model.gravity.linear[2] / 4)
contact_force_in_world = [0.0, 0.0, normal_force] * 4
contact_frame = ["lf_contact", "lb_contact", "rf_contact", "rb_contact"]

f_ext_local = [pin.Force.Zero() for j in range(model.njoints)]
JtF = np.zeros(model.nq)

for i in range(len(contact_frame)):
    frameIndex = model.getFrameId(contact_frame[i])
    jointIndex = model.frames[frameIndex].parentJoint
    print("jointIndex: ", jointIndex)
    translation = model.frames[frameIndex].placement.translation
    rotation = data.oMi[jointIndex].rotation
    contactForce = rotation.T @ contact_force_in_world[3 * i : 3 * i + 3]
    f_ext_local[jointIndex].linear += contactForce
    f_ext_local[jointIndex].angular += np.cross(translation, contactForce)

    J = pin.getFrameJacobian(
        model, data, frameIndex, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED
    )[:3, :]
    JtF = JtF + J.T @ contact_force_in_world[3 * i : 3 * i + 3]

nle = pin.nonLinearEffects(model, data, q, v)
M = pin.crba(model, data, q)
print("torque_from_EOM = ", M @ a + nle - JtF)
print("torque_from_rnea= ", pin.rnea(model, data, q, v, a, f_ext_local))
```

closed 2023-04-11 by jcarpent