stack-of-tasks/pinocchio #1917: pin.buildModelsFromUrdf for Hello Stretch robot

arjung128, 2023-04-10:
Hello, I obtained the URDF for the Hello Stretch robot and am trying to use `pin.buildModelsFromUrdf` to obtain a model for it, but am running into the following issue:

```
>>> import pinocchio as pin
>>> model, geom_model, visual_model = pin.buildModelsFromUrdf('stretch/stretch-re2-2025/exported_urdf/stretch.urdf', 'stretch/stretch-re2-2025/exported_urdf/', pin.JointModelFreeFlyer())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "~/venv-ompl/lib/python3.8/site-packages/pinocchio/shortcuts.py", line 45, in buildModelsFromUrdf
    geom_model = pin.buildGeomFromUrdf(model, filename, geometry_type, package_dirs)
ValueError: Could not load resource ./meshes/base_link.STL
Unable to open file "./meshes/base_link.STL".
Hint: the mesh directory may be wrong.
```

The directory I passed into the function (`stretch/stretch-re2-2025/exported_urdf/`) contains a folder called `meshes` which contains `base_link.STL`, amongst many other `.STL` files. Any ideas for how I can go about debugging this? Thanks in advance!

jcarpent, 2023-04-10:
Could you please share the model or any link to the description ?

fabinsch, 2023-04-10:
Which version of Pinocchio are you using ? 
You can do `print(pin.__version__)` to check in your code .

arjung128, 2023-04-10:
Thanks so much for your prompt responses!

Here is the `exported_urdf` folder from my original code snippet: [exported_urdf.zip](https://github.com/stack-of-tasks/pinocchio/files/11188906/exported_urdf.zip)

And I'm using Pinocchio version 2.5.6.

Please let me know if there's any additional information I can provide.

fabinsch, 2023-04-10:
I think upgrading Pinocchio to the latest version could fix your issue. In the PR [here](https://github.com/stack-of-tasks/pinocchio/pull/1748), we fixed some relative path loading issues related to meshes. It was integrated in Pinocchio 2.6.11 in September '22.

jcarpent, 2023-04-10:
I confirm that everything works well with the latest release of Pinocchio (version 2.6.17) installed via conda.