stack-of-tasks/pinocchio #1932: Unsure about usage of pin.updateFramePlacements(model, data)

arjung128, 2023-04-22:
Hello, I would like to create a new frame based on two existing frames. Namely, `joint_gripper_fingertip_right` and `joint_gripper_fingertip_left` exist, and I would like to keep track of the center point between these two frames, i.e. the average position of the fingertips. 

Here is my code:

```
model, geom_model, visual_model = pin.buildModelsFromUrdf('stretch/stretch-re2-2025/exported_urdf/stretch.urdf', 'stretch/stretch-re2-2025/exported_urdf/')

frame = pin.Frame(
    name="joint_gripper_fingertip_center",
    parent_joint=11, # model.parents[-2] = model.parents[-3] = 11 (parents of the right and left fingertip joints)
    parent_frame=81, # model.frames[81].name = 'joint_gripper_fingertip_right'
    placement=pin.SE3(
        rotation=np.eye(3),
        translation=np.array([0, 0, 0]),
    ),
    type=pin.FrameType.OP_FRAME,
)
model.addFrame(frame)

data, geom_data, visual_data = pin.createDatas(model, geom_model, visual_model)

q = pin.randomConfiguration(model)
pin.forwardKinematics(model, data, q)
pin.framesForwardKinematics(model, data, q)

# set the translation for joint_gripper_fingertip_center to be the average of joint_gripper_fingertip_right and joint_gripper_fingertip_left
data.oMf[model.getFrameId('joint_gripper_fingertip_center')].translation = (data.oMf[model.getFrameId('joint_gripper_fingertip_right')].translation + data.oMf[model.getFrameId('joint_gripper_fingertip_left')].translation) / 2
```

However, this doesn't seem to affect `pin.computeFrameJacobian(model,data,q,model.getFrameId('joint_gripper_fingertip_center'))`, and calling `pin.updateFramePlacements(model, data)` reverts the `joint_gripper_fingertip_center` frame to a completely different pose. Any ideas what I'm missing to get an accurate Jacobian (using pin.computeFrameJacobian) and frame placement (using pin.updateFramePlacements) after manually setting the translation to be the midpoint between the two finger joints?

Any help will be much appreciated, thanks in advance!

jcarpent, 2023-04-22:
@arjung128 Please do not open an issue for Question, and rather use the Discussions workflow for that.
In addition, your request is related to #1927, so please keep the same track; otherwise, it will be hard for others to follow the thread properly!

You should update the frame.placement quantity according to the middle position between the two other frames, and this depends on the config vector of course. So, you should:

1.  do an updateFramePlacements(model,data,q)  // it will update all the oMf quantities
2. update frame.placement of the frame with name "joint_gripper_fingertip_center"
3. updateFramePlacement(model,data,model.getFrameId("joint_gripper_fingertip_center"))
4. compute your frame Jacobian.

Hope it is clear.

arjung128, 2023-04-22:
Hello @jcarpent,

Thanks for your prompt response! And I apologize, I will make use of the discussion feature from now onwards.

I'm not sure I fully understood your suggestion. I replaced the last line of my code snippet above with the following block of code:

```
# save jacobian (for comparison) and average of left / right finger positions
joint_gripper_fingertip_center_jacobian = pin.computeFrameJacobian(model,data,q,model.getFrameId('joint_gripper_fingertip_center')).copy()
joint_gripper_fingertip_center_new_pos  = ((data.oMf[model.getFrameId('joint_gripper_fingertip_right')].translation + data.oMf[model.getFrameId('joint_gripper_fingertip_left')].translation) / 2).copy()

# do an updateFramePlacements(model,data,q) // it will update all the oMf quantities
pin.updateFramePlacements(model,data)
# update frame.placement of the frame with name "joint_gripper_fingertip_center"
data.oMf[model.getFrameId('joint_gripper_fingertip_center')].translation = joint_gripper_fingertip_center_new_pos
# updateFramePlacement(model,data,model.getFrameId("joint_gripper_fingertip_center"))
pin.updateFramePlacement(model,data,model.getFrameId("joint_gripper_fingertip_center"))
# compute your frame Jacobian.
joint_gripper_fingertip_center_jacobian_new = pin.computeFrameJacobian(model,data,q,model.getFrameId('joint_gripper_fingertip_center')).copy()

# did the jacobian change? --> no, as the two are equal
print((joint_gripper_fingertip_center_jacobian == joint_gripper_fingertip_center_jacobian_new).all())

# did the position of joint_gripper_fingertip_center get updated to the desired position? --> False.
print((data.oMf[model.getFrameId('joint_gripper_fingertip_center')].translation == joint_gripper_fingertip_center_new_pos).all())
```

As shown, the jacobian does not seem to change, and the position did not seem to get updated to the desired position. Any ideas what I am doing wrong, and what I can do to implement this new frame correctly?

Thanks once again for all your help and support!

jcarpent, 2023-04-22:
```python
# do an updateFramePlacements(model,data,q) // it will update all the oMf quantities
pin.computeJointJacobians(model,data,q)
pin.updateFramePlacements(model,data)
# update frame.placement of the frame with name "joint_gripper_fingertip_center"
data.oMf[model.getFrameId('joint_gripper_fingertip_center')].translation = joint_gripper_fingertip_center_new_pos
# compute your frame Jacobian.
joint_gripper_fingertip_center_jacobian_new = pin.getFrameJacobian(model,data,q,model.getFrameId('joint_gripper_fingertip_center')).copy()
```

arjung128, 2023-04-22:
Hello @jcarpent,

By the last `pin.getFrameJacobian`, did you mean `pin.computeFrameJacobian`? With `pin.getFrameJacobian`, I am getting an error, but it seems to work with `pin.computeFrameJacobian`.

With `pin.computeFrameJacobian`, the position seems to be accurately updated -- thank you! But the jacobian still seems to be the same as before -- is this expected?

Thanks for your prompt help!

jcarpent, 2023-04-22:
For `pin.getFrameJacobian`, you need to also specify the reference Frame in which the Jacobian is expressed (LOCAL, WORLD, or LOCAL_WORLD_ALIGNED).

closed 2023-04-22 by jcarpent

arjung128, 2023-04-23:
@jcarpent Thanks a lot for your response! With your suggestion, I am able to do IK for an existing frame, but somehow not for the new frame I added. My suspicion is that I am not updating the new frame correctly.

Here is my code:

```
# imports
import pinocchio as pin
from pinocchio.visualize import MeshcatVisualizer
import numpy as np
from scipy.spatial.transform import Rotation as R
from numpy.linalg import norm, solve

# initialize model and add frame
model, geom_model, visual_model = pin.buildModelsFromUrdf('stretch/stretch-re2-2025/exported_urdf/stretch.urdf', 'stretch/stretch-re2-2025/exported_urdf/')
frame = pin.Frame(
    name="joint_gripper_fingertip_center",
    parent_joint=11, # model.parents[-2] = model.parents[-3] = 11 (parents of the right and left fingertip joints)
    parent_frame=81, # model.frames[81].name = 'joint_gripper_fingertip_right'
    placement=pin.SE3(
        rotation=np.eye(3),
        translation=np.array([0, 0, 0]),
    ),
    type=pin.FrameType.OP_FRAME,
)
model.addFrame(frame)
data, geom_data, visual_data = pin.createDatas(model, geom_model, visual_model)

# IK function
def inverse_kinematics_stretch(pos, rot, frame, eps=1e-3, DT=1e-1, damp=1e-12):
    q = pin.neutral(model)
    oMdes = pin.SE3(rot, pos)  
    low = np.array([-1.01, -1.01, -3.9, -1.53, 0., 0., 0., 0., 0., -1.75, -1.57, -3.14, -0.6, -0.6, -1.01, -1.01]) + 1e-5
    high = np.array([1.01, 1.01, 1.5, 0.79, 1.1, 0.13, 0.13, 0.13, 0.13, 4., 0.56, 3.14, 0.6 , 0.6 , 1.01, 1.01]) - 1e-5
    i=0
    while True:
        pin.forwardKinematics(model, data, q)
        pin.framesForwardKinematics(model, data, q)
        # do an updateFramePlacements(model,data,q) // it will update all the oMf quantities
        pin.computeJointJacobians(model,data,q)
        pin.updateFramePlacements(model,data)

        if frame == 'right':
            frame_id = model.getFrameId('joint_gripper_fingertip_right')
        else:
            frame_id = model.getFrameId('joint_gripper_fingertip_center')
            # update frame.placement of the frame with name "joint_gripper_fingertip_center"
            data.oMf[frame_id].translation = (data.oMf[model.getFrameId('joint_gripper_fingertip_right')].translation + data.oMf[model.getFrameId('joint_gripper_fingertip_left')].translation) / 2
            data.oMf[frame_id].rotation = data.oMf[65].rotation # R.from_euler('xyz', [-3.11745943, 0.01096767, 0.03274201]).as_matrix()

        dMi = oMdes.actInv(data.oMf[frame_id])
        err = pin.log(dMi).vector
        if norm(err) < eps:
            success = True
            break
        if i >= 5000:
            success = False
            break

        # compute your frame Jacobian.
        J = pin.getFrameJacobian(model,data,frame_id,pin.LOCAL)
        v = - J.T.dot(solve(J.dot(J.T) + damp * np.eye(6), err))
        q = pin.integrate(model,q,v*DT)
        q = np.clip(q, low, high) # enforce joint limits
        q[:4] = np.array([1., 0., 0., 0.]) # fix other unused joints
        i += 1

    center_trans = (data.oMf[model.getFrameId('joint_gripper_fingertip_right')].translation + data.oMf[model.getFrameId('joint_gripper_fingertip_left')].translation) / 2
    center_rot = data.oMf[65].rotation # R.from_matrix(data.oMf[65].rotation).as_euler('xyz')

    print(norm(err))
[... 57 lines cut ...]
# do an updateFramePlacements(model,data,q) // it will update all the oMf quantities
pin.computeJointJacobians(model,data,q)
pin.updateFramePlacements(model,data)

center_trans_neutral = (data.oMf[model.getFrameId('joint_gripper_fingertip_right')].translation + data.oMf[model.getFrameId('joint_gripper_fingertip_left')].translation) / 2
center_rot_neutral = data.oMf[65].rotation # R.from_matrix(data.oMf[65].rotation).as_euler('xyz')

q, success, _, _ = inverse_kinematics_stretch(center_trans_neutral, center_rot_neutral, 'center')

# OUTPUT:
# 0.0
```

However, if I perturb the target position by just a little bit, IK is unable to find a solution:

```
q, success, _, _ = inverse_kinematics_stretch(center_trans_neutral + np.array([0., -0.01, 0.]), center_rot_neutral, 'center')

# OUTPUT:
# 0.012159574260018216
```

Any ideas what I might be doing wrong here?

Any help will be much appreciated, thanks in advance!