stack-of-tasks/pinocchio #1938: Strange error after conda installation: module 'pinocchio.pinocchio_pywrap' has no attribute 'getNumpyType'

Flo-Wo, 2023-04-26:
Hello guys,

after using pinocchio for quite a while (thank you so much for this amazing tool!), I needed to reinstall the package on a different machine. After following the instructions and installation pinocchio via conda, the ``RobotWrapper`` does not seem to work for me. Originally I tried to install pinocchio via the installation of crocoddyl, but to test the error, I installed pinocchio in a plain environment and tried to open a simple urdf-File (the file is not corrupted, it already worked with pinocchio on a different machine).

The error is the following:
```python
>>> from pinocchio import RobotWrapper
>>> test = RobotWrapper.BuildFromURDF("rot_wam_pend_temp_0.3.urdf")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/florianwolf/miniconda3/envs/pin_test/lib/python3.11/site-packages/pinocchio/robot_wrapper.py", line 16, in BuildFromURDF
    robot = RobotWrapper()
            ^^^^^^^^^^^^^^
  File "/Users/florianwolf/miniconda3/envs/pin_test/lib/python3.11/site-packages/pinocchio/robot_wrapper.py", line 32, in __init__
    self.v0 = utils.zero(self.nv)
              ^^^^^^^^^^^^^^^^^^^
  File "/Users/florianwolf/miniconda3/envs/pin_test/lib/python3.11/site-packages/pinocchio/utils.py", line 40, in zero
    if pin.getNumpyType()==np.matrix:
       ^^^^^^^^^^^^^^^^
AttributeError: module 'pinocchio.pinocchio_pywrap' has no attribute 'getNumpyType'
```
I also tried to switch to python 3.9 but that didn't help. The environment contains the following packages

```shell
# packages in environment at /Users/florianwolf/miniconda3/envs/pin_test_2:
#
# Name                    Version                   Build  Channel
assimp                    5.2.5                h276577b_0    conda-forge
boost                     1.78.0          py311h5c64717_4    conda-forge
boost-cpp                 1.78.0               h9ed8d21_3    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
ca-certificates           2022.12.7            h4653dfc_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
eigen                     3.4.0                hc021e02_0    conda-forge
eigenpy                   3.0.0           py311h6857c3b_1    conda-forge
hpp-fcl                   2.3.1           py311he75a86d_1    conda-forge
icu                       72.1                 he12128b_0    conda-forge
libblas                   3.9.0           16_osxarm64_openblas    conda-forge
libcblas                  3.9.0           16_osxarm64_openblas    conda-forge
libcxx                    16.0.2               h4653b0c_0    conda-forge
libexpat                  2.5.0                hb7217d7_1    conda-forge
libffi                    3.4.2                h3422bc3_5    conda-forge
libgfortran               5.0.0           12_2_0_hd922786_31    conda-forge
libgfortran5              12.2.0              h0eea778_31    conda-forge
liblapack                 3.9.0           16_osxarm64_openblas    conda-forge
libopenblas               0.3.21          openmp_hc731615_3    conda-forge
libsqlite                 3.40.0               h76d750c_1    conda-forge
libzlib                   1.2.13               h03a7124_4    conda-forge
llvm-openmp               16.0.2               h1c12783_0    conda-forge
ncurses                   6.3                  h07bb92c_1    conda-forge
numpy                     1.24.3          py311hb8f3215_0    conda-forge
octomap                   1.9.8                hffc8910_0    conda-forge
openssl                   3.1.0                h53f4e23_2    conda-forge
pinocchio                 2.6.17          py311ha2fd7aa_2    conda-forge
pip                       23.1.2             pyhd8ed1ab_0    conda-forge
python                    3.11.3          h1456518_0_cpython    conda-forge
python_abi                3.11                    3_cp311    conda-forge
qhull                     2020.2               hc021e02_2    conda-forge
readline                  8.2                  h92ec313_1    conda-forge
setuptools                67.7.2             pyhd8ed1ab_0    conda-forge
tinyxml                   2.6.2                h260d524_2    conda-forge
tk                        8.6.12               he1e0b03_0    conda-forge
tzdata                    2023c                h71feb2d_0    conda-forge
urdfdom                   3.1.1                h19b2bf3_0    conda-forge
urdfdom_headers           1.1.0                hf86a087_0    conda-forge
wheel                     0.40.0             pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h57fd34a_0    conda-forge
zlib                      1.2.13               h03a7124_4    conda-forge
zstd                      1.5.2                hf913c23_6    conda-forge
```

Any help is appreciated, thank you very much!

jcarpent, 2023-04-26:
Hi @Flo-Wo,
The issue comes from the recent release of EigenPy 3.0, which suppresses the support of numpy.matrix. 
I still need to update software which rely on EigenPy to account for this change in the conda workflow.

In the meantime, you can simply do `conda install -c conda-forge eigenpy==2.9.2`

Flo-Wo, 2023-04-26:
Thank you so much for the quick response and your help. That solved the issue, works like a charm.

closed 2023-04-26 by Flo-Wo

IbrahimTeej, 2023-10-09:
Hello All,

I am currently using Ubuntu 22.04 with python 3.10 version. I successfully installed pinocchio through robotpkg using the sudo apt. However, i get a similar error attributed to ''module 'pinocchio.pinocchio_pywrap_default' has no attribute 'getNumpyType' ''

Kindly see enclosed below:
itijjani@ITIJJANI-P14-U:~/DEVEL/exo_walk$ ipython3 -i static_gait_exper3_IB.py -- --dir_robot 4steps_IBRAHIM
Python 3.10.12 (main, Jun 11 2023, 05:26:28) [GCC 11.4.0]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.31.1 -- An enhanced Interactive Python. Type '?' for help.
Loading the viewer ...
+ /opt/openrobots/bin/gepetto-gui
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
~/DEVEL/exo_walk/static_gait_exper3_IB.py in <module>
     84 
     85 # Load the full model
---> 86 robot = RobotWrapper.BuildFromURDF(modelPath + URDF_SUBPATH, [modelPath], pinocchio.JointModelFreeFlyer())
     87 
     88 #****************************************************************************

~/install/lib/python3/dist-packages/pinocchio/robot_wrapper.py in BuildFromURDF(filename, package_dirs, root_joint, verbose, meshLoader)
     14     @staticmethod
     15     def BuildFromURDF(filename, package_dirs=None, root_joint=None, verbose=False, meshLoader=None):
---> 16         robot = RobotWrapper()
     17         robot.initFromURDF(filename, package_dirs, root_joint, verbose, meshLoader)
     18         return robot

~/install/lib/python3/dist-packages/pinocchio/robot_wrapper.py in __init__(self, model, collision_model, visual_model, verbose)
     41         self.data, self.collision_data, self.visual_data = createDatas(model,collision_model,visual_model)
     42 
---> 43         self.v0 = utils.zero(self.nv)
     44         self.q0 = pin.neutral(self.model)
     45 

~/install/lib/python3/dist-packages/pinocchio/utils.py in zero(n)
     38 
     39 def zero(n):
---> 40     if pin.getNumpyType()==np.matrix:
     41         return np.matrix(np.zeros([n, 1] if isinstance(n, int) else n))
     42     else:

AttributeError: module 'pinocchio.pinocchio_pywrap_default' has no attribute 'getNumpyType'

In [1]: pinocchio.__version__
Out[1]: '2.9.1'

In [2]: 

I have crocoddyl version 2.0.1, and eigenpy 3.1.1. Since eigenpy3.0 suppresses numpy.matrix support, i also could not downgrade my eigenpy using ''sudo apt-get install eigenpy==2.9.2''. It says ''unable to locate package eigenpy''

Any help would be helpful. Thanks

Best regards,
Ibrahim Tijjani

IbrahimTeej, 2023-10-09:
Hello, @nim65s.

Thank you for your message and also pointing me to the right direction.
I actually thought this issue is similar to mine, that's the reason for wanting to downgrade but now it's clear.
I will format my code as you suggested. Please, could you point me to the flavored markdown? 
I am not sure i know exactly the exact code file i need to modify mine to, and perhaps on the master branch? Thank you

Best regards,
Ibrahim