stack-of-tasks/pinocchio #1940: Issue reading URDF files in pybindings

kingjin94, 2023-04-28:
This might be related to / was hidden by #1938.

At the moment it seems that pinocchio's python bindings are unable to load URDFs. The error is `AttributeError: module 'pinocchio.pinocchio_pywrap' has no attribute 'getNumpyType'`

A simple script to demonstrate (using timor-python as a source for a simple demo URDF):

```
docker run -ti python:3.10 bash
pip install timor-python
python
import pinocchio as pin
from timor.utilities.file_locations import robots
panda_urdf = robots['panda'].joinpath('urdf').joinpath('panda.urdf')
pin.RobotWrapper.BuildFromURDF(panda_urdf, panda_urdf.parent)
```

kingjin94, 2023-04-28:
Even shorter:
```

docker run -ti python:3.10 bash
pip install pinocchio
python
import pinocchio as pin
pin.RobotWrapper.BuildFromURDF("notImportantFile.urdf", "/dummy/package/dir/")
```
 Will result in the same error.

stephane-caron, 2023-04-28:
Running into the same issue in this workflow: https://github.com/tasts-robots/pink/actions/runs/4830671896/jobs/8607169482?pr=50

```
  AttributeError: module 'pinocchio.pinocchio_pywrap' has no attribute 'getNumpyType'
```

fabinsch, 2023-04-28:
fix is on the way #1941

jcarpent, 2023-04-29:
A new release in in the pipe and should be available via conda in few hours.

jcarpent, 2023-05-01:
The release update is in the pipe: https://github.com/conda-forge/pinocchio-feedstock/pull/97

closed 2023-05-01 by jcarpent