stack-of-tasks/pinocchio #1948: Error finding parent and previousFrame

jameswzhu, 2023-05-03:
Hello, I have a fresh install of the pinocchio3-preview branch. I am trying to run the `simulation-contact-dynamics.py` example, and as I build the model from the urdf, I find that at line 66, `contact_model = pin.RigidConstraintModel(pin.ContactType.CONTACT_6D,frame.parent,frame.placement)`, the `frame.parent` and `frame.previousFrame` contain the error: `Traceback (most recent call last):
  File "<string>", line 1, in <module>
Boost.Python.ArgumentError: Python argument types in
    None.None(Frame)
did not match C++ signature:
    None(pinocchio::ModelItem<pinocchio::FrameTpl<double, 0> > {lvalue})`. Let me know if I have done something wrong with the installation here? Thanks!

jcarpent, 2023-05-03:
Could you provide the steps to reproduce your issue? Could you provide a docker image?

jameswzhu, 2023-05-04:
I uploaded the docker image here: [(https://hub.docker.com/r/zhuj1198/pinocchio)]. Thank you!

jameswzhu, 2023-05-11:
Hello, have you had a chance to take a look at this?

jcarpent, 2023-05-11:
@MegMll Could you check this issue?

MegMll, 2023-05-11:
Hello, 

I checked your installation in the docker and it seems that you compiled pinocchio without collision support, which is necessary for contact dynamics.
So I would suggest, you rebuild pinocchio by adding collision support (with `-DBUILD_WITH_COLLISION_SUPPORT=ON`) which requires you to first compile the hpp-fcl library.
I tried to compile it in your docker, but it seems that you have some missing dependencies.

closed 2023-05-14 by jcarpent

jameswzhu, 2023-05-16:
Looks like it's working, thanks!