stack-of-tasks/pinocchio #1950: pb compiling api

sancelot, 2023-05-04:
Hi, 
I have a problem compiling the api, gcc reports :
```
In file included from /home/robot/pinocchio/build/include/pinocchio/bindings/python/multibody/liegroups.hpp:12:0,
                 from /home/robot/pinocchio/bindings/python/multibody/expose-liegroups.cpp:6:
/home/robot/pinocchio/build/include/pinocchio/multibody/liegroup/liegroup-generic.hpp: In instantiation of ‘pinocchio::LieGroupGenericTpl<LieGroupCollection>::LieGroupGenericTpl(const pinocchio::LieGroupGenericTpl<LieGroupCollection>&) [with LieGroupCollection = pinocchio::LieGroupCollectionDefaultTpl<double>]’:
/usr/include/c++/7/ext/new_allocator.h:136:4:   required from ‘void __gnu_cxx::new_allocator<_Tp>::construct(_Up*, _Args&& ...) [with _Up = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >; _Args = {const pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double, 0> >&}; _Tp = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >]’
/usr/include/c++/7/bits/alloc_traits.h:243:4:   required from ‘static std::_Require<typename std::allocator_traits<_Alloc>::__construct_helper<_Tp, _Args>::type> std::allocator_traits<_Alloc>::_S_construct(_Alloc&, _Tp*, _Args&& ...) [with _Tp = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >; _Args = {const pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double, 0> >&}; _Alloc = Eigen::aligned_allocator<pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> > >; std::_Require<typename std::allocator_traits<_Alloc>::__construct_helper<_Tp, _Args>::type> = void]’
/usr/include/c++/7/bits/alloc_traits.h:344:16:   required from ‘static decltype (_S_construct(__a, __p, (forward<_Args>)(std::allocator_traits::construct::__args)...)) std::allocator_traits<_Alloc>::construct(_Alloc&, _Tp*, _Args&& ...) [with _Tp = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >; _Args = {const pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double, 0> >&}; _Alloc = Eigen::aligned_allocator<pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> > >; decltype (_S_construct(__a, __p, (forward<_Args>)(std::allocator_traits::construct::__args)...)) = <type error>]’
/usr/include/c++/7/bits/stl_vector.h:943:30:   required from ‘void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >; _Alloc = Eigen::aligned_allocator<pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> > >; std::vector<_Tp, _Alloc>::value_type = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >]’
/home/robot/pinocchio/build/include/pinocchio/multibody/liegroup/cartesian-product-variant.hxx:17:3:   required from ‘void pinocchio::CartesianProductOperationVariantTpl<Scalar, Options, LieGroupCollectionTpl>::append(const LieGroupGeneric&) [with _Scalar = double; int _Options = 0; LieGroupCollectionTpl = pinocchio::LieGroupCollectionDefaultTpl; pinocchio::CartesianProductOperationVariantTpl<Scalar, Options, LieGroupCollectionTpl>::LieGroupGeneric = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >]’
/home/robot/pinocchio/build/include/pinocchio/multibody/liegroup/cartesian-product-variant.hpp:64:13:   required from ‘pinocchio::CartesianProductOperationVariantTpl<Scalar, Options, LieGroupCollectionTpl>::CartesianProductOperationVariantTpl(const LieGroupGeneric&) [with _Scalar = double; int _Options = 0; LieGroupCollectionTpl = pinocchio::LieGroupCollectionDefaultTpl; pinocchio::CartesianProductOperationVariantTpl<Scalar, Options, LieGroupCollectionTpl>::LieGroupGeneric = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >]’
/home/robot/pinocchio/bindings/python/multibody/expose-liegroups.cpp:29:58:   required from here
/home/robot/pinocchio/build/include/pinocchio/multibody/liegroup/liegroup-generic.hpp:46:5: warning: base class ‘struct pinocchio::LieGroupBase<pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> > >’ should be explicitly initialized in the copy constructor [-Wextra]
     LieGroupGenericTpl(const LieGroupGenericTpl & lg_generic)
     ^~~~~~~~~~~~~~~~~~
In file included from /opt/kdl/include/boost/variant.hpp:17:0,
                 from /home/robot/pinocchio/build/include/pinocchio/multibody/liegroup/liegroup-collection.hpp:13,
                 from /home/robot/pinocchio/build/include/pinocchio/multibody/liegroup/cartesian-product-variant.hpp:9,
                 from /home/robot/pinocchio/build/include/pinocchio/bindings/python/multibody/liegroups.hpp:11,
                 from /home/robot/pinocchio/bindings/python/multibody/expose-liegroups.cpp:6:
/opt/kdl/include/boost/variant/variant.hpp: In instantiation of ‘boost::variant<T0, TN>::variant(const T&) [with T = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >; T0_ = pinocchio::SpecialOrthogonalOperationTpl<2, double, 0>; TN = {pinocchio::SpecialOrthogonalOperationTpl<3, double, 0>, pinocchio::SpecialEuclideanOperationTpl<2, double, 0>, pinocchio::SpecialEuclideanOperationTpl<3, double, 0>, pinocchio::VectorSpaceOperationTpl<1, double, 0>, pinocchio::VectorSpaceOperationTpl<2, double, 0>, pinocchio::VectorSpaceOperationTpl<3, double, 0>, pinocchio::VectorSpaceOperationTpl<-1, double, 0>}]’:
/home/robot/pinocchio/build/include/pinocchio/multibody/liegroup/liegroup-generic.hpp:47:22:   required from ‘pinocchio::LieGroupGenericTpl<LieGroupCollection>::LieGroupGenericTpl(const pinocchio::LieGroupGenericTpl<LieGroupCollection>&) [with LieGroupCollection = pinocchio::LieGroupCollectionDefaultTpl<double>]’
/usr/include/c++/7/ext/new_allocator.h:136:4:   required from ‘void __gnu_cxx::new_allocator<_Tp>::construct(_Up*, _Args&& ...) [with _Up = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >; _Args = {const pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double, 0> >&}; _Tp = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >]’
/usr/include/c++/7/bits/alloc_traits.h:243:4:   required from ‘static std::_Require<typename std::allocator_traits<_Alloc>::__construct_helper<_Tp, _Args>::type> std::allocator_traits<_Alloc>::_S_construct(_Alloc&, _Tp*, _Args&& ...) [with _Tp = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >; _Args = {const pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double, 0> >&}; _Alloc = Eigen::aligned_allocator<pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> > >; std::_Require<typename std::allocator_traits<_Alloc>::__construct_helper<_Tp, _Args>::type> = void]’
/usr/include/c++/7/bits/alloc_traits.h:344:16:   required from ‘static decltype (_S_construct(__a, __p, (forward<_Args>)(std::allocator_traits::construct::__args)...)) std::allocator_traits<_Alloc>::construct(_Alloc&, _Tp*, _Args&& ...) [with _Tp = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >; _Args = {const pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double, 0> >&}; _Alloc = Eigen::aligned_allocator<pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> > >; decltype (_S_construct(__a, __p, (forward<_Args>)(std::allocator_traits::construct::__args)...)) = <type error>]’
/usr/include/c++/7/bits/stl_vector.h:943:30:   required from ‘void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >; _Alloc = Eigen::aligned_allocator<pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> > >; std::vector<_Tp, _Alloc>::value_type = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >]’
/home/robot/pinocchio/build/include/pinocchio/multibody/liegroup/cartesian-product-variant.hxx:17:3:   required from ‘void pinocchio::CartesianProductOperationVariantTpl<Scalar, Options, LieGroupCollectionTpl>::append(const LieGroupGeneric&) [with _Scalar = double; int _Options = 0; LieGroupCollectionTpl = pinocchio::LieGroupCollectionDefaultTpl; pinocchio::CartesianProductOperationVariantTpl<Scalar, Options, LieGroupCollectionTpl>::LieGroupGeneric = pinocchio::LieGroupGenericTpl<pinocchio::LieGroupCollectionDefaultTpl<double> >]’
[... cut ...]
bindings/python/CMakeFiles/pinocchio_pywrap.dir/build.make:470: recipe for target 'bindings/python/CMakeFiles/pinocchio_pywrap.dir/multibody/expose-liegroups.cpp.o' failed
make[2]: *** [bindings/python/CMakeFiles/pinocchio_pywrap.dir/multibody/expose-liegroups.cpp.o] Error 1
make[2]: Leaving directory '/home/robot/pinocchio/build'
CMakeFiles/Makefile2:1786: recipe for target 'bindings/python/CMakeFiles/pinocchio_pywrap.dir/all' failed
make[1]: *** [bindings/python/CMakeFiles/pinocchio_pywrap.dir/all] Error 2
make[1]: Leaving directory '/home/robot/pinocchio/build'
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2
```

jcarpent, 2023-05-04:
Do you need a so old version of boost?

sancelot, 2023-05-10:
> Closing as there is no reply.
My target system uses opensuse 42.3 for  some reasons I can not offer to upgrade it as I wanted since it is a realtime system .
and unfortunately I can not use python bindings because I have only python 3.4 available.
but I managed to compile using boost 1.81 .