stack-of-tasks/pinocchio #1951: Support to continuous joints and transmission

Cfather, 2023-05-04:
Hi,

I'm not sure if pinocchio supports the following features from the urdf:

1. [continuous](http://wiki.ros.org/urdf/XML/joint) joint type.I tried some robots with 'continuous' joint types in their urdf and it seems that pinocchio can not recognize those joints properly.
2. [transmission](http://wiki.ros.org/urdf/XML/Transmission), or in other words, [armature](https://en.wikipedia.org/wiki/Armature_(electrical)). The motor will create a resistance force = armature * acceleration. As a result, we have to figure out a way to add this number to the diagonal entries of the inertia matrix when we compute forward or inverse dynamics. I'm not sure if pinocchio can handle this feature or if there are already some similar functions that deal with this.

I'm glad to provide more information. Thanks!

stephane-caron, 2023-05-05:
As far as I can tell continuous joints are handled all right? For instance the [Mini Cheetah description](https://github.com/Derek-TH-Wang/mini_cheetah_urdf/blob/1988bceb26e81f28594a16e7d5e6abe5cbb27ace/urdf/mini_cheetah.urdf) all joints are continuous:

```xml
    <joint name="RR_upper_leg_2_hip_motor_joint" type="continuous">
    <axis xyz="1 0 0"/>
    <parent link="RR_hip_motor"/>
    <child link="RR_upper_leg"/>
    <origin rpy="0 0 0" xyz="-0.053565 0 0"/>
    <limit effort="100" velocity="100"/>
    <joint_properties damping=".0" friction=".0"/>
  </joint>
```

And they are properly loaded as unbounded revolute joints by Pinocchio:

```python
In [1]: from robot_descriptions.loaders.pinocchio import load_robot_description

In [2]: mini_cheetah = load_robot_description("mini_cheetah_description")

In [3]: mini_cheetah.model.joints.tolist()
Out[3]: 
[JointModelRX
   index: 18446744073709551615
   index q: -1
   index v: -1
   nq: 1
   nv: 1,
 JointModelRUBX
   index: 1
   index q: 0
   index v: 0
   nq: 2
   nv: 1,
 JointModelRevoluteUnboundedUnaligned
   index: 2
   index q: 2
   index v: 1
   nq: 2
   nv: 1,
 JointModelRevoluteUnboundedUnaligned
   index: 3
   index q: 4
   index v: 2
   nq: 2
   nv: 1,
 JointModelRUBX
   index: 4
   index q: 6
   index v: 3
   nq: 2
   nv: 1,
 JointModelRevoluteUnboundedUnaligned
   index: 5
   index q: 8
   index v: 4
   nq: 2
   nv: 1,
...
```

Re. armature, have you checked existing issues? I just did a quick search on "armature" in existing issues, I see e.g. this point on [why you don't need to add diagonal entries to the inertia matrix](https://github.com/stack-of-tasks/pinocchio/issues/845#issuecomment-518566405) in computing $M(q)^{-1}$. There is also [this issue](https://github.com/stack-of-tasks/pinocchio/issues/671) and [this one](https://github.com/stack-of-tasks/pinocchio/issues/440) where new contributions are welcome.

jcarpent, 2023-05-14:
The armature and other transmission data will be fully handled in Pinocchio 3x. In the meantime, I would recommend taking them into account manually.

closed 2023-05-14 by jcarpent