stack-of-tasks/pinocchio #1953: After installing Pinocchio in C++, an error will be reported when importing URDF

xh666-666, 2023-05-08:
1.‘pinocchio::urdf::details::parseRootTree(std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const&amp;, pinocchio::urdf::details::UrdfVisitorBaseTpl&lt;double, 0&gt;&amp;)’When I reported this error, I added in the cmakeLists file:#link_libraries(/usr/local/lib/libhpp-fcl.so)
  #link_libraries(/usr/local/lib/libpinocchio.so)
  # link_libraries(/usr/local/lib/libpinocchio.so.2.6.17)
it can be compiled successfully.
However,when i run the code, now a new problem arises:Error:   File path_to_the_model_dir/home/xh/kortex-master/api_cpp/examples/GEN3-Lite.urdf does not exist
         at line 53 in /build/urdfdom-YMMa9X/urdfdom-1.0.0/urdf_parser/src/model.cpp
terminate called after throwing an instance of 'std::invalid_argument'
  what():  The file path_to_the_model_dir/home/xh/api_cpp/examples/GEN3-Lite.urdf does not contain a valid URDF model.
Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
  I made sure I have a correct URDF file.
I want to know how to solve such a problem, and hope to get your timely reply, thank you

stephane-caron, 2023-05-08:
From:

```
Error: File path_to_the_model_dir/home/xh/kortex-master/api_cpp/examples/GEN3-Lite.urdf 
```

You need to remove ``path_to_the_model_dir`` before ``/home``.

In the example that you are adapting, ``path_to_the_model_dir`` was an indication for you. It was meant to be replaced with the actual path (rather than adding it afterwards).

closed 2023-05-08 by stephane-caron

xh666-666, 2023-05-08:
Thank you very much for your help, this problem was successfully solved by me.