stack-of-tasks/pinocchio #1954: Does Pinocchio support non-chain robots ?

Acwok, 2023-05-10:
Hello,

My robot is a free-flyer cube with two ur5 arms. I'd like to use one arm tip as a fixed base and the other as the end-effector.  
I've tried MoveIt 2 but their ik solvers don't support urdf with non-chain structure:
**chain robot example:** base -> ... -> tip_link 
**non-chain robot example:** tip_link_1 -> ... -> base <- ... <- tip_link_2

I've looked at some tutorials, but I only found examples with a single UR5 with a fixed base.
However, as far as I understand, Pinocchio should also work with legged robots (which are very similar to our robot).  
Therefore I am planning to create a MoveIt 2 kinematics plugin using Pinocchio.  

I feel that integrating Pinocchio in a MoveIt plugin can be done easily, but I'm wondering if Pinocchio the right solution for my problem. 
**Does Pinocchio support non-chain robots ?**  
If yes, do you have examples using inverse kinematics on robots with floating base ?

jcarpent, 2023-05-10:
Thanks @Acwok for your interest and your suggestion of creating a MoveIt 2 module based on Pinocchio.

> Does Pinocchio support non-chain robots ?

The short answer is yes. Pinocchio has been designed for generic tree-like robots and has been recently extended for robots with closed kinematic chains.

> If yes, do you have examples using inverse kinematics on robots with floating base ?

Please see this tiny example in Python https://github.com/stack-of-tasks/pinocchio/blob/dc8333cb38e63a37e87f9b26c3ad7330b85b56d1/examples/inverse-kinematics.py

closed 2023-05-10 by jcarpent

Acwok, 2023-05-11:
Thank you very much! It seems to be exactly what I was looking for 😃  
I'm going to test it.

GuoPingPan, 2024-08-13:
how to deal will a dual-arm robot? Do I need to separate the left and right arms in the urdf of the humanoid robot？

stephane-caron, 2024-08-16:
You can find answers to your questions by looking at how the dual-arm robot is built up in the [`flying_dual_arm_ur3.py`](https://github.com/stephane-caron/pink/blob/5a506b8d9ea2ac0d993c4857fab0e7fdf60a4736/examples/flying_dual_arm_ur3.py) example linked above, specifically [this part](https://github.com/stephane-caron/pink/blob/5a506b8d9ea2ac0d993c4857fab0e7fdf60a4736/examples/flying_dual_arm_ur3.py#L87-L113).

GuoPingPan, 2024-08-20:
Thank you very much. But I found it is hard to install pink.