stack-of-tasks/pinocchio #1959: Pinocchio_The problem of inverse kinematics of robots

DawnQiu, 2023-05-11:
Dear Pinocchio team,

I wanted to start off by expressing my gratitude for all the hard work you put into creating such a powerful and user-friendly tool. Pinocchio has been an invaluable asset to me, and I appreciate all the effort that went into its development.

However, I have encountered an issue while using Pinocchio for computing the inverse kinematics of my robot. Specifically, I am having trouble with the convergence of the results. 

To give some context, I am trying to use the end-effector frame of my robot to calculate its transformation matrix relative to the base frame. I have successfully imported my URDF and identified the end-effector frame index using the code `pinocchio::FrameIndex end_effector_id = model.getFrameId("EndEffector_FRAME")`. From there, I defined a desired pose in Cartesian space and set up an iterative algorithm to compute the necessary joint angles. 

However, despite adjusting the iteration parameters such as increasing the number of iterations and adjusting the time step, I am still unable to achieve convergence. Additionally, the resulting joint angle values exceed the physical limits of my robot.

I have attached my source code and output results for reference.

```
#include "pinocchio/parsers/sample-models.hpp"
#include "pinocchio/spatial/explog.hpp"
#include "pinocchio/algorithm/kinematics.hpp"
#include "pinocchio/algorithm/jacobian.hpp"
#include "pinocchio/algorithm/joint-configuration.hpp"
#include <pinocchio/parsers/urdf.hpp>
#include <pinocchio/algorithm/frames.hpp>
#include <pinocchio/spatial/se3.hpp>
#include <pinocchio/spatial/motion.hpp>
#include <pinocchio/spatial/explog.hpp>
#include <pinocchio/multibody/fwd.hpp>
#include <pinocchio/multibody/joint/joint-generic.hpp>

int main(int /* argc */, char ** /* argv */)
{
    // Load robot model from URDF file
    pinocchio::Model model;
    pinocchio::urdf::buildModel("/home/dawn/kortex-master/api_cpp/examples/GEN3Lite.urdf", model);
    pinocchio::Data data(model);

    // Get number of degrees of freedom of robot
    //const int nq = model.nq;

    // Get the end-effector placement
    pinocchio::FrameIndex end_effector_id = model.getFrameId("EndEffector_FRAME");

    // Set the Desired Pose
    const pinocchio::SE3 oMdes(Eigen::Matrix3d::Identity(), Eigen::Vector3d(0.55, 0., 0.4662));

    // Init the configuration
    Eigen::VectorXd q = pinocchio::neutral(model);

    const double eps  = 1e-4;
    const int IT_MAX  = 1000;
    const double DT   = 1e-1;
    const double damp = 1e-6;

    pinocchio::Data::Matrix6x J(6,model.nv);
    J.setZero();

    bool success = false;

    typedef Eigen::Matrix<double, 6, 1> Vector6d;
    Vector6d err;

    Eigen::VectorXd v(model.nv);

    for (int i=0;;i++)
    {
        pinocchio::forwardKinematics(model,data,q);
        pinocchio::updateFramePlacement(model,data,end_effector_id);
        const pinocchio::SE3 end_effector_transform = data.oMf[end_effector_id];
        const pinocchio::SE3 dMi = oMdes.actInv(end_effector_transform);
        err = pinocchio::log6(dMi).toVector();

        if(err.norm() < eps)
        {
[... 124 lines cut ...]
880: error =   -0.0248392   0.00201194  -0.00192485  1.19502e-06  0.000256354 -6.04328e-06
890: error =   -0.0358908   0.00301602  -0.00224474  1.51943e-06  0.000142919 -2.10007e-06
900: error =   -0.0259404   0.00219278  -0.00166564   8.7028e-07  7.28922e-05 -1.03761e-06
910: error =   -0.0622967   0.00527467  -0.00520663  3.19359e-06  0.000157958 -7.12425e-06
920: error =   -0.0242975   0.00205872  -0.00204825  1.08232e-06  5.88353e-05 -2.74691e-06
930: error =   -0.0241535    0.0020501  -0.00173197  6.74483e-07  4.78535e-05 -1.32398e-06
940: error =   -0.0310633   0.00263229  -0.00264858  1.48034e-06  7.41931e-05 -3.67962e-06
950: error =   -0.0569686   0.00482879  -0.00481659  2.68885e-06  0.000131373 -6.40592e-06
960: error =   -0.0266532     0.002261  -0.00218587  9.00871e-07   5.7257e-05 -2.57486e-06
970: error =   -0.0314408   0.00266966   -0.0023435  8.89692e-07  5.86769e-05 -1.96623e-06
980: error =  -0.0241839  0.00205417 -0.00172568 4.98766e-07 4.32033e-05  -1.147e-06
990: error =   -0.0808512      0.00684  -0.00699465  4.34226e-06  0.000217189 -1.10219e-05

Warning: the iterative algorithm has not reached convergence to the desired precision

result: -0.0856276   -1.64197   -0.29861    39.2699   -1.34329   -39.1843

final error:    -0.029882   0.00252969  -0.00262287  1.54569e-06  7.76168e-05 -4.12558e-06
```

I am wondering if you have any insights into what may be causing this issue. Could there be an error in my code, or is there something else that I am overlooking? Any guidance or suggestions would be greatly appreciated.

Thank you again for all your hard work and dedication to creating such a valuable tool.

Best regards!

stephane-caron, 2023-05-11:
Be careful that your error vector here is expressed in the target frame `d` ($log_6({}_a M_b)$ is a vector in $a$):

```c++
err = pinocchio::log6(dMi).toVector();
```

But your Jacobian is in the body frame `i`:

```c++
pinocchio::computeFrameJacobian(model,data,q,end_effector_id,J);
```

Error and Jacobian should be in the same frame, e.g. in the body (end-effector) frame [as in this note](https://scaron.info/robotics/jacobian-of-a-kinematic-task-and-derivatives-on-manifolds.html#pose-task-residual).

I checked in the docs in this inconsistency is [already in the example](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_b-examples_d-inverse-kinematics.html#autotoc_md44). Perhaps the author of the example can chime in? Or shall we correct it to put everything in the body frame?

jcarpent, 2023-05-11:
You can also change the `dt` to do smaller steps.

DawnQiu, 2023-05-11:
Thank you for getting back to me. I think I understand what you're saying now. The issue is that the error coordinate system is different from the Jacobian matrix coordinate system.

The error reference coordinate system is based on the desired end-effector pose coordinate system, while the Jacobian matrix's reference coordinate system is based on the base coordinate system. However, I'm not sure how to solve this problem. Do you have any suggestions on how I can align the err reference coordinate system with the base coordinate system?

Thank you again for your help and I look forward to hearing from you.

stephane-caron, 2023-05-11:
@DawnQiu You can check out #1963 for a change set to apply to the example to get everything in the local frame (NB: not reviewed yet).

jcarpent, 2023-05-13:
Fix in #1963

closed 2023-05-13 by jcarpent