stack-of-tasks/pinocchio #1960: libboost not found while installing on Jetson Orin

Windson9, 2023-05-11:
Hello. Thanks for great work. I am trying to install Pin on a Jetson Orin but facing issues where I am not able to import the library. Below are some logs for reference. I am using Python3.8 and I tried to follow the steps mentioned in similar issues but no luck so far.

- `orin-2@orin-2:~/mayank/megapose6d$ python3 -m virtualenv -p python3 virtual_env
created virtual environment CPython3.8.10.final.0-64 in 261ms
  creator CPython3Posix(dest=/home/orin-2/mayank/megapose6d/virtual_env, clear=False, global=False)
  seeder FromAppData(download=False, pip=latest, setuptools=latest, wheel=latest, pkg_resources=latest, via=copy, app_data_dir=/home/orin-2/.local/share/virtualenv/seed-app-data/v1.0.1.debian.1)
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
orin-2@orin-2:~/mayank/megapose6d$ source virtual_env/bin/activate
(virtual_env) orin-2@orin-2:~/mayank/megapose6d$ python -c "import pinocchio"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'pinocchio'
(virtual_env) orin-2@orin-2:~/mayank/megapose6d$ pip install -U pip
Collecting pip
  Using cached pip-23.1.2-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.0.2
    Uninstalling pip-20.0.2:
      Successfully uninstalled pip-20.0.2
Successfully installed pip-23.1.2
(virtual_env) orin-2@orin-2:~/mayank/megapose6d$ pip install pin --no-cache-dir
Collecting pin
  Downloading pin-2.6.18-0-cp38-cp38-manylinux_2_28_aarch64.whl (5.8 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 5.8/5.8 MB 11.4 MB/s eta 0:00:00
Collecting cmeel (from pin)
  Downloading cmeel-0.43.1-py3-none-any.whl (16 kB)
Collecting hpp-fcl (from pin)
  Downloading hpp_fcl-2.3.2-0-cp38-cp38-manylinux_2_28_aarch64.whl (2.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.3/2.3 MB 12.0 MB/s eta 0:00:00
Collecting cmeel-urdfdom (from pin)
  Downloading cmeel_urdfdom-3.1.0.2-0-py3-none-manylinux_2_17_aarch64.whl (430 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 430.3/430.3 kB 13.7 MB/s eta 0:00:00
Collecting cmeel-console-bridge (from pin)
  Downloading cmeel_console_bridge-1.0.2.1-2-cp38-cp38-manylinux_2_28_aarch64.whl (24 kB)
Collecting cmeel-tinyxml (from pin)
  Downloading cmeel_tinyxml-2.6.2.1-2-cp38-cp38-manylinux_2_28_aarch64.whl (61 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.9/61.9 kB 82.2 MB/s eta 0:00:00
Collecting tomli<3.0.0,>=2.0.1 (from cmeel->pin)
  Downloading tomli-2.0.1-py3-none-any.whl (12 kB)
Collecting cmeel-assimp<6 (from hpp-fcl->pin)
  Downloading cmeel_assimp-5.2.5-10-py3-none-manylinux_2_17_aarch64.whl (12.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 12.9/12.9 MB 11.7 MB/s eta 0:00:00
Collecting cmeel-octomap<2 (from hpp-fcl->pin)
  Downloading cmeel_octomap-1.9.8.1-2-cp38-cp38-manylinux_2_28_aarch64.whl (672 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 672.7/672.7 kB 12.7 MB/s eta 0:00:00
Collecting eigenpy<4 (from hpp-fcl->pin)
  Downloading eigenpy-3.0.0-7-cp38-cp38-manylinux_2_28_aarch64.whl (2.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.4/2.4 MB 11.8 MB/s eta 0:00:00
Collecting cmeel-boost<1.83.0 (from eigenpy<4->hpp-fcl->pin)
  Downloading cmeel_boost-1.82.0-0-cp38-cp38-manylinux_2_28_aarch64.whl (34.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 34.4/34.4 MB 11.7 MB/s eta 0:00:00
Collecting numpy<1.25,>=1.23 (from cmeel-boost<1.83.0->eigenpy<4->hpp-fcl->pin)
  Downloading numpy-1.24.3-cp38-cp38-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (14.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 14.0/14.0 MB 11.7 MB/s eta 0:00:00
Installing collected packages: tomli, numpy, cmeel, cmeel-urdfdom, cmeel-tinyxml, cmeel-octomap, cmeel-console-bridge, cmeel-boost, cmeel-assimp, eigenpy, hpp-fcl, pin
Successfully installed cmeel-0.43.1 cmeel-assimp-5.2.5 cmeel-boost-1.82.0 cmeel-console-bridge-1.0.2.1 cmeel-octomap-1.9.8.1 cmeel-tinyxml-2.6.2.1 cmeel-urdfdom-3.1.0.2 eigenpy-3.0.0 hpp-fcl-2.3.2 numpy-1.24.3 pin-2.6.18 tomli-2.0.1
(virtual_env) orin-2@orin-2:~/mayank/megapose6d$ python
Python 3.8.10 (default, Mar 13 2023, 10:26:41) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/orin-2/mayank/megapose6d/virtual_env/lib/python3.8/site-packages/cmeel.prefix/lib/python3.8/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ImportError: libboost_python38.so.1.81.0: cannot open shared object file: No such file or directory
>>> quit()
(virtual_env) orin-2@orin-2:~/mayank/megapose6d$ pip --version
pip 23.1.2 from /home/orin-2/mayank/megapose6d/virtual_env/lib/python3.8/site-packages/pip (python 3.8)
(virtual_env) orin-2@orin-2:~/mayank/megapose6d$ python3 -c "import hppfcl;print(hppfcl.__version__)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/orin-2/mayank/megapose6d/virtual_env/lib/python3.8/site-packages/cmeel.prefix/lib/python3.8/site-packages/hppfcl/__init__.py", line 34, in <module>
    from .hppfcl import *
ImportError: libboost_python38.so.1.81.0: cannot open shared object file: No such file or directory`

nim65s, 2023-05-11:
a rebuild is in progress, I'll keep you updated when it's Ok, but it should require some hours.

jcarpent, 2023-05-11:
@Windson9 An alternative solution is to rely on conda.

Windson9, 2023-05-11:
The issue is I cannot use Pytorch with Cuda enabled as there is no whl package for Jetson orin in conda environment. So I have to use either virtual env or local setup

nim65s, 2023-05-11:
meanwhile, you may try to downgrade boost: `pip install "cmeel-boost < 1.82.0"`

Windson9, 2023-05-11:
The error still persist after installing cmeel-boost < 1.82.0.

(virtual_env) orin-2@orin-2:/mayank/megapose6d$ pip install "cmeel-boost < 1.82.0"
Collecting cmeel-boost<1.82.0
  Downloading cmeel_boost-1.81.0-12-cp38-cp38-manylinux_2_28_aarch64.whl (33.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 33.9/33.9 MB 789.1 kB/s eta 0:00:00
Requirement already satisfied: cmeel in ./virtual_env/lib/python3.8/site-packages (from cmeel-boost<1.82.0) (0.43.1)
Requirement already satisfied: numpy>=1.23 in ./virtual_env/lib/python3.8/site-packages (from cmeel-boost<1.82.0) (1.24.3)
Requirement already satisfied: tomli<3.0.0,>=2.0.1 in ./virtual_env/lib/python3.8/site-packages (from cmeel->cmeel-boost<1.82.0) (2.0.1)
Installing collected packages: cmeel-boost
  Attempting uninstall: cmeel-boost
    Found existing installation: cmeel-boost 1.82.0
    Uninstalling cmeel-boost-1.82.0:
      Successfully uninstalled cmeel-boost-1.82.0
Successfully installed cmeel-boost-1.81.0
(virtual_env) orin-2@orin-2:/mayank/megapose6d$ python3 -c "import pinocchio"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/orin-2/mayank/megapose6d/virtual_env/lib/python3.8/site-packages/cmeel.prefix/lib/python3.8/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ImportError: libboost_python38.so.1.82.0: cannot open shared object file: No such file or directory
(virtual_env) orin-2@orin-2:~/mayank/megapose6d$

nim65s, 2023-05-13:
Hi,

This should be fixed.

Can you confirm on your sides, @Windson9 & @stephane-caron ?

kingjin94, 2023-05-15:
Thx, that solved the problem for >= Py3.8. Sadly a new problem appeared for 3.7 (if you still like to support it):

```/usr/local/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/pinocchio/__init__.py:6: in <module>
    from .pinocchio_pywrap import *
E   ImportError: /usr/local/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/pinocchio/../../../../lib/libhpp-fcl.so: undefined symbol: _ZNK6Assimp8IOSystem16CurrentDirectoryEv```

stephane-caron, 2023-05-15:
Gotcha. Yes, that would be nice for 3.7 users to still be able to install a legacy version of the software. (As far as Pink is concerned it will just follow your move and set 3.8 as the minimum Python version for the next release.)

stephane-caron, 2023-05-16:
Potentially not the same issue but I get the following import error with Python 3.10:

```
$ conda create -n test4 python=3.10
$ conda activate test4
$ pip install pin
Collecting pin
  Downloading pin-2.6.18-2-cp310-cp310-manylinux_2_28_x86_64.whl (6.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 6.1/6.1 MB 1.0 MB/s eta 0:00:00
Collecting cmeel (from pin)
  Using cached cmeel-0.43.1-py3-none-any.whl (16 kB)
Collecting cmeel-console-bridge<2,>=1.0.2.2 (from pin)
  Using cached cmeel_console_bridge-1.0.2.2-1-py3-none-manylinux_2_28_x86_64.whl (23 kB)
Collecting cmeel-tinyxml<3,>=2.6.2.2 (from pin)
  Using cached cmeel_tinyxml-2.6.2.2-1-py3-none-manylinux_2_28_x86_64.whl (62 kB)
Collecting cmeel-urdfdom<4,>=3.1.0.3 (from pin)
  Using cached cmeel_urdfdom-3.1.0.3-0-py3-none-manylinux_2_28_x86_64.whl (405 kB)
Collecting hpp-fcl<4,>=2.3.3 (from pin)
  Downloading hpp_fcl-2.3.3-3-cp310-cp310-manylinux_2_28_x86_64.whl (2.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.3/2.3 MB 1.2 MB/s eta 0:00:00
Collecting cmeel-assimp<6,>=5.2.5.1 (from hpp-fcl<4,>=2.3.3->pin)
  Using cached cmeel_assimp-5.2.5.1-0-py3-none-manylinux_2_28_x86_64.whl (13.6 MB)
Collecting cmeel-octomap<2,>=1.9.8.2 (from hpp-fcl<4,>=2.3.3->pin)
  Using cached cmeel_octomap-1.9.8.2-0-py3-none-manylinux_2_28_x86_64.whl (694 kB)
Collecting eigenpy<4 (from hpp-fcl<4,>=2.3.3->pin)
  Downloading eigenpy-3.0.0-7-cp310-cp310-manylinux_2_28_x86_64.whl (2.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.5/2.5 MB 1.5 MB/s eta 0:00:00
Collecting tomli<3.0.0,>=2.0.1 (from cmeel->pin)
  Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
Collecting cmeel-boost<1.83.0 (from eigenpy<4->hpp-fcl<4,>=2.3.3->pin)
  Downloading cmeel_boost-1.82.0-0-cp310-cp310-manylinux_2_28_x86_64.whl (34.6 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 34.6/34.6 MB 1.7 MB/s eta 0:00:00
Collecting numpy<1.25,>=1.23 (from cmeel-boost<1.83.0->eigenpy<4->hpp-fcl<4,>=2.3.3->pin)
  Using cached numpy-1.24.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (17.3 MB)
Installing collected packages: tomli, numpy, cmeel, cmeel-urdfdom, cmeel-tinyxml, cmeel-octomap, cmeel-console-bridge, cmeel-boost, cmeel-assimp, eigenpy, hpp-fcl, pin
python
Successfully installed cmeel-0.43.1 cmeel-assimp-5.2.5.1 cmeel-boost-1.82.0 cmeel-console-bridge-1.0.2.2 cmeel-octomap-1.9.8.2 cmeel-tinyxml-2.6.2.2 cmeel-urdfdom-3.1.0.3 eigenpy-3.0.0 hpp-fcl-2.3.3 numpy-1.24.3 pin-2.6.18 tomli-2.0.1
$ python
Python 3.10.11 | packaged by conda-forge | (main, May 10 2023, 18:58:44) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/scaron/.micromamba/envs/test4/lib/python3.10/site-packages/cmeel.prefix/lib/python3.10/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ImportError: /lib/x86_64-linux-gnu/libboost_python38.so.1.71.0: undefined symbol: _Py_fopen
```

Do you want to spin a separate issue for this?

nim65s, 2023-05-16:
@stephane-caron : I can't reproduce that one.
```Dockerfile
FROM ubuntu:22.04

ADD https://repo.anaconda.com/archive/Anaconda3-2022.10-Linux-x86_64.sh /install-conda

RUN bash /install-conda -b
RUN eval "$(/root/anaconda3/bin/conda shell.bash hook)" \
 && conda init

CMD eval "$(/root/anaconda3/bin/conda shell.bash hook)" \
 && conda create -n test4 python=3.10 -y \
 && conda activate test4 \
 && pip install pin \
 && python -c "import pinocchio; print(pinocchio.__file__)"
 ```
 
 correctly shows:
 ```
 /root/anaconda3/envs/test4/lib/python3.10/site-packages/cmeel.prefix/lib/python3.10/site-packages/pinocchio/__init__.py
 ```
 
 I made another test with:
 ```Dockerfile
 FROM continuumio/miniconda3

RUN conda create -n test4 python=3.10 -y
SHELL ["conda", "run", "-n", "test4", "/bin/bash", "-c"]
CMD pip install pin && python -c "import pinocchio; print(pinocchio.__file__)"
```

correctly shows:
```
/opt/conda/envs/test4/lib/python3.10/site-packages/cmeel.prefix/lib/python3.10/site-packages/pinocchio/__init__.py
```
 
 Am I messing with something here ?

stephane-caron, 2023-05-16:
Hmm, that's curious. Suspecting a cache issue I tried

```
# in the conda environment
pip install pin --no-cache-dir --upgrade --force-reinstall
```

But it yields the same error.

stephane-caron, 2023-05-16:
It seems that importing Pinocchio looks for Boost::Python in `/lib`, where it is indeed installed on my system (not sure why). What I don't understand is why it would try to read that one rather than the correct one in `${CONDA_PREFIX}/lib/python3.10/site-packages/cmeel.prefix/lib/libboost_python310.so`.

Going to `${CONDA_PREFIX}/lib/python3.10/site-packages/cmeel.prefix/lib/python3.10/site-packages/pinocchio` and running `ldd` yields:

```console
$ ldd pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so | grep boost_python
	libboost_python310.so.1.82.0 => /home/scaron/.micromamba/envs/test4bis/lib/python3.10/site-packages/cmeel.prefix/lib/python3.10/site-packages/pinocchio/./../../../../lib/libboost_python310.so.1.82.0 (0x00007f77f46c3000)
	libboost_python38.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_python38.so.1.71.0 (0x00007f77f3c97000)
```

So something there is linked against Python 3.8?

There are a couple of other py38 boost libraries in there:

```console
$ ldd pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so | grep 1.71
	libboost_python38.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_python38.so.1.71.0 (0x00007fd721488000)
	libboost_filesystem.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007fd721468000)
	libboost_system.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_system.so.1.71.0 (0x00007fd721463000)
	libboost_serialization.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_serialization.so.1.71.0 (0x00007fd72141f000)
	libboost_atomic.so.1.82.0 => /home/scaron/.micromamba/envs/test4bis/lib/python3.10/site-packages/cmeel.prefix/lib/python3.10/site-packages/pinocchio/./../../../../lib/libboost_atomic.so.1.82.0 (0x00007fd721371000)
	libboost_chrono.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_chrono.so.1.71.0 (0x00007fd720716000)
```

For instance the same happens with `boost_filesystem`:

```console
$ ldd pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so | grep boost_filesystem
	libboost_filesystem.so.1.82.0 => /home/scaron/.micromamba/envs/test4bis/lib/python3.10/site-packages/cmeel.prefix/lib/python3.10/site-packages/pinocchio/./../../../../lib/libboost_filesystem.so.1.82.0 (0x00007f7bfd79f000)
	libboost_filesystem.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007f7bfcd77000)
```

stephane-caron, 2023-05-16:
@nim65s I don't see this error in GitHub Actions, so I'm not sure what's going on. Perhaps the issue can be reproduced with:

```dockerfile
FROM ubuntu:20.04
```

Otherwise I'm concerned something else would be causing this on my system. I checked again with `python=3.9` and I get a similar linker issue now, while it worked yesterday :thinking:

nim65s, 2023-05-16:
@stephane-caron maybe something in your env variables or in your /etc/ld.so.conf ?

nim65s, 2023-05-22:
```
docker run --rm -it python:3.7 bash -c "pip install -U pip && pip install pin && python -c 'import pinocchio; print(pinocchio.__version__)'"
[…]
2.6.18
```

Fixed for Python 3.7: Pinocchio 2.6.18 will be the latest available version.

@Windson9 @stephane-caron : I'll close this. If you still have issues, don't hesitate to re-open if you think this is the same, or open another one :)

closed 2023-05-22 by nim65s

stephane-caron, 2023-05-23:
Will do, thanks for following up.

haivuvan, 2023-06-02:
@nim65s I have same error when run in virtual enviroment.
    from .pinocchio_pywrap import *
ImportError: libboost_python38.so.1.80.0: cannot open shared object file: No such file or directory

I tried with python 3.8.10 and python 3.9.16 and both version have same issue.
I install lib using conda enviroment the issue not occur. But I need to run in Jetson Orin. How I can fix this issue in virtual enviroment ?

nim65s, 2023-06-02:
@haivuvan : you need to upgrade your packages. Maybe something like:
```
python -m pip install -U pip
python -m pip install -U cmeel-boost eigenpy hpp-fcl pin
```
After that, if you still have issues, please post the result of `python -m pip freeze` so we can inspect this a bit further.

haivuvan, 2023-06-05:
@nim65s, after upgrade package the issue still occur.
please check attached pip3 enviroments
[pip3_8_10_env.txt](https://github.com/stack-of-tasks/pinocchio/files/11648257/pip3_8_10_env.txt)
[pip3_9_16_env.txt](https://github.com/stack-of-tasks/pinocchio/files/11648259/pip3_9_16_env.txt)

nim65s, 2023-06-05:
you still have eigenpy v2.9.2, while the latest version should be v3.1.0. And this version is available both for x86_64 and aarch64 as a manylinux_2_28.

What OS are you running ?

Does `python -m pip install -U pip && python -m pip install eigenpy` show any erro ?

Did you ever try to install eigenpy from another source on this machine ?

haivuvan, 2023-06-06:
@nim65s I use eigenpy v2.9.2 because later version remove function that I want to use.
I'm running on Ubuntu 20.04 for both laptop and Jetson Orin.

nim65s, 2023-06-06:
This is a totally different issue then. 

You told me you upgraded, while you're actively trying not to upgrade one package. 

This is really not nice for the time I spend helping you, and this is not nice for every other people subscribed to this issue getting notifications about your issue which has nothing to to with them.

Anyway, we have a [XY problem](https://xyproblem.info/) here, please discuss about that eigenpy function, I'm not going to spend time finding a set of versions that workaround your initial problem until we agree this is the right option or the only one.

NB: Pinning a package is fine, but you must pin also pin all its dependencies, as we can't ensure every version of something will work with every version of everything else, and we can't provide pre-built binaries for such cases.

CodingCatMountain, 2024-08-07:
Hi, @nim65s . I also met the problem in the python3.10 virtual environment created via miniforge3 today.
```
from .pinocchio_pywrap import *
ImportError: /lib/x86_64-linux-gnu/libboost_python38.so.1.71.0: undefined symbol: _Py_fopen
```

I tried your solution mentioned above: 
```
python -m pip install -U pip
python -m pip install -U cmeel-boost eigenpy hpp-fcl pin
```
While the problem is not solved. I just want to kindly report this problem to you and maybe the solution I tried can help other people. :) 
What I finally solve this problem is running the following command:
```
conda install pinocchio
```

By the way, my os  environment is Ubuntu20.04.

nim65s, 2024-08-07:
@CodingCatMountain : this issue is closed, and what you experience has nothing to do with it. Also don't mix pin pypi packaage with pinocchio conda package. use one or the other.

CodingCatMountain, 2024-08-08:
@nim65s OK. It's screechy to say my experience has nothing to do with it.. But you are right, I shall not mix pin pypi package and pinocchio conda package even though my program can run now.