stack-of-tasks/pinocchio #1965: ubuntu 22.04 gepetto viewer can not be initialized

sancelot, 2023-05-12:
I setted up pinocchio in an ubuntu 22.04 system

I launched gepetto-gui 

and tried to run this example program
https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_b-examples_display_a-gepetto-viewer.html

I changed the urdf filename with mine.

unfortunately it replies with this message 
Check whether gepetto-viewer is properly started

Error while loading the viewer model. It seems you should start gepetto-viewer

'GepettoVisualizer' object has no attribute 'viewer'

jcarpent, 2023-05-12:
Please switch to meshcat for simplicity. Gepetto-gui is no more recommended.

jcarpent, 2023-05-12:
@nim65s Could you provide some guidance for that issue?

sancelot, 2023-05-12:
From my developer experience I would prefer web 3D applications like meshcat.
I managed to use meshcat . But from my viewpoint it is not easy to without advanced documentation for noobs :
export ROS_PACKAGE_PATH , to define where you stored your meshes if you defined your urdf robot
pip install --user meshcat
overload your PYTHONPATH to ~/.local/lib/python.../site-packages (otherwise browser replies with Uncaught ReferenceError: MeshCat is not defined)

jcarpent, 2023-05-12:
Normally, you should not use PYTHONPATH if meshcat is installed alongside Pinocchio.
The ROS_PACKAGE_PATH is used by the Pinocchio parser to retrieve the geometries. But in general, you can change your URDF to just use the `file://` option

jcarpent, 2023-05-12:
@sancelot Could you provide a tiny example of using meschat with the documentation to help us improving the doc in general?

closed 2023-05-12 by jcarpent

sancelot, 2023-05-30:
@jcarpent 
Installing meshcat in your home dir : 
pip install --user meshcat 

meshcat is now setted up in your ~ folder , in the directory ~/.local/lib/pythonXX/site-packages (where XX should reflect  your python version)

FAQ
When launching meshcat, console logs Uncaught ReferenceError: MeshCat is not defined:
overload your PYTHONPATH using this command before launching meshcat :
export PYTHONPATH=~/.local/lib/pythonXX/site-packages # where XX reflects your python version

your robot meshes are not displayed and not found : 
set ROS_PACKAGE_PATH variable to follow the folder where your urdf meshes are located

jcarpent, 2023-05-30:
Thanks a lot @sancelot for this tiny example!