stack-of-tasks/pinocchio #1967: CI failing with libboost_python38.so.1.81.0 not found

stephane-caron, 2023-05-12:
Since yesterday we have observed the Pink CI failing with:

```
  File "/home/runner/work/pink/pink/pink/configuration.py", line 28, in <module>
    import pinocchio as pin
  File "/home/runner/work/pink/pink/.tox/coverage/lib/python3.8/site-packages/cmeel.prefix/lib/python3.8/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ImportError: libboost_python38.so.1.81.0: cannot open shared object file: No such file or directory
```

The dependencies configured in test environments are:

```yaml
    numpy
    pin >=2.6.4
    qpsolvers >=2.7.2
    quadprog >=0.1.11
    robot_descriptions >=1.4.1
```

stephane-caron, 2023-05-12:
This issue may be related to https://github.com/stack-of-tasks/pinocchio/issues/1960

stephane-caron, 2023-05-12:
Timing-wise, this issue started happening yesterday:

[image]

nim65s, 2023-05-12:
This is the same issue: I'm rebuilding everything for boost 1.82

jcarpent, 2023-05-12:
Close as duplicated with #1960.

closed 2023-05-12 by jcarpent