stack-of-tasks/pinocchio #1974: Question about Jacobian Matrix

Wayne-Z, 2023-05-26:
Hello,
    I tried to calculate the Jacobian matrix of the end effector of the kinova-gen3 robot in the world frame to calculate `v_ee` by `v_ee=Je*q_d`, I used `getFrameJacobian(model,data,end_effector_joint_id, WORLD.Je)`  and `getJointJacobian(model,data,7,WORLD,Je)` to get the Jacobian matrix, their results are the same, but different from the correct Jocabian matrix of the end effector, as shown in the figure1.
    Then I calculate the jacobian matrix of another link  use `getFrameJacobian`, and I found that the Jacobian matrix of this link is same to the end_effector_joint with the last few columns of the matrix become 0. But they are different point in space and the matrix should be different.
    So, I would like to ask which point in space the Jacobi matrix obtained from Pinocchio corresponds to？Please correct me if I have used it incorrectly.
[image]

Wayne-Z, 2023-05-27:
I think I found the issue. The jacobian matrix  in world frame get from pinocchio is relative to `World Coordinate Origin` rather than `target point`.  It means that, it is `z_i \times ^i p_o^o` rather than `z_i \times ^i p_n^o` in the first three rows of J.[image]
[image]

jcarpent, 2023-05-27:
Yes, you're right. You should work with LOCAL or LOCAL_WORLD_ALIGNED type of frame.

closed 2023-05-27 by jcarpent