stack-of-tasks/pinocchio #1994: Visual and Collision meshes not on same pose in Meshcat

joao-pm-santos96, 2023-06-16:
Hi!
When I run the meshcat-viewer.py example, I notice that the collision and visual meshes are not in the same place. In this a bug or the intended behavior?

**VISUAL ONLY**
[image]

**COLLISION ONLY**
[image]

**BOTH**
[image]

jcarpent, 2023-06-16:
By default, the visual information is not displayed. You should activate it via `viewer.displayVisuals(True)`

jcarpent, 2023-06-16:
We should simplify the behavior

joao-pm-santos96, 2023-06-16:
That did solve it! Thanks!

closed 2023-06-16 by joao-pm-santos96

stephane-caron, 2023-06-19:
OK, I see why:

```python
if convex is not None:
    placement = pin.SE3.Identity()
    placement.translation[0] = 2.0
    geometry = pin.GeometryObject("convex", 0, convex, placement)
    geometry.meshColor = np.ones((4))
    visual_model.addGeometryObject(geometry)
```

I don't understand what this block is trying to showcase, but if breaks down future calls to `viz.display`:

```
Traceback (most recent call last):
  File "/.../src/pinocchio/examples/meshcat-viewer.py", line 101, in <module>
    viz.display()
  File "/.../.micromamba/envs/pin-310/lib/python3.10/site-packages/pinocchio/visualize/meshcat_visualizer.py", line 523, in display
    self.updatePlacements(pin.GeometryType.VISUAL)
  File "/.../.micromamba/envs/pin-310/lib/python3.10/site-packages/pinocchio/visualize/meshcat_visualizer.py", line 537, in updatePlacements
    M = geom_data.oMg[geom_model.getGeometryId(visual.name)]
IndexError: Index out of range
double free or corruption (!prev)
```

@jcarpent Let's remove it from this example? Or if it is part of a good workflow (which one?) let's move it to a separate example?

jcarpent, 2023-06-19:
Could you tell me how to reproduce the current issue?

jcarpent, 2023-06-19:
> Or if it is part of a good workflow (which one?) let's move it to a separate example?

This example highlights the fact that we can load convex geometries provided by hppfcl.

joao-pm-santos96, 2023-06-20:
> I tested it on the following two setups:
> 
> * Conda, Python 3.10, Pinocchio 2.6.19 (from yesterday)
> * PyPI, Python 3.8, Pinocchio 2.6.19

How did you update your conda env Pinocchio version? I've not been able to, so far...

stephane-caron, 2023-06-20:
> How did you update your conda env Pinocchio version? I've not been able to, so far...

You can try `conda update pinocchio` again today. I think 2.6.19 was released on conda yesterday :wink: