stack-of-tasks/pinocchio #1995: HEAD build failure

chenrui333, 2023-06-18:
👋 While regression build 2.6.18, I ran into some cmake config error as below. And I found it also reproducible with the head build as well.

The cmake config error message is as below:
```
-- Found Boost: /opt/homebrew/include (found version "1.82.0") found components: chrono serialization
-- Minimal C++ standard upgraded to 14
-- Incompatible C++ standard detected: upgrade required from 11 to >= 14
CMake Error at /opt/homebrew/lib/cmake/hpp-fcl/cxx-standard.cmake:125 (message):
  CMAKE_CXX_STANDARD_REQUIRED set - cannot upgrade incompatible standard
Call Stack (most recent call first):
  /opt/homebrew/lib/cmake/hpp-fcl/hpp-fclConfig.cmake:186 (CHECK_MINIMAL_CXX_STANDARD)
  cmake/package-config.cmake:110 (find_package)
  CMakeLists.txt:184 (ADD_PROJECT_DEPENDENCY)
```

relates to: https://github.com/Homebrew/homebrew-core/pull/134062

jcarpent, 2023-06-18:
Could you provide the bash lines to reproduce this issue ?

jcarpent, 2023-06-18:
I've been able to reproduce the issue with `brew install --only-dependencies --verbose --build-bottle pinocchio`.
I will provide a fix soon.

jcarpent, 2023-06-18:
The issue has been fixed in the current devel branch of Pinocchio. I will proceed to a new release today ;)

chenrui333, 2023-06-19:
> I've been able to reproduce the issue with `brew install --verbose --build-bottle pinocchio`. I will provide a fix soon.

Sorry about the late reply, good to know you nailed it. :)

chenrui333, 2023-06-19:
I am also gonna try your commit in the meanwhile.

chenrui333, 2023-06-19:
Looks like there are lots of changes on devel branch XD, let me know when you can cut a new release

chenrui333, 2023-06-19:
I can confirm 2.6.19 rel works for me in my local. And I just filed a new version bump PR.

jcarpent, 2023-06-19:
Thanks @chenrui333 for the nice feedback. I will close the issue then.

closed 2023-06-19 by jcarpent