stack-of-tasks/pinocchio #2003: how to separately get the 'P' or  'R'  from pinocchio.pinocchio_pywrap.SE3

mzq740556549, 2023-06-27:
I got a TypeError  which is 'numpy.ndarray' object is not callable, when I use 'position = data.oMf[idx].translation()' to get the translation or rotation

```
Traceback (most recent call last):
  File "/home/ask-opc/catkin_kinova/src/demos/nodes/arm_catch.py", line 259, in <module>
    se,pose,rot = forwardk(joint_id, q)
  File "/home/ask-opc/catkin_kinova/src/demos/nodes/Kinematics.py", line 37, in forwardk
    position = data.oMf[idx].translation()
TypeError: 'numpy.ndarray' object is not callable
```

MegMll, 2023-06-27:
Hello ! 
`translation` is not a function it's just an attribute, so no need to use parenthesis.
This should work :
`position = data.oMf[idx].translation`
Have a look at the example folder, if you want to get more info.

mzq740556549, 2023-06-27:
OH ! It works ! Thank you very much !!!