stack-of-tasks/pinocchio #2005: pinocchio::sdf::buildModel error

areofire, 2023-07-01:
I'm using pinocchio3-preview and trying to load a robot model with function "pinocchio::sdf::buildModel" .

Part of my code is:   
  Model model;
  pinocchio::urdf::buildModel(urdf_filename,model);
  PINOCCHIO_STD_VECTOR_WITH_EIGEN_ALLOCATOR(RigidConstraintModel) contact_models;
  pinocchio::sdf::buildModel(urdf_filename,model,contact_models);

But a segmentation fault come out in parseRootTree， const std::string childName=
       jotingElement->GetElement("child")->Get<std::string>();;

Maybe i miss something or the model i used is not correct, I  tried the model from here:
[models.zip](https://github.com/stack-of-tasks/pinocchio/files/10475001/models.zip), from #1839;
and here:
https://github.com/osrf/gazebo_tutorials/tree/master/kinematic_loop/four_bar_split_fixed_sdf/model.sdf

Both of them come out the same error above.

Is there a example of sdf model?

Many thanks

jcarpent, 2023-07-01:
Please provide a reproducible example and the way to compile and launch it.

areofire, 2023-07-01:
[pin_c_test_.zip](https://github.com/stack-of-tasks/pinocchio/files/11925608/pin_c_test_.zip)
cd build
cmake ..
make

closed 2023-07-03 by areofire

jcarpent, 2023-07-03:
@areofire Did you find a fix for your issue?

areofire, 2023-07-03:
Not yet. I have spend much time on installing pin-view3, but all failed. As in example above, to use pinnocchio3, i have to include relevant  file in my project. 

I got  a pinocchio::sdf::buildModel error and  still working on fixing it.

areofire, 2023-07-04:
Maybe the model i used is wrong, there wasn't a "static " parameter in graph.mapOfJoints. 
[image]

areofire, 2023-07-04:
Could you give an example of sdf file, thank you !

areofire, 2023-07-04:
@jcarpent