stack-of-tasks/pinocchio #2012: DLL load failed while importing pinocchio_pywrap： Can not find module. In windows 11, conda environment.

xiangyushao-cn, 2023-07-11:
## Bug description

Can not find pinocchio_pywrap when run the code,   in windows-anaconda environment

  File "d:/2-PostDoc/_Postdoc1stYear/11-Humanoid Robots/__Pragramms/From Ding/locomotion_nlp_kmp_pybullet_python-main/locomotion_nlp_kmp_pybullet_python-main/robot_nmpc_nlp_ik.py", line 12, in <module>
    import pinocchio as pin
  File "C:\Users\xiang\.conda\envs\humanoid_python30810\lib\site-packages\pinocchio\__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ImportError: DLL load failed while importing pinocchio_pywrap:

The pip list is as follows: 
Brotli             1.0.9
certifi            2023.5.7
charset-normalizer 3.2.0
idna               3.4
Mosek              10.0.46
numpy              1.24.4
packaging          23.1
pip                23.1.2
platformdirs       3.8.1
pooch              1.7.0
pybullet           3.2.4
PySocks            1.7.1
requests           2.31.0
scipy              1.10.1
setuptools         68.0.0
typing_extensions  4.7.1
urllib3            2.0.3
wheel              0.40.0
win-inet-pton      1.1.0

jcarpent, 2023-07-11:
Could you try to install Pinocchio in a fresh and clean conda environment and provide the output of `python -c import pinocchio`.

jcarpent, 2023-07-11:
> _File "C:\Users\xiang.conda\envs\humanoid_python30810\lib\site-packages\pinocchio__init__.py", line 6, in

This file does not belong to the Pinocchio project. At first glance, the issue seems to be on your side.

jcarpent, 2023-07-13:
Closing this issue as there is not feedback from @xiangyushao-cn

closed 2023-07-13 by jcarpent