stack-of-tasks/pinocchio #2020: ImportError in Conda Installation

shallwing, 2023-07-21:
I have installed pinocchio for Conda in this tutorial：      [Conda Install](https://stack-of-tasks.github.io/pinocchio/download.html)
But when I import pinocchio in python, It reported this error：

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/shallwing/anaconda3/envs/openmmlab/lib/python3.8/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
  File "/opt/ros/noetic/lib/python3/dist-packages/hppfcl/__init__.py", line 34, in <module>
    from .hppfcl import *
ImportError: /opt/ros/noetic/lib/python3/dist-packages/hppfcl/hppfcl.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN7eigenpy9NumpyType7getTypeEv

So what is it？ Some libs that I forget to install?

I have install ROS Noetic in Ubuntu 20.04 and create a conda environment for python3.8.

nim65s, 2023-07-21:
the python interpreters from conda and your system are not compatible. You can either remove conda and install pinocchio from another source (ros, robotpkg, manually, etc.) to get a pinocchio working with your ROS, or remove your ROS and install it with conda

shallwing, 2023-07-23:
> the python interpreters from conda and your system are not compatible. You can either remove conda and install pinocchio from another source (ros, robotpkg, manually, etc.) to get a pinocchio working with your ROS, or remove your ROS and install it with conda

Then I selected the measure of Linux install [Linux Install](https://stack-of-tasks.github.io/pinocchio/download.html) to restart, I successfully imported the submodule pinocchio.buildModelFromUrdf to build my own arm mode. Thank you!