stack-of-tasks/pinocchio #2022: Missing computeDistances() in pinocchio.casadi

joao-pm-santos96, 2023-07-24:
Hello once again!

I've been trying to compute some derivatives using pinocchio and casadi following https://github.com/stack-of-tasks/pinocchio/discussions/2019 .

Yet, I've encountered the following error `AttributeError: module 'pinocchio.casadi' has no attribute 'computeDistances'`. Is this method planned to be integrated? 

I've installed though this tutorial: https://github.com/nmansard/jnrh2023#conda

My `pinocchio.__version__` returns '2.99.0'

jcarpent, 2023-07-24:
This is expected as [HPP-FCL](https://github.com/humanoid-path-planner/hpp-fcl), the framework for collision detection used in Pinocchio, is not compatible with autodiff.
You can see other developments by our team going toward this goal in this [paper](https://hal.science/hal-03780482v2/file/Differentiable_collision_detection.pdf).

closed 2023-07-24 by jcarpent

joao-pm-santos96, 2023-07-25:
Thank you. Only recently I've discovered **HAL** it it has been a great source of knowledge. 

So, as I understand from your answer and referred paper, has of now there is no way to compute the jacobian of collisions or collision distances, is that right?

jcarpent, 2023-07-25:
Yes, this intrinsic to the collision problem.